shivasiddharth/GassistPi #1054: USB MIC & AUDIO JACK setup issue - routes through USB audio

legoszm, 2020-09-09:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
GassistPi 
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 00000000eed7e8dd

```
Paste your Board and OS details here.   
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:
```
I follow the steps for USB MIC and AUDIO JACK during setup. Speaker test at steps 3.4 and 5 routes the audio through the audio jack as expected. Once setup is complete and the assistant is running (as a service), the audio reverts to USB audio (my microphone is a Jabra speaker & mic).

Speaker tests after the installation route audio to the USB audio device. I have attached aplay & arecord output
[aplay_arecord_list.txt](https://github.com/shivasiddharth/GassistPi/files/5197301/aplay_arecord_list.txt)

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
Maybe, probably not.

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

legoszm, 2020-09-09:
I note that other than audio output not as desired, the assistant functions as expected.

shivasiddharth, 2020-09-09:
can you post the output of 
```
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
```

I guess the audio config files are getting overwritten after restart. Is that USB device 2 in 1  with both audio out and mic ?

legoszm, 2020-09-09:
Thanks for your prompt reply! Below is the output as requested.

And yes, the USB device is a 2 in one with both audio out and mic.

sudo nano /etc/asound.conf

cm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

sudo nano ./.asoundrc

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

shivasiddharth, 2020-09-10:
Can you post the output .asoundrc ?

legoszm, 2020-09-10:
Apologies if I'm doing this wrong... I'm someone new to this.

**.asoundrc**

```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

shivasiddharth, 2020-09-10:
Can you post the output of:
```
aplay -l
arecord -l
```

legoszm, 2020-09-10:
**aplay -l**

```
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 6/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: USB [Jabra SPEAK 510 USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

**arecord -l**

```
**** List of CAPTURE Hardware Devices ****
card 1: USB [Jabra SPEAK 510 USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

shivasiddharth, 2020-09-11:
From the raspi-config. Choose 3.5mm jack as the audio device.

legoszm, 2020-09-11:
Here are the steps I did:

1) Rebooted. Assistant started with USB audio. Audio test `speaker-test t wav` also through USB audio.

2) I then ran `sudo raspi-config`, and set the "force to 3.5 mm jack" option

3) Return to terminal, this is the output:

```
numid=3,iface=MIXER,name='Master Playback Volume'
  ; type=INTEGER,access=rw------,values=2,min=0,max=65536,step=1
  : values=1,1
```

4) Run `speaker-test t wav` again, no audio from any output

5) Try to use assistant, no responses.

6) Wait a few moments, assistant returns to working normally. Speaker test still yields no audio.

7) Reboot Pi, all changes are undone and everything operates as described in (1) above.

gelaw2, 2020-09-21:
@legoszm @shivasiddharth I had a similar issue with my system.  I found this solution worked for me:
I was able to resolve my sound issue by 1) rerun the USB Mic and Audio Jack Users Script. 2) open .asoundrc and change the hardware definitions to match my configuration. Reboot.
Maybe this will help you?

Friday8229, 2020-11-23:
This is my error can you help me 

sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh  
Renaming /etc/asound.conf to /etc/asound.conf.bak...
amixer: Cannot find the given element from control default

Output of aplay -l

aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
  
 Output of arecord -l

arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
  
Output of all the hree commands is same
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
sudo nano .asoundrc: 
 
  pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:0,0"
  }
}

shivasiddharth, 2020-12-29:
Audio issues have been solved with the Dec release.

closed 2020-12-29 by shivasiddharth