shivasiddharth/GassistPi #1063: Audio (speaker & mic) broken after Gassistpi install with Matrix Creator

michaelbeljaars, 2020-10-16:
### Which branch of GassistPi are you using ?
```
    Master branch
    "SDK":060,
    "Features":036,
    "Revision":001
```
### Board and OS details:    
Open a terminal and run   
cat /proc/cpuinfo
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000002603431e
Model		: Raspberry Pi 3 Model B Rev 1.2
```

cat /etc/os-release  

```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"```
[... 324 lines cut ...]
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dsnoop.c:575:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: [MODEL_ID]
device_id: [DEVICE_ID]

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

```

shivasiddharth, 2020-10-16:
1. These are not errors, just warnings.
2. This project comes with Snowboy built-in any reason for compiling it seperately ?
3. Do :
`
sudo apt-get purge pulseaudio 
`
Reboot and give it a try.

michaelbeljaars, 2020-10-16:
Thanks for your quick response. I compiled Snowboy separately because it was suggested in the tutorial I followed (https://airbornesurfer.com/how-to-use-a-custom-wake-word-with-google-assistant/). I knew that Gassistpi interfaces with Snowboy, I was not aware that it is actually integrated into it. Not sure why the tutorial suggests to install it separately.

I tried your suggestion of purging pulseaudio. After rebooting and manually starting Gassistpi I get this (plus all the warnings from above):
```
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Fatal Python error: Aborted

Current thread 0x76fd6ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 103 in __init__
  File "/home/pi/GassistPi/src/main.py", line 1014 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
Aborted
```

shivasiddharth, 2020-10-16:
Can you share the output of:
`sudo nano ./.asoundrc`
`sudo nano /etc/asound.conf`

michaelbeljaars, 2020-10-16:
.asoundrc:
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

pcm.mic {
   type softvol
   slave {
     pcm "array"
   }
   control {
     name "MicArray Master"
        card "MATRIXIOSOUND"
   }
}

pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
    rate 16000
  }
}

pcm.array {
  type plug
  slave {
    pcm "hw:MATRIXIOSOUND"
  }
}

pcm.channel_1 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }
    bindings.0  0
}

pcm.channel_2 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  1
} 

pcm.channel_3 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  2
} 

pcm.channel_4 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  3
} 

pcm.channel_5 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  4
} 

pcm.channel_6 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  5
} 

pcm.channel_7 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  6
} 

pcm.channel_8 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  7
} 

pcm.all_channels {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings {
            0 0
            0 1
            0 2
            0 3
            0 4
            0 5
            0 6
            0 7
    }
} 

pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}
```
/etc/asound.conf
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

pcm.mic {
   type softvol
   slave {
     pcm "array"
   }
   control {
     name "MicArray Master"
        card "MATRIXIOSOUND"
   }
}

pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
    rate 16000
  }
}

pcm.array {
  type plug
  slave {
    pcm "hw:MATRIXIOSOUND"
  }
}

pcm.channel_1 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }
    bindings.0  0
}

pcm.channel_2 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  1
} 

pcm.channel_3 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  2
} 

pcm.channel_4 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  3
} 

pcm.channel_5 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  4
} 

pcm.channel_6 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  5
} 

pcm.channel_7 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  6
} 

pcm.channel_8 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  7
} 

pcm.all_channels {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings {
            0 0
            0 1
            0 2
            0 3
            0 4
            0 5
            0 6
            0 7
    }
} 
```

shivasiddharth, 2020-10-16:
Can you share the output of:
`aplay -l`
`arecord -l`
I need some more details about this matrix card. Does it have the audio output also or the Pi jack is the output ?

michaelbeljaars, 2020-10-16:
aplay -l:
```
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: MATRIXIOSOUND [MATRIXIO-SOUND], device 1: matrixio.pcm-out.0 snd-soc-dummy-dai-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

arecord -l:
```
**** List of CAPTURE Hardware Devices ****
card 2: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 3: MATRIXIOSOUND [MATRIXIO-SOUND], device 0: matrixio.mic.0 snd-soc-dummy-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

There is no audio output on the Matrix Creator. I am indeed using the audio jack on the pi.

shivasiddharth, 2020-10-16:
Open one terminal and run:
`arecord -f S16_LE -d 20 -r 16000 --channels 2 --device="hw:MATRIXIOSOUND" /tmp/test-mic.wav
`
Open second terminal and run:
`speaker-test -t wav`

Tell me what happens

michaelbeljaars, 2020-10-16:
Both commands executed simultaneously. I am currently working remotely on this pi, so I can do all the troubleshooting, but I currently cannot record anything or verify that there is sound coming from the speaker. 

```pi@MatrixVoice:~ $ arecord -f S16_LE -d 20 -r 16000 --channels 2 --device="hw:MATRIXIOSOUND" /tmp/test-mic.wav
Recording WAVE '/tmp/test-mic.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Stereo
arecord: pcm_read:2145: read error: Input/output error```

```pi@MatrixVoice:~ $ arecord -f S16_LE -d 20 -r 16000 --channels 2 --device="hw:MATRIXIOSOUND" /tmp/test2-mic.wav
arecord: main:828: audio open error: Device or resource busy```

shivasiddharth, 2020-10-16:
The audio config files need to be modified. I don't have a matrix hat to try. Actually, you need to open up an issue with Matrix hat git saying that the config files that they have provided does not permit the mics to be used simultaneously by multiple programs coz it's an issue with their config files. So try using the USB-DAC option in my audio setup guide. You will still be using matrix hat, but the config files will be USB-DAC's. After that, do the sudo nano for the audio configs that you did above and change 1,0 for mic to 2,0 in both the files.

shivasiddharth, 2020-10-16:
Another puzzling thing is, if the matrix hat does not have an output, why is it getting listed under aplay -l.

michaelbeljaars, 2020-10-16:
> Another puzzling thing is, if the matrix hat does not have an output, why is it getting listed under aplay -l.

Yeah, I saw that too. All I can tell you is that it does not have any audio out connectors. So the chip might support audio out but it may not be physically connected to a jack port

michaelbeljaars, 2020-10-16:
> The audio config files need to be modified. I don't have a matrix hat to try. Actually, you need to open up an issue with Matrix hat git saying that the config files that they have provided does not [...]

After running the script for USB-DAC and editing the audio config files, this is what manually running Gassistpi gives me:
```ALSA lib pcm_dsnoop.c:693:(snd_pcm_dsnoop_open) unable to open slave
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Fatal Python error: Aborted

Current thread 0x76f33ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 103 in __init__
  File "/home/pi/GassistPi/src/main.py", line 1014 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
```

Something to know is that the Matrix Creator is not just a hat with 1 microphone. It has an array of 8 microphones which can either be used individually or as an array (for improving sound quality or getting directional information).

shivasiddharth, 2020-10-16:
I hope you rebooted the Pi after using the USB-DAC config. In GassitPi two threads simultaneously try to use the mic, which the matrix hat is not permitting. Without matrix hat in my hand, troubleshooting this is going to take some time due to trial and errors. Here is a couple of things I would like you to try.
1. Revert to the original matrix hat audio config files.
2. Rename the existing GassitPi folder to something.
3. Clone my other Just Google Assistant branch.
4. Run that manually.
Tell me what happens.
This is close to midnight in my place. Will try and figure out a fix for the config file tomorrow. If possible share a link to the product page.

michaelbeljaars, 2020-10-16:
No worries, I appreciate your help greatly. There is no urgency to getting this sorted out. I will try your suggestions and get back to you.

michaelbeljaars, 2020-10-16:
> 4\. If possible share a link to the product page.
https://www.matrix.one/products/creator
https://matrix-io.github.io/matrix-documentation/matrix-creator/device-setup/

shivasiddharth, 2020-10-17:
1. Change the audio config files both .asoundrc and asound.conf to the following:
```
pcm.!default {
  type asym
  capture.pcm "all_channels"
  playback.pcm "speaker"
}

pcm.mic {
   type softvol
   slave {
     pcm "array"
   }
   control {
     name "MicArray Master"
        card "MATRIXIOSOUND"
   }
}

pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
    rate 16000
  }
}

pcm.array {
  type plug
  slave {
    pcm "hw:MATRIXIOSOUND"
  }
}

pcm.channel_1 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }
    bindings.0  0
}

pcm.channel_2 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  1
} 

pcm.channel_3 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  2
} 

pcm.channel_4 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  3
} 

pcm.channel_5 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  4
} 

pcm.channel_6 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  5
} 

pcm.channel_7 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  6
} 

pcm.channel_8 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  7
} 

pcm.all_channels {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings {
            0 0
            0 1
            0 2
            0 3
            0 4
            0 5
            0 6
            0 7
    }
} 
```
2. Reboot your Pi.
3. Make sure that both the config files are same.
4. Start the assistant manually (The Just Google Assistant branch).
5. Rename the GassistPi folder of the Just Google Assistant branch to something else and revert to the original GassistPi folder (rename it back to GassistPi)
6. Now try to manually start this assistant.

Please get back on this.

michaelbeljaars, 2020-10-17:
> I hope you rebooted the Pi after using the USB-DAC config. In GassitPi two threads simultaneously try to use the mic, which the matrix hat is not permitting. Without matrix hat in my hand, [...]

I performed all the steps above, this is the result when manually running the Just Google Assistant:
```
pi@MatrixVoice:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi_JGA/src/main.py --project_id [PROJECT_ID] --device_model_id [MODEL_ID]
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: [MODE_ID]
device_id: [DEVICE_ID]

ON_MUTED_CHANGED:
  {"is_muted": false}
Mic mute is set to: False
ON_START_FINISHED
ON_MEDIA_STATE_IDLE
```
Since I currently have no physical access to the pi, it is impossible to check whether it is now actually correctly using the mic and speaker. Because of a week of holidays, I will be able to check this earliest october 26th. If this did not solve the problem, I will try the suggestions you posted last.

On a related topic: the python library for the Google Assistant is deprecated (hence all the warnings). Do you plan changing the implementation of the Google Assistant? Is that even possible?

shivasiddharth, 2020-10-17:
So Just Google Assistant branch is working fine. See my last comment, please work on that and revert back.

> On a related topic: the python library for the Google Assistant is deprecated (hence all the warnings). Do you plan changing the implementation of the Google Assistant? Is that even possible?

The warnings are all from deprecated packages that are being used by other APIs for the different features. If you are into just the assistant and do not want any of the custom features like custom wakewording, you can stick to the Just Google Assistant branch. Or if you want just the custom wakewording and not the other features, you can use the Google Assistant Custom Wakeword  branch that I have got. There is so much of flexibility for the users.

Library will be supported until Google completely pulls the plug and refuse to run the code at all. By switching to Google Assistant Service you loose a lot of features, the trade off is too much, no alarms, timers etc. Also the audio helpers are still unpolished and its really crappy.

Do check my last comment for proceeding further with the troubleshooting process.

michaelbeljaars, 2020-10-19:
> 1. Change the audio config files both .asoundrc and asound.conf to the following:
> 
> ```
> pcm.!default {
>   type asym
>   capture.pcm "all_channels"
>   playback.pcm "speaker"
> }
> 
> pcm.mic {
>    type softvol
>    slave {
>      pcm "array"
>    }
>    control {
>      name "MicArray Master"
>         card "MATRIXIOSOUND"
>    }
> }
> 
> pcm.speaker {
>   type plug
>   slave {
>     pcm "hw:1,0"
>     rate 16000
>   }
> }
> 
> pcm.array {
>   type plug
>   slave {
>     pcm "hw:MATRIXIOSOUND"
>   }
> }
> 
> pcm.channel_1 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }
>     bindings.0  0
> }
> 
> pcm.channel_2 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  1
> } 
> 
> pcm.channel_3 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  2
> } 
> 
> pcm.channel_4 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  3
> } 
> 
> pcm.channel_5 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  4
> } 
> 
> pcm.channel_6 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  5
> } 
> 
> pcm.channel_7 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  6
> } 
> 
> pcm.channel_8 {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings.0  7
> } 
> 
> pcm.all_channels {
>     type dsnoop
>     ipc_key 234884
>     slave {
>         pcm "hw:2,0"
>         channels 8
>     }    
>     bindings {
>             0 0
>             0 1
>             0 2
>             0 3
>             0 4
>             0 5
>             0 6
>             0 7
>     }
> } 
> ```
> 
> 1. Reboot your Pi.
> 2. Make sure that both the config files are same.
> 3. Start the assistant manually (The Just Google Assistant branch).

I have performed these 3 steps, this is the result:
```
pi@MatrixVoice:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi_JGA/src/main.py --project_id [PROJECT_ID] --device_model_id [MODEL_ID]
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
ALSA lib pcm_dsnoop.c:693:(snd_pcm_dsnoop_open) unable to open slave
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Fatal Python error: Aborted

Current thread 0x76fb7ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 103 in __init__
  File "/home/pi/GassistPi_JGA/src/main.py", line 145 in main
  File "/home/pi/GassistPi_JGA/src/main.py", line 174 in <module>
Aborted
```

Also manually recording with arecord fails:
```
pi@MatrixVoice:~ $ arecord --duration 5 --rate 16000 --format S16_LE test.wav && aplay ./test.wav
Recording WAVE 'test.wav' : Signed 16 bit Little Endian, Rate 16000 Hz, Mono
Warning: rate is not accurate (requested = 16000Hz, got = 48000Hz)
         please, try the plug plugin (-Dplug:all_channels)
arecord: pcm_local.h:593: snd_pcm_channel_area_addr: Assertion `bitofs % 8 == 0' failed.
Aborted by signal Aborted...
```

michaelbeljaars, 2020-10-19:
It did get the Just Google Assistant to work with these files (as from Matrix):
```
pi@MatrixVoice:~ $ cat .asoundrc
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

pcm.mic {
   type softvol
   slave {
     pcm "array"
   }
   control {
     name "MicArray Master"
        card "MATRIXIOSOUND"
   }
}

pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}

pcm.array {
  type plug
  slave {
    pcm "hw:MATRIXIOSOUND"
  }
}

pcm.channel_1 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }
    bindings.0  0
}

pcm.channel_2 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  1
} 

pcm.channel_3 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  2
} 

pcm.channel_4 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  3
} 

pcm.channel_5 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  4
} 

pcm.channel_6 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  5
} 

pcm.channel_7 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  6
} 

pcm.channel_8 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  7
} 

pcm.all_channels {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings {
            0 0
            0 1
            0 2
            0 3
            0 4
            0 5
            0 6
            0 7
    }
} 

pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}
pi@MatrixVoice:~ $ 
```
```
pi@MatrixVoice:~ $ cat /etc/asound.conf
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}

pcm.mic {
   type softvol
   slave {
     pcm "array"
   }
   control {
     name "MicArray Master"
        card "MATRIXIOSOUND"
   }
}

pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}

pcm.array {
  type plug
  slave {
    pcm "hw:MATRIXIOSOUND"
  }
}

pcm.channel_1 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }
    bindings.0  0
}

pcm.channel_2 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  1
} 

pcm.channel_3 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  2
} 

pcm.channel_4 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  3
} 

pcm.channel_5 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  4
} 

pcm.channel_6 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  5
} 

pcm.channel_7 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  6
} 

pcm.channel_8 {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings.0  7
} 

pcm.all_channels {
    type dsnoop
    ipc_key 234884
    slave {
        pcm "hw:2,0"
        channels 8
    }    
    bindings {
            0 0
            0 1
            0 2
            0 3
            0 4
            0 5
            0 6
            0 7
    }
} 
pi@MatrixVoice:~ $ 
```

They only differ in the last part:
```
pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}
```
What I also discovered is that not only does Gassistpi (full functionality) not work (no response to hotwords), it also breaks something with respect to the mic, meaning that if I run Just Google Assistant after, it doesn't work either. A reboot is needed to restore functionality.

shivasiddharth, 2020-10-19:
Try this:
1. sudo rm /etc/asound.conf
2. Change ./.asoundrc to:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:2,0"
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
3. Also try changing the .asoundrc to:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:2,0"
        rate 16000
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
                rate 16000 
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
4. There should be no /etc/asound.conf file. Thats why I have asked you to delete it. You must reboot each time after changing the .asoundrc file for changes to take effect.

michaelbeljaars, 2020-10-19:
1. Removed /etc/asound.conf
2. Changed, rebooted and tested with JGA: speaker working, no mic (not responding to wakeword)
3. Changed, rebooted and tested with JGA: speaker not working (no sound), no mic (not responding to wakeword).

shivasiddharth, 2020-10-19:
Ok, we are making progress. 

Try this:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:3,0"
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

Set the .asoundrc to the above piece of code and follow the same set of procedures. Reboot, etc.

michaelbeljaars, 2020-10-19:
With this .asoundrc file: JGA works, Gassistpi master does not:

```
pi@MatrixVoice:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id [PROJECT_ID] --device_model_id [MODEL_ID]
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: [MODEL_ID]
device_id: [DEVICE_ID]

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE
```

And as before, running Gassistpi seem to break the mic access. Running JGA subsequently does not work (no response to wakeword).

shivasiddharth, 2020-10-19:
What you have posted is from the Master branch or JGA ? Can you post the output of the Master branch ?

michaelbeljaars, 2020-10-19:
> What you have posted is from the Master branch or JGA ? Can you post the output of the Master branch ?

This is the output from the master branch.

shivasiddharth, 2020-10-19:
So just to clarify, JGA is responding your queries and Master branch is not ?
Can you check the config.yaml in the master branch to see if custom wakeword and ok Google is enabled. The default custom wakeword is Alexa.

michaelbeljaars, 2020-10-19:
> So just to clarify, JGA is responding your queries and Master branch is not ?
> Can you check the config.yaml in the master branch to see if custom wakeword and ok Google is enabled. The default custom wakeword is Alexa.

That is correct. JGA is detects wakeword (OK, Google) and is responding, master does not detect any wakeword (I tried OK Google, Alexa, Snowboy) and is thus obviously not responding. See below the section of my config.yaml that configures the wakeword

```
#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'
```

shivasiddharth, 2020-10-19:
Can you disable the custom wakeword in the config.yaml file and then try. If even that does not work, only thing that I can suggest is trying from scratch again on a fresh copy of OS without doing the steps from the third party site that you carried out (snowboy compilation). The change to the matrix hat audio config still stands for the new installation.

michaelbeljaars, 2020-10-19:
Changed the config.yaml to:
```
#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Disabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'
```

I rebooted the pi and ran the master manually: no change.

michaelbeljaars, 2020-10-19:
> If even that does not work, only thing that I can suggest is trying from scratch again on a fresh copy of OS without doing the steps from the third party site that you carried out (snowboy compilation). The change to the matrix hat audio config still stands for the new installation.

Let's see if I have the steps right:

1. Fresh install of Raspbian
2. Install of Matrix Creator (First 2 steps of https://github.com/matrix-io/google-assistant-matrixio)
3. Clone Gassistpi and install (not running any audio configuration scripts)
4. Purge Pulseaudio
5. Remove /etc/asound.conf (if existent) and edit contents of .asoundrc to:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:3,0"
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

shivasiddharth, 2020-10-19:
Yes that's correct.

michaelbeljaars, 2020-10-20:
> 6. Manually run Gassistpi

First attempt:
```
pi@MatrixCreator:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id [PROJECT_ID] --device_model_id [MODEL_ID]
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "nu even niet"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_RENDER_RESPONSE:
  {"text": "Duidelijk 😌", "type": 0}

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```

So 3 times is a charm, it responds to all 3 wakewords (OK Google, Alexa, Snowboy),but I wonder what goes wrong the first 2 times. Moreover if I terminate the script after the 3rd time, I need to reboot to get the Gassistpi to work again.

michaelbeljaars, 2020-10-20:
I've been playing with it a bit more and it is very unpredictable, sometimes it starts after a few attempts, sometimes it doesn't.

shivasiddharth, 2020-10-20:
Can you install pulseaudio using:
`sudo apt-get install pulseaudio`

Now try it out, it should not crash.

michaelbeljaars, 2020-10-20:
> Can you install pulseaudio using:
> `sudo apt-get install pulseaudio`
> 
> Now try it out, it should not crash.

With reinstalling pulseaudio indeed it does not crash, but I have no speaker and no mic response.

shivasiddharth, 2020-10-20:
Can you remove this part and try:
```
pcm.output {
	type hw
	card 1
}

ctl.!default {
	type hw
	card 1
}
```
Alternately,
If the OS version is not critical, can you downgrade to Stretch from Buster?  This Crashing issue is due to the upgraded Alsa version in Buster that conflicts with SOX. By the end of this year I will switch from Snowboy to Picovoice that will save us from this misery.

michaelbeljaars, 2020-10-20:
Removing those lines in .asoundrc does not change anything. I don't have functioning sound or mic. Tests with arecord give me an empty wav file. Playing music does not lead to sound coming out of the speaker. It seems that audio does not function with pulseaudio installed, even though it does not generate an error. If the ALSA version is the problem, how does pulsaudio tie into this?

Is there an option to just downgrade alsa for now?

shivasiddharth, 2020-10-20:
Check this https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-635077153

michaelbeljaars, 2020-10-20:
So if I understand correctly, **with** pulseaudio as an intermediary, the problem with ALSA does not arise, but my problem with pulseaudio is that it can't port my audio devices for some reason. **Without** pulseaudio, I do get my audio devices directly (through ALSA) but somehow that leads to crashing of Gassistpi.

I am currently compiling and installing a previous version of ALSA. I will purge pulseaudio, revert to the .asoundrc from above, reboot and test again.

michaelbeljaars, 2020-10-20:
With compiled version of alsa and alsa-tools (1.1.3), same problem: without pulseaudio, Gassistpi crashes, with pulseaudio there is no audio playback and no recording.

shivasiddharth, 2020-10-20:
You need to perform all the steps in the comment that I highlighted previously. The user had identified issues with mpg123 and portaudio as well. 

Each DAC is different and unfortunately there is no universal solution. For AIY-Kit, pulseaudio fixed the alsa bug, for seeed studio Respeaker, pulseaudio was not required at all (if you see the respeaker setup, you will see a pulseaudio purge), just changes to the audio config files worked. Fortunately, I had samples of the DACs to work with and I was in a position to fix them fast. 

I just went through the Matrix community page and apparently, the Matrix creator does not like pulseaudio.

michaelbeljaars, 2020-10-21:
Aside from alsa and alsa-tools, I also already had compiled and installed the other suggested version of portaudio. Additionally today I have compiled and installed mpg123 and followed the rest of the instructions in the suggested comment, still getting the same crash:
```
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted
```

shivasiddharth, 2020-10-21:
> Do you have an idea what makes Gassistpi crash in the first place? Is it because it tries to access the mic from two different threads?

Not exactly. dsnoop plugin in the audio config file allows the mic to be used by two threads without any issue at all. The problem is between pyaudio-alsa-portaudio. There was some update to Alsa after 1.1.4 that started this conflict. Like I said before, the fix was easy for the other cards as I had them in my hand, I was able to test it out. That is also a reason why I dont have the Matrix card under the supported audio installations.

I have couple of suggestions for you.
1.  Revert back to Stretch.
2.  Let Matrix guys figure out the issue with their card (infact its not their audio config that is working but my modification).

I saw your post in Matrix forum. If your query is project specific, I doubt if you would ever get a reply. Instead I would suggest you to post that the default config is not using dsnoop plugin. There is also an existing issue https://community.matrix.one/t/microphone-array-separately/2266/10

michaelbeljaars, 2020-10-21:
I will give Stretch a try. I also tried recompiling portaudio with this patch (https://sources.debian.org/patches/portaudio19/19.6.0-1.1/944509-crash.patch/) which is supposed to solve this error: 
```
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted
```

Not sure if I did something wrong, but it did not seem to change anything. I started with a fresh build of portaudio, applied the patch, compiled, installed and rebooted, but the issue remains.

michaelbeljaars, 2020-10-21:
> I will give Stretch a try. I also tried recompiling portaudio with this patch (https://sources.debian.org/patches/portaudio19/19.6.0-1.1/944509-crash.patch/) which is supposed to solve this error:
> 
> ```
> python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
> Fatal Python error: Aborted
> ```
> 
> Not sure if I did something wrong, but it did not seem to change anything. I started with a fresh source code of portaudio, applied the patch, compiled, installed and rebooted, but the issue remains.

michaelbeljaars, 2020-10-21:
> 1. Fresh install of Raspbian
> 2. Install of Matrix Creator (First 2 steps of https://github.com/matrix-io/google-assistant-matrixio)
> 3. Clone Gassistpi and install (not running any audio configuration scripts)
> 4. Purge Pulseaudio
> 5. Remove /etc/asound.conf (if existent) and edit contents of .asoundrc to:
> 
> ```
> pcm.dsnooper {
>     type dsnoop
>     ipc_key 816357492
>     ipc_key_add_uid 0
>     ipc_perm 0666
>     slave {
>         pcm "hw:3,0"
>     }
> }
> 
> pcm.!default {
>         type asym
>         playback.pcm {
>                 type plug
>                 slave.pcm "hw:1,0"
>         }
>         capture.pcm {
>                 type plug
>                 slave.pcm "dsnooper"
>         }
> }
> ```
> 
> 6. Manually run Gassistpi

I went through all these steps using the latest release of Stretch (2019-04-08). I adjusted the settings in the .asoundrc file to reflect the hardware under Stretch:
```
pi@MatrixCreator:~ $ cat .asoundrc 
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:2,0"
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
```
pi@MatrixCreator:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 2: MATRIXIOSOUND [MATRIXIO-SOUND], device 1: matrixio.pcm-out.0 snd-soc-dummy-dai-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #1019 
```
```
pi@MatrixCreator:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Dummy [Dummy], device 0: Dummy PCM [Dummy PCM]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 2: MATRIXIOSOUND [MATRIXIO-SOUND], device 0: matrixio.mic.0 snd-soc-dummy-dai-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

I don't get any errors running Gassistpi, but also no response to any wakeword. Recording with arecord works fine, as well does playback with aplay.

shivasiddharth, 2020-10-21:
Did you reboot after changing the audio config?

michaelbeljaars, 2020-10-21:
> Did you reboot after changing the audio config?

I did.

shivasiddharth, 2020-10-21:
Can you manually start the assistant and share the terminal contents? Do you hear the startup chime ?

michaelbeljaars, 2020-10-21:
> Can you manually start the assistant and share the terminal contents? Do you hear the startup chime ?

Yes, I can start the assistant and I hear the chime. It seems to be listening, but it does not respond to any wakeword (OK Google, Alexa, Snowboy)
I am currently travelling. I will share the output when I am back.

michaelbeljaars, 2020-10-21:
> and share the terminal contents?

```
pi@MatrixCreator:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id smartassistant-5d680-matrixassistant-pxcxy0 --project_id smartassistant-5d680
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.5/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: smartassistant-5d680-matrixassistant-pxcxy0
device_id: 9E584D8D6D1C56C8686B1D985AEE5C44

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE
```

shivasiddharth, 2020-10-21:
Frankly, looks all ok to me. Can you try changing the card number of the mic to 1 in the .asoundrc to see if it make any difference. I am just giving a shot in the dark. To be honest I dont have a clue as to why the Matrix card is taking two card ids and is also appearing under playback devices. After changing the number see if arecord works.

michaelbeljaars, 2020-10-21:
> Frankly, looks all ok to me. Can you try changing the card number of the mic to 1 in the .asoundrc to see if it make any difference. I am just giving a shot in the dark. To be honest I dont have a [...]

Then it fails:
```
/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
ALSA lib pcm_dsnoop.c:673:(snd_pcm_dsnoop_open) unable to open slave
[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Fatal Python error: Aborted

Current thread 0x76f23640 (most recent call first):
  File "/home/pi/env/lib/python3.5/site-packages/google/assistant/library/assistant.py", line 103 in __init__
  File "/home/pi/GassistPi/src/main.py", line 1014 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
```

I am not too sure either about the multiple devices, but I think it has to do with the fact that it allows for addressing both the individual microphones as well as the array as a whole.

shivasiddharth, 2020-10-21:
Oh god. Revert back then to the old value. Can you double check if you have purged pulseaudio ? Can you see if JGA works ? In any case this is the best I can do. I have nothing more to suggest or do. Will see if I can reach out to some one from Matrix, the state of the forum is not very promising.

michaelbeljaars, 2020-10-21:
> Oh god. Revert back then to the old value.
Done

> Can you double check if you have purged pulseaudio ?
Checked, and is indeed not present on my system

> Can you see if JGA works ?
JGA works as long as I don't run the master first. Again, as with Buster, running the master seems to break the access to the mic. After running the master, I can't record anymore  with arecord even though it does not generate an error and also JGA does not respond anymore (but doesn't give an error).

shivasiddharth, 2020-10-22:
Ok, I may have an answer as to why the mic may not be working with pulseaudio. By default, pulseaudio could be using the first recording card which in your case is that dummy which obviously does not help. Try this:
Install pulse audio:
`sudo apt-get pulseaudio`
Install pulseaudio control
`sudo apt install pavucontrol`
Open the pulseaudio control, try choosing the recording and playback devices manually and now give it a try.

michaelbeljaars, 2020-10-22:
[image]

I used pavucontrol to disable the dummy profile. Pavucontrol shows activity on the mic prior to starting Gassistpi. Upon running Gassistpi it keeps showing activity (I was continuously making noise) until it comes to the part where it outputs the model id and device id (or maybe just prior to that). Then it freezes and becomes unresponsive.

michaelbeljaars, 2020-10-22:
JGA, pavucontrol recording window, noise level indicator bar showing activity
[image]

Gassistspi, pavucontrol recording window, no noise level indicator bar
[image]

shivasiddharth, 2020-10-22:
This is the best I could do. I have nothing more to suggest or do. I shall leave the issue open. The project will have an option to switch the custom wakeword engine to picovoice soon. Lets see if that works out. Thanks for your cooperation.

[12 more replies]

closed 2020-12-29 by shivasiddharth