shivasiddharth/GassistPi #1064: Unable to install - keep getting "Request contains an invalid argument"

shankargopal, 2020-10-18:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

Contents of version file:

{
  "Version":{
    "SDK":060,
    "Features":036,
    "Revision":001
  }
}

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 270.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : c03112
Serial          : 10000000322508ab
Model           : Raspberry Pi 4 Model B Rev 1.2

```
Paste your Board and OS details here.   
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

### Describe the bug:

I followed the installation instructions given in the Readme.  I've also installed GassistPI long ago (on a Raspberry 3 B+) and it worked brilliantly. Thank you for such a great project!

The installation appears to proceed normally.  But when I try to manually start the assistant, using the instructions under Ok-Google Hotword/Pi3/Pi2/Armv7 users (there seems to be no specific option for Raspberry 4 users), I get this:

```

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1051, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 1029, in main
    device_model_id, device_id, args.nickname)
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device INVALID_ARGUMENT (400): Request contains an invalid argument.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
Yes

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.

As above.

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
As above

shivasiddharth, 2020-10-18:
Please double check the command that you used to manually start. Can post the command here as it is. Do not remove any part of it.

shankargopal, 2020-10-18:
Thank you! This is the command I entered (model ID redacted):

```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id xxxxxx --project_id DoonPi3
```

shivasiddharth, 2020-10-18:
I need to see the thing as you post in the terminal.  That is why I asked you not to remove any part of it. Nobody can do anything with your model id. Anyways, you have missed the quotes. Try changing the order of parameters. First use the project_id and then the device_model_id

shankargopal, 2020-10-19:
Sorry, I thought the model ID might be sensitive.  I've retried the command with quotes and with the order reversed, like this:

```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'DoonPi3' --device_model_id 'doonpi3-doonpioct182020-9d6n18'
```

I still get the same error.

shivasiddharth, 2020-10-19:
Can you share the terminal contents as such after you execute this.

shivasiddharth, 2020-10-19:
Your ids seem off. Can you double check ? Coz the first part of model is usually the project id. The project id and the model id are in different cases here. Open the credentials json file that you downloaded in the home/pi directory that will have th correct ids please cross check with that

shankargopal, 2020-10-19:
I don't have a credentials.json file (is that the problem?) - only a client_secret json file, and that doesn't have either of these ids in it.  This is the full terminal output, including the command. I have changed the cases in the Model ID to match those of the Project Name, but one thing to note is that the Google Developers' Console shows the Model ID in all lower case (as above).  In any case the output appears to be the same as before.  Here it is:

```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'DoonPi3' --device_model_id 'DoonPi3-DoonPiOct182020-9d6n18'
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
  File "/home/pi/GassistPi/src/main.py", line 1051, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 1029, in main
    device_model_id, device_id, args.nickname)
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device INVALID_ARGUMENT (400): Request contains an invalid argument.
```

shivasiddharth, 2020-10-19:
The client_secret is what I meant. Did the authentication process go through properly ?

shankargopal, 2020-10-20:
Yes, it didn't report any errors as far as I could see.

If I was to wipe everything and try again, what all should I delete? Do you think that might be helpful?

shivasiddharth, 2020-10-29:
Do you have any updates ?

shankargopal, 2020-10-29:
Not yet, sorry. I assumed that I would have to reflash the SD card and start from scratch, and I haven't had time to do that yet as there is other data on the Pi that needs to be backed up.

Is there an easier way to restart from a clean start? I know that just deleting ~/Gassistpi is unlikely to be sufficient?

shivasiddharth, 2020-10-29:
Just deleting the folder will not work. You still will have the dependencies installed

shankargopal, 2020-10-31:
Ok then I'll try to reflash the entire system today.

On Thu, 29 Oct 2020 09:14:18 -0700
shivasiddharth <notifications@github.com> wrote:

> Just deleting the folder will not work. You still will have the
> dependencies installed 
>

shivasiddharth, 2020-11-26:
Do we have an update on this ?

shankargopal, 2020-11-26:
Not yet, very sorry. I simply haven't had the time to reflash the SD
card from scratch (I need to back up data, get a working SD card writer
etc.).  I hope to do so by this weekend.

On Wed, 25 Nov 2020 23:35:31 -0800
shivasiddharth <notifications@github.com> wrote:

> Do we have an update on this ?
>

shivasiddharth, 2020-12-29:
Closing this for inactivity. Please reopen once you have something to look at or discuss.

closed 2020-12-29 by shivasiddharth