shivasiddharth/GassistPi #1066: OSError: [Errno -9997] Invalid sample rate

beastnani, 2020-10-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Master branch 

### Board and OS details:
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   

processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 51.20
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 000000005c6bb4a5
Model		: Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster

### Describe the bug:
```
[... 341 lines cut ...]
ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, self->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, sampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
2020-10-27 13:15:13,387 : ERROR : [Errno -9997] Invalid sample rate
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1051, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 197, in __init__
    self.detector = snowboydecoder.HotwordDetector(models, sensitivity=self.sensitivity)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9997] Invalid sample rate

```

shivasiddharth, 2020-10-27:
share the output of:
```
aplay -l
arecord -l
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
```

beastnani, 2020-10-27:
*aplay -l
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 H
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

*arecord -l
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

*sudo nano ./.asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
*sudo nano /etc/asound.conf

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

shivasiddharth, 2020-10-27:
You have not provided full list with `aplay -l`

beastnani, 2020-10-27:
so what should I do? I am confused ?

shivasiddharth, 2020-10-27:
`aplay -l` will give all the audio output devices. You have a Pi 3 so at least the HDMI port should appear in the list. That is why I am saying that the list is incomplete.

beastnani, 2020-10-27:
I started form scratche again its working now, 
But 
Now i can't make assistant headless. When i reboot, it dosen't give any response or tone 
Can you help please

shivasiddharth, 2020-10-27:
share the output of `sudo systemctl status gassistpi.service`

beastnani, 2020-10-27:
pi@raspberrypi:~ $ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Tue 2020-10-27 22:54:32 IST; 10s ago
 Main PID: 6157 (python)
    Tasks: 1 (limit: 2065)
   CGroup: /system.slice/gassistpi.service
           └─6157 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packag
Oct 27 22:54:36 raspberrypi python[6157]:   options=None, file=DESCRIPTOR),
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packag
Oct 27 22:54:36 raspberrypi python[6157]:   index=0, containing_type=None, field
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packag
Oct 27 22:54:36 raspberrypi python[6157]:   serialized_end=727,
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packag
Oct 27 22:54:36 raspberrypi python[6157]:   options=None, file=DESCRIPTOR),
Oct 27 22:54:38 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packag
Oct 27 22:54:38 raspberrypi python[6157]:   from collections import Mapping, Con
lines 1-18/18 (END)...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-10-27 22:54:32 IST; 10s ago
 Main PID: 6157 (python)
    Tasks: 1 (limit: 2065)
   CGroup: /system.slice/gassistpi.service
           └─6157 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id raspi1-271118 --device_model_id raspi1-271118-raspi1-n4j65l

Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:201: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked desc
Oct 27 22:54:36 raspberrypi python[6157]:   options=None, file=DESCRIPTOR),
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked desc
Oct 27 22:54:36 raspberrypi python[6157]:   index=0, containing_type=None, fields=[]),
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descripto
Oct 27 22:54:36 raspberrypi python[6157]:   serialized_end=727,
Oct 27 22:54:36 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked desc
Oct 27 22:54:36 raspberrypi python[6157]:   options=None, file=DESCRIPTOR),
Oct 27 22:54:38 raspberrypi python[6157]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8
Oct 27 22:54:38 raspberrypi python[6157]:   from collections import Mapping, Container
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-18/18 (END)

shivasiddharth, 2020-10-27:
I see that your assistant is running without any issue. You should hear the startup chime and the assistant should work.

shivasiddharth, 2020-10-27:
SHare the output of `journalctl -u gassistpi.service `

beastnani, 2020-10-27:
But I am unable to hear any chime or assistant voice 

pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Tue 2020-10-27 23:00:37 IST, end at Tue 2020-10-27 23:34:08 IST
Oct 27 23:04:52 raspberrypi systemd[1]: Started Google Assistant-Voice.
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   serialized_pb=b'\n\x15google/api/http
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   serialized_end=121,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-package
lines 1-23...skipping...
-- Logs begin at Tue 2020-10-27 23:00:37 IST, end at Tue 2020-10-27 23:34:08 IST. --
Oct 27 23:04:52 raspberrypi systemd[1]: Started Google Assistant-Voice.
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:05:02 raspberrypi python[455]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Plea
Oct 27 23:05:02 raspberrypi python[455]:   serialized_end=121,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
[... cut ...]
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:97: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
Oct 27 23:05:02 raspberrypi python[455]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:115: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Oct 27 23:05:02 raspberrypi python[455]:   file=DESCRIPTOR,
lines 1-62

shivasiddharth, 2020-10-27:
Can you paste the output of` journalctl -u gassistpi.service` seems like some lines are truncated. You are saying that the manual start is working fine but not the service ?

beastnani, 2020-10-27:
Yes, the AutoStart of assistant on boot doesn't 

This is all the output in terminal :
pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Tue 2020-10-27 23:00:37 IST, end at Tue 2020-10-27 23:54:02 IST. --
Oct 27 23:43:00 raspberrypi systemd[1]: Started Google Assistant-Voice.
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:43:09 raspberrypi python[456]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Plea
Oct 27 23:43:09 raspberrypi python[456]:   serialized_end=121,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:297: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors is going to go away
Oct 27 23:43:09 raspberrypi python[456]:   fields=[],
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:301: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Ple
Oct 27 23:43:09 raspberrypi python[456]:   serialized_end=381,
Oct 27 23:43:09 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:328: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
[... cut ...]
Oct 27 23:43:10 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:97: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Oct 27 23:43:10 raspberrypi python[456]:   file=DESCRIPTOR,
Oct 27 23:43:10 raspberrypi python[456]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:115: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Oct 27 23:43:10 raspberrypi python[456]:   file=DESCRIPTOR,
lines 1-62

shivasiddharth, 2020-10-27:
Reboot the Pi allow it to stand for 2-3 mins dont try to manually start. Just dont do anything. Then share the output of:
```
sudo systemctl status gassistpi.service
journalctl -u gassistpi.service
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
aplay -l 
arecord -l
```
I still need to know as to why you have not given me the full `aplay -l` list

beastnani, 2020-10-27:
I did as you said, here are the outputs after 5 min

 pi@raspberrypi:~ $ sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Wed 2020-10-28 00:14:53 IST; 109ms ago
 Main PID: 22355 (python)
    Tasks: 1 (limit: 2065)
   CGroup: /system.slice/gassistpi.service
           └─22355 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --p

Oct 28 00:14:53 raspberrypi systemd[1]: Started Google Assistant-Voice.
lines 1-9/9 (END)

pi@raspberrypi:~ $ journalctl -u gassistpi.service
-- Logs begin at Tue 2020-10-27 23:00:37 IST, end at Wed 2020-10-28 00:16:38 IST. --
Oct 27 23:55:25 raspberrypi systemd[1]: Started Google Assistant-Voice.
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: Deprecat
Oct 27 23:55:34 raspberrypi python[446]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x1
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: Deprecat
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: Deprecat
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: Deprecat
Oct 27 23:55:34 raspberrypi python[446]:   serialized_end=121,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:297: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   fields=[],
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:301: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   serialized_end=381,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:328: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:346: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:358: Depreca
Oct 27 23:55:34 raspberrypi python[446]:   serialized_end=430,
Oct 27 23:55:34 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/annotations_pb2.py:42: D
Oct 27 23:55:34 raspberrypi python[446]:   google_dot_protobuf_dot_descriptor__pb2.DESCRIPTOR,
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/api/annotations_pb2.py:64: D
Oct 27 23:55:35 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:39: Deprec
Oct 27 23:55:35 raspberrypi python[446]:   dependencies=[google_dot_protobuf_dot_any__pb2.DESCRIPTOR],
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:66: Deprec
Oct 27 23:55:35 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:84: Deprec
Oct 27 23:55:35 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:102: Depre
Oct 27 23:55:35 raspberrypi python[446]:   file=DESCRIPTOR,
Oct 27 23:55:35 raspberrypi python[446]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:114: Depre
[... cut ...]
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Sound Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

shivasiddharth, 2020-10-27:
Do you have a USB DAC ? What is you mic and what is your audio out device ?

beastnani, 2020-10-27:
I have a pi3 and I am using usb mic as input and 3.5mm onboard jack for audio output

shivasiddharth, 2020-10-27:
Which version of buster are you using ? Date of buster ?

shivasiddharth, 2020-10-27:
Can you share a screenshot of the terminal with `aplay -l` command. The devices that you have and the output just does not add up.

beastnani, 2020-10-27:
It's NOOBS_v3_4_1 and I think I downloaded it in month of august around 15th

beastnani, 2020-10-27:
[image]

beastnani, 2020-10-27:
I think sound isn't a problem because I get audio output

shivasiddharth, 2020-10-27:
Are you using a lite version of Raspbian OS ? May be thats why HDMI is not coming up. The audio config that you are using is for Desktop version. Thats what is causing the no audio issue with the service.

Use the desktop version of Raspbian. Download the image from Pi website. Write the image without using Noobs. Give it a try and get back.

beastnani, 2020-10-28:
Should  I also install gassistpi?

shivasiddharth, 2020-10-28:
Download the image from here and see http://downloads.raspberrypi.org/raspbian_full/images/raspbian_full-2020-02-14/

beastnani, 2020-10-28:
which etcher sould i use?

shivasiddharth, 2020-10-28:
Etcher has nothing to do with kernel error. You need to format the SD card first before you write the image. I always use SD formatter for formatting and Win32 disk imager. Or else You can use the Pi imager software https://www.raspberrypi.org/downloads/ I am not going to do the hand holding with burning images here.

beastnani, 2020-10-28:
Should I send output of commands given by you before installing gassistpi or after?

beastnani, 2020-10-28:
[image]

beastnani, 2020-10-28:
here is output before installing gassistpi

shivasiddharth, 2020-10-28:
This is a Feb build. So there is no need to run the audio modifications I have given for post May builds. All I am interested is in the working of the assistant.

beastnani, 2020-10-28:
I have installed Assistant but, same as before works really good manually but doesn't auto start on boot

beastnani, 2020-10-28:
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 7/7
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 IEC958/HDMI [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: ALSA [bcm2835 ALSA], device 2: bcm2835 IEC958/HDMI1 [bcm2835 IEC958/HDMI1]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

sudo nano ./.asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

sudo nano /etc/asound.conf:

pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

shivasiddharth, 2020-10-28:
Do:
`sudo nano ./.asoundrc`

Remove everything in that and paste the following
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

After that do:
```
sudo rm /etc/asound.conf
sudo cp ./.asoundrc /etc/asound.conf
```

Restart the Pi and assistant should work.

If it does not work, then I you need to share the output of:
```
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
sudo nano /home/${USER}/GassistPi/systemd/gassistpi.service
```

beastnani, 2020-10-28:
Tank you soo much, It worked! finally!

shivasiddharth, 2020-10-28:
Closing this issue.

closed 2020-10-28 by shivasiddharth