shivasiddharth/GassistPi #1068: Unable to start assistant because of psutil python module.

Tech-User42, 2020-10-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```

```
Paste your Board and OS details here.   
```
processor       : 0
vendor_id       : GenuineIntel
cpu family      : 6
model           : 42
model name      : Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz
stepping        : 7
microcode       : 0x2f
cpu MHz         : 797.020
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 0
initial apicid  : 0
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 4589.61
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:

processor       : 1
vendor_id       : GenuineIntel
cpu family      : 6
model           : 42
model name      : Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz
stepping        : 7
microcode       : 0x2f
cpu MHz         : 798.196
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 1
cpu cores       : 2
apicid          : 2
initial apicid  : 2
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 4589.61
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:

processor       : 2
vendor_id       : GenuineIntel
cpu family      : 6
model           : 42
model name      : Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz
stepping        : 7
microcode       : 0x2f
cpu MHz         : 798.305
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 0
cpu cores       : 2
apicid          : 1
initial apicid  : 1
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 4589.61
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:

processor       : 3
vendor_id       : GenuineIntel
cpu family      : 6
model           : 42
model name      : Intel(R) Core(TM) i5-2410M CPU @ 2.30GHz
stepping        : 7
microcode       : 0x2f
cpu MHz         : 800.307
cache size      : 3072 KB
physical id     : 0
siblings        : 4
core id         : 1
cpu cores       : 2
apicid          : 3
initial apicid  : 3
fpu             : yes
fpu_exception   : yes
cpuid level     : 13
wp              : yes
flags           : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx est tm2 ssse3 cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer aes xsave avx lahf_lm epb pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
bugs            : cpu_meltdown spectre_v1 spectre_v2 spec_store_bypass l1tf mds swapgs itlb_multihit
bogomips        : 4589.61
clflush size    : 64
cache_alignment : 64
address sizes   : 36 bits physical, 48 bits virtual
power management:

NAME="Ubuntu"
VERSION="18.04.5 LTS (Bionic Beaver)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 18.04.5 LTS"
VERSION_ID="18.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=bionic
UBUNTU_CODENAME=bionic
### Describe the bug:
```
I got a missing python module "Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/pushbutton.py", line 36, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'"
but i installed that plugin in all python version and reboot 3 times
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes/No
NO
### 2. Is the issue related to assistant installation ?
Yes/No
NO
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No
YES AND NO 
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
 /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --device-model-id  BLABLA --project-id BLABLA
Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/pushbutton.py", line 36, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'
```

### Paste the contents of the terminal below
```
 /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --device-model-id  BLABLA --project-id BLABLA
Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/pushbutton.py", line 36, in <module>
    import psutil
ModuleNotFoundError: No module named 'psutil'
```
```

shivasiddharth, 2020-10-28:
You should be running the main.py and not pushbutton.py.

Tech-User42, 2020-10-28:
I got another error with main.py with module kodijson
``Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'``
But i have installed it.

shivasiddharth, 2020-10-28:
can you post the output of:
```
source /env/bin/activate
pip freeze
```

Tech-User42, 2020-10-28:
``eeze-bash: /env/bin/activate: Aucun fichier ou dossier de ce  type`
"pip freeze" return that :"
``DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
appdirs==1.4.4
asn1crypto==0.24.0
cachetools==3.1.1
certifi==2020.6.20
cffi==1.14.3
chardet==3.0.4
click==6.7
configparser==4.0.2
contextlib2==0.6.0.post1
cryptography==2.1.4
distlib==0.3.1
dnspython==1.15.0
enum34==1.1.10
faulthandler==2.4
filelock==3.0.12
futures==3.3.0
google-assistant-grpc==0.2.1
google-assistant-sdk==0.6.0
google-auth==1.22.1
google-auth-oauthlib==0.4.1
googleapis-common-protos==1.52.0
grpcio==1.33.1
gyp==0.1
idna==2.6
importlib-metadata==2.0.0
importlib-resources==3.1.1
ipaddress==1.0.17
keyring==10.6.0
keyrings.alt==3.0
kodi-json==1.0.0
monotonic==1.5
oauthlib==3.1.0
pathlib2==2.3.5
pipenv==2020.8.13
protobuf==3.13.0
psutil==5.7.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycairo==1.16.2
pycparser==2.20
pycrypto==2.6.1
pygobject==3.26.1
pyOpenSSL==19.1.0
pyxdg==0.25
requests==2.24.0
requests-oauthlib==1.3.0
rsa==4.5
scandir==1.10.0
SecretStorage==2.3.1
singledispatch==3.4.0.3
six==1.15.0
sounddevice==0.3.15
tenacity==4.12.0
typing==3.7.4.3
urllib3==1.25.11
virtualenv==20.0.35
virtualenv-clone==0.5.4
wxPython==3.0.2.0
wxPython-common==3.0.2.0
zipp==1.2.0``

shivasiddharth, 2020-10-28:
can you post the output of:

source /env/bin/activate
pip3 freeze

Tech-User42, 2020-10-28:
source /env/bin/activate return that is no file or folder `-bash: /env/bin/activate: Aucun fichier ou dossier de ce type`
pip3 freeze return that `DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
appdirs==1.4.4
asn1crypto==0.24.0
cachetools==3.1.1
certifi==2020.6.20
cffi==1.14.3
chardet==3.0.4
click==6.7
configparser==4.0.2
contextlib2==0.6.0.post1
cryptography==2.1.4
distlib==0.3.1
dnspython==1.15.0
enum34==1.1.10
faulthandler==2.4
filelock==3.0.12
futures==3.3.0
google-assistant-grpc==0.2.1
google-assistant-sdk==0.6.0
google-auth==1.22.1
google-auth-oauthlib==0.4.1
googleapis-common-protos==1.52.0
grpcio==1.33.1
gyp==0.1
idna==2.6
importlib-metadata==2.0.0
importlib-resources==3.1.1
ipaddress==1.0.17
keyring==10.6.0
keyrings.alt==3.0
kodi-json==1.0.0
monotonic==1.5
oauthlib==3.1.0
pathlib2==2.3.5
pipenv==2020.8.13
protobuf==3.13.0
psutil==5.7.3
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycairo==1.16.2
pycparser==2.20
pycrypto==2.6.1
pygobject==3.26.1
pyOpenSSL==19.1.0
pyxdg==0.25
requests==2.24.0
requests-oauthlib==1.3.0
rsa==4.5
scandir==1.10.0
SecretStorage==2.3.1
singledispatch==3.4.0.3
six==1.15.0
sounddevice==0.3.15
tenacity==4.12.0
typing==3.7.4.3
urllib3==1.25.11
virtualenv==20.0.35
virtualenv-clone==0.5.4
wxPython==3.0.2.0
wxPython-common==3.0.2.0
zipp==1.2.0`

shivasiddharth, 2020-10-28:
In the command that you use for manually starting can you try replacing python with python3

Tech-User42, 2020-10-28:
I got the same issue ` /home/${USER}/env/bin/python3 -u /home/${USER}/GassistPi/src/main.py --device-model-id  BLABLABLA --project-id BLABLABLA
Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 20, in <module>
    from kodijson import Kodi, PLAYER_VIDEO
ModuleNotFoundError: No module named 'kodijson'`

shivasiddharth, 2020-10-28:
Switch to the environment using:
`source /env/bin/activate`
then try manually running the assistant using:
`python3 -u .....`
No need to prefix /home/${USER}/env/bin/

Just try this.

Tech-User42, 2020-10-28:
It return me that 
```
tech-user@Toshiba:~$ source /env/bin/activate
-bash: /env/bin/activate: Aucun fichier ou dossier de ce type
```

shivasiddharth, 2020-10-28:
Sorry a typo can you share the output of:
```
source env/bin/activate
pip freeze
```

Tech-User42, 2020-10-28:
```
tech-user@Toshiba:~$ source env/bin/activate
(env) tech-user@Toshiba:~$ pip freeze
cachetools==4.1.1
certifi==2020.6.20
cffi==1.14.3
chardet==3.0.4
click==6.7
cryptography==3.2
enum34==1.1.10
futures==3.1.1
google-assistant-grpc==0.2.1
google-assistant-library==1.1.0
google-assistant-sdk==0.6.0
google-auth==1.22.1
google-auth-oauthlib==0.4.1
googleapis-common-protos==1.52.0
grpcio==1.33.1
idna==2.10
oauthlib==3.1.0
pathlib2==2.3.5
pkg-resources==0.0.0
protobuf==3.13.0
pyasn1==0.4.8
pyasn1-modules==0.2.8
pycparser==2.20
pyOpenSSL==19.1.0
python-vlc==3.0.11115
requests==2.24.0
requests-oauthlib==1.3.0
rsa==4.6
six==1.15.0
sounddevice==0.3.15
tenacity==4.12.0
urllib3==1.25.11
youtube-dl==2020.9.20
(env) tech-user@Toshiba:~$
```

shivasiddharth, 2020-10-28:
most of the packages have not been installed. try:
```
source env/bin/activate
pip install -r /home/${USER}/GassistPi/Requirements/GassistPi-pip-requirements.txt
```

Tech-User42, 2020-10-28:
I got and error and i try to restart with how you tell me to do and failed.
```
(env) tech-user@Toshiba:~$ pip install -r /home/${USER}/GassistPi/Requirements/GassistPi-pip-requirements.txt
Collecting git+https://github.com/plamere/spotipy.git (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 17))
  Cloning https://github.com/plamere/spotipy.git to /tmp/pip-req-build-n621k6dy
Collecting git+https://github.com/shivasiddharth/py-googletrans.git (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 23))
  Cloning https://github.com/shivasiddharth/py-googletrans.git to /tmp/pip-req-build-ry1_4wf8
Collecting pyaudio>=0.2.11
  Downloading PyAudio-0.2.11.tar.gz (37 kB)
Collecting aftership>=0.2
  Downloading aftership-1.1.0-py3-none-any.whl (8.3 kB)
Collecting feedparser>=5.2.1
  Downloading feedparser-6.0.2-py3-none-any.whl (80 kB)
     |████████████████████████████████| 80 kB 1.3 MB/s
Collecting kodi-json>=1.0.0
  Downloading kodi_json-1.0.0-py2.py3-none-any.whl (3.8 kB)
Collecting gmusicapi>=11.1.1
  Downloading gmusicapi-13.0.0.tar.gz (173 kB)
     |████████████████████████████████| 173 kB 1.5 MB/s
Collecting gTTS==2.0.2
  Downloading gTTS-2.0.2.tar.gz (24 kB)
Collecting gTTS-token==1.1.3
  Downloading gTTS-token-1.1.3.tar.gz (3.4 kB)
Requirement already satisfied: youtube_dl>=2018.11.23 in ./env/lib/python3.6/site-packages (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 8)) (2020.9.20)
Collecting PyChromecast>=2.3.0
  Downloading PyChromecast-7.5.1-py2.py3-none-any.whl (56 kB)
     |████████████████████████████████| 56 kB 1.7 MB/s
Collecting pafy>=0.5.4
  Downloading pafy-0.5.5-py2.py3-none-any.whl (35 kB)
Processing ./.cache/pip/wheels/42/32/da/8b12fd6b138c733efd03cfde6c6c8191a32842f9e82aa45fbf/psutil-5.7.3-cp36-cp36m-linux_x86_64.whl
Requirement already satisfied: urllib3>=1.24.1 in ./env/lib/python3.6/site-packages (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 12)) (1.25.11)
Collecting google-api-python-client>=1.7.5
  Downloading google_api_python_client-1.12.5-py2.py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 1.6 MB/s
Requirement already satisfied: requests>=2.20.1 in ./env/lib/python3.6/site-packages (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 14)) (2.24.0)
Collecting pushbullet.py>=0.11.0
  Downloading pushbullet.py-0.12.0-py2.py3-none-any.whl (10 kB)
Requirement already satisfied: python-vlc>=3.0.4106 in ./env/lib/python3.6/site-packages (from -r /home/tech-user/GassistPi/Requirements/GassistPi-pip-requirements.txt (line 16)) (3.0.11115)
Collecting pyyaml==3.13
  Downloading PyYAML-3.13.tar.gz (270 kB)
     |████████████████████████████████| 270 kB 2.0 MB/s
Collecting spidev>=3.2
  Downloading spidev-3.5.tar.gz (10 kB)
Collecting gpiozero>=1.4.1
  Downloading gpiozero-1.5.1-py2.py3-none-any.whl (138 kB)
     |████████████████████████████████| 138 kB 1.7 MB/s
Collecting numpy>=1.15.4
  Downloading numpy-1.19.2-cp36-cp36m-manylinux2010_x86_64.whl (14.5 MB)
     |████████████████████████████████| 14.5 MB 1.9 MB/s
Collecting pycryptodomex==3.7.2
  Downloading pycryptodomex-3.7.2-cp36-cp36m-manylinux1_x86_64.whl (7.5 MB)
     |████████████████████████████████| 7.5 MB 1.9 MB/s
Collecting paho-mqtt==1.4.0
  Downloading paho-mqtt-1.4.0.tar.gz (88 kB)
     |████████████████████████████████| 88 kB 1.9 MB/s
Collecting google-cloud-texttospeech==0.3.0
  Downloading google_cloud_texttospeech-0.3.0-py2.py3-none-any.whl (40 kB)
     |████████████████████████████████| 40 kB 1.5 MB/s
Collecting google-cloud-speech==0.36.0
  Downloading google_cloud_speech-0.36.0-py2.py3-none-any.whl (54 kB)
     |████████████████████████████████| 54 kB 1.6 MB/s
Collecting adafruit-io==2.1
  Downloading adafruit-io-2.1.tar.gz (43 kB)
     |████████████████████████████████| 43 kB 1.3 MB/s
Collecting pywemo==0.4.39
  Downloading pywemo-0.4.39-py2.py3-none-any.whl (47 kB)
     |████████████████████████████████| 47 kB 1.5 MB/s
Collecting cryptography==2.4.2
  Downloading cryptography-2.4.2-cp34-abi3-manylinux1_x86_64.whl (2.1 MB)
     |████████████████████████████████| 2.1 MB 2.2 MB/s
[... 114 lines cut ...]
    Running setup.py install for pyyaml ... done
    Running setup.py install for spidev ... done
    Running setup.py install for paho-mqtt ... done
    Running setup.py install for adafruit-io ... done
  Attempting uninstall: cryptography
    Found existing installation: cryptography 3.2
    Uninstalling cryptography-3.2:
      Successfully uninstalled cryptography-3.2
    Running setup.py install for spotipy ... done
    Running setup.py install for googletrans ... done
ERROR: After October 2020 you may experience errors when installing or updating packages. This is because pip will change the way that it resolves dependency conflicts.

We recommend you use --use-feature=2020-resolver to test your packages with the new resolver before it becomes the default.

pyopenssl 19.1.0 requires cryptography>=2.8, but you'll have cryptography 2.4.2 which is incompatible.
Successfully installed MechanicalSoup-0.12.0 PyChromecast-7.5.1 adafruit-io-2.1 aftership-1.1.0 appdirs-1.4.4 asn1crypto-1.4.0 beautifulsoup4-4.9.3 bs4-0.0.1 casttube-0.2.1 colorzero-1.1 coverage-5.3 coveralls-2.1.2 cryptography-2.4.2 decorator-4.4.2 docopt-0.6.2 feedparser-6.0.2 gTTS-2.0.2 gTTS-token-1.1.3 gmusicapi-13.0.0 google-api-core-1.23.0 google-api-pytho [... cut ...] e-2.0.1 spidev-3.5 spotipy-2.16.1 uritemplate-3.0.1 validictory-1.1.2 websocket-client-0.57.0 zeroconf-0.28.6
(env) tech-user@Toshiba:~$ python3 -u /home/${USER}/GassistPi/src/main.py
Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 36, in <module>
    import snowboydecoder
  File "/home/tech-user/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
ModuleNotFoundError: No module named 'snowboydetect'
(env) tech-user@Toshiba:~$
```

shivasiddharth, 2020-10-28:
The install script should have installed everything. Seems like the installation was broken. Did it stop in between ? Can you try running the installer again as given in README.
If you dont need all the features of the master branch, you can use the Just Google Assistant branch.

Tech-User42, 2020-10-28:
I try that and i return to you

Tech-User42, 2020-10-28:
I have a realaease upgrade of ubuntu the 20.04.1 lts do you think i should do it ?

Tech-User42, 2020-10-28:
When i run the script i get that : 
`tech-user@Toshiba:~$ sudo  ./GassistPi/scripts/gassist-installer.sh
This script must run as tech-user, trying to change user...

Enter the your full credential file name including the path and .json extension: /home/tech-user/credentials.json

Enter the your Google Cloud Console Project-Id: home-buntu-blabla

Enter the modelid that was generated in the actions console: 3898blablabla

Atteint :1 http://fr.archive.ubuntu.com/ubuntu bionic InRelease
Atteint :2 http://ppa.launchpad.net/graphics-drivers/ppa/ubuntu bionic InRelease
Réception de :3 http://security.ubuntu.com/ubuntu bionic-security InRelease [88,7 kB]
Atteint :4 http://fr.archive.ubuntu.com/ubuntu bionic-updates InRelease
Atteint :5 http://fr.archive.ubuntu.com/ubuntu bionic-backports InRelease
Atteint :6 http://ppa.launchpad.net/lutris-team/lutris/ubuntu bionic InRelease
Ign :7 http://ppa.launchpad.net/wine/wine-builds/ubuntu bionic InRelease
Err :8 http://ppa.launchpad.net/wine/wine-builds/ubuntu bionic Release
  404  Not Found [IP : 91.189.95.83 80]
Lecture des listes de paquets... Fait
E: Le dépôt http://ppa.launchpad.net/wine/wine-builds/ubuntu bionic Release n'a pas de fichier Release.
N: Les mises à jour depuis un tel dépôt ne peuvent s'effectuer de manière sécurisée, et sont donc désactivées par défaut.
N: Voir les pages de manuel d'apt-secure(8) pour la création des dépôts et les détails de configuration d'un utilisateur.
tech-user@Toshiba:~$`

shivasiddharth, 2020-10-28:
404 means that the network is down. Thats all i can say as I dont know the other language.

Tech-User42, 2020-10-28:
So i start manually and i hear a sound followed witgh a great "Hello !" but i got that :
```
tech-user@Toshiba:~$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device-model-id  home-buntu-da303 --project-id 389824728727
/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: home-buntu-da303
device_id: 83A6F5A05E5B43169E8662AFDCFA0F52

Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 174, in <module>
    main()
  File "/home/tech-user/GassistPi/src/main.py", line 156, in main
    device_model_id, device_id, args.nickname)
  File "/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device INVALID_ARGUMENT (400): Request contains an invalid argument.
tech-user@Toshiba:~$
```

shivasiddharth, 2020-10-28:
The order of the arguments should be changed. first it should be project id and second should be the model id

Tech-User42, 2020-10-28:
I think i reach the limit for today ;) 
```
tech-user@Toshiba:~$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device-model-id  389824728727  --project-id home-buntu-da303
/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: 389824728727
device_id: 83A6F5A05E5B43169E8662AFDCFA0F52

Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 174, in <module>
    main()
  File "/home/tech-user/GassistPi/src/main.py", line 156, in main
    device_model_id, device_id, args.nickname)
  File "/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device RESOURCE_EXHAUSTED (429): Quota exceeded for quota metric 'embedded-assistant-prod/converse_requests' and limit 'ConverseRequestsPerDayPerProject' of service 'embeddedassistant.googleapis.com' for consumer 'project_number:389824728727'.
```
I think my probleme is solved i try tommorow to test and i closed this tomorrow thanks for the help !

Tech-User42, 2020-10-29:
I try to restart the AI this morning and got that : 
```
tech-user@Toshiba:~$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device-model-id  389824728727  --project-id home-buntu-da303
/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: 389824728727
device_id: 83A6F5A05E5B43169E8662AFDCFA0F52

Registering...Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 174, in <module>
    main()
  File "/home/tech-user/GassistPi/src/main.py", line 156, in main
    device_model_id, device_id, args.nickname)
  File "/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/device_helpers.py", line 85, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device NOT_FOUND (404): Requested entity was not found.
tech-user@Toshiba:~$ 
```

shivasiddharth, 2020-10-29:
Please check the order of the argument. Project id should come first and device id afterwards.

Tech-User42, 2020-10-29:
I didnt understand what goes wrong 

```
tech-user@Toshiba:~$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project-id mon--home --device-model-id mon--home-mon-google-home-qxc3cq  
/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: mon--home-mon-google-home-qxc3cq
device_id: 83A6F5A05E5B43169E8662AFDCFA0F52

Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 174, in <module>
    main()
  File "/home/tech-user/GassistPi/src/main.py", line 156, in main
    device_model_id, device_id, args.nickname)
  File "/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device INVALID_ARGUMENT (400): Request contains an invalid argument.
tech-user@Toshiba:~$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --device-model-id mon--home-mon-google-home-qxc3cq --project-id mon--home 
/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: mon--home-mon-google-home-qxc3cq
device_id: 83A6F5A05E5B43169E8662AFDCFA0F52

Traceback (most recent call last):
  File "/home/tech-user/GassistPi/src/main.py", line 174, in <module>
    main()
  File "/home/tech-user/GassistPi/src/main.py", line 156, in main
    device_model_id, device_id, args.nickname)
  File "/home/tech-user/env/lib/python3.6/site-packages/google/assistant/library/device_helpers.py", line 88, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device INVALID_ARGUMENT (400): Request contains an invalid argument.
tech-user@Toshiba:~$ 

```

shivasiddharth, 2020-10-29:
Please check the arguments. The project id and model id should have underscores and not hyphens.

shivasiddharth, 2020-12-01:
Closing this issue for inactivity.

closed 2020-12-01 by shivasiddharth