shivasiddharth/GassistPi #1069: Issue on systemed service startup

jlaitila, 2020-10-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@magicmirroros:~ $ journalctl -u gassistpi.service 
-- Logs begin at Wed 2020-10-28 07:53:48 GMT, end at Wed 2020-10-28 16:28:11 GMT. --
Oct 28 07:54:01 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros python[518]: Traceback (most recent call last):
Oct 28 07:54:03 magicmirroros python[518]:   File "/home/pi/GassistPi/src/main.py", line 18, in <module>
Oct 28 07:54:03 magicmirroros python[518]:     import faulthandler
Oct 28 07:54:03 magicmirroros python[518]: ImportError: No module named faulthandler
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 1.
Oct 28 07:54:03 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros python[553]: Traceback (most recent call last):
Oct 28 07:54:03 magicmirroros python[553]:   File "/home/pi/GassistPi/src/main.py", line 18, in <module>
Oct 28 07:54:03 magicmirroros python[553]:     import faulthandler
Oct 28 07:54:03 magicmirroros python[553]: ImportError: No module named faulthandler
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 2.
Oct 28 07:54:03 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros python[554]: Traceback (most recent call last):
Oct 28 07:54:03 magicmirroros python[554]:   File "/home/pi/GassistPi/src/main.py", line 18, in <module>
Oct 28 07:54:03 magicmirroros python[554]:     import faulthandler
Oct 28 07:54:03 magicmirroros python[554]: ImportError: No module named faulthandler
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 07:54:03 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 3.
Oct 28 07:54:03 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 07:54:03 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 07:54:04 magicmirroros python[560]: Traceback (most recent call last):
Oct 28 07:54:04 magicmirroros python[560]:   File "/home/pi/GassistPi/src/main.py", line 18, in <module>
Oct 28 07:54:04 magicmirroros python[560]:     import faulthandler
Oct 28 07:54:04 magicmirroros python[560]: ImportError: No module named faulthandler
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 4.
Oct 28 07:54:04 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 07:54:04 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 07:54:04 magicmirroros python[563]: Traceback (most recent call last):
Oct 28 07:54:04 magicmirroros python[563]:   File "/home/pi/GassistPi/src/main.py", line 18, in <module>
Oct 28 07:54:04 magicmirroros python[563]:     import faulthandler
Oct 28 07:54:04 magicmirroros python[563]: ImportError: No module named faulthandler
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Oct 28 07:54:04 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Start request repeated too quickly.
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
[... 383 lines cut ...]
Oct 28 16:38:10 magicmirroros python[8211]: ImportError: No module named faulthandler
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Oct 28 16:38:10 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Start request repeated too quickly.
Oct 28 16:38:10 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 16:38:10 magicmirroros systemd[1]: Failed to start Google Assistant-Voice.
lines 244-319/319 (END)

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

jlaitila, 2020-10-28:
pi@magicmirroros:~/GassistPi/systemd $ cat gassistpi.service 
[Unit]
Description=Google Assistant-Voice
Wants=network-online.target
After=network-online.target

[Service]
Environment=VIRTUAL_ENV=/home/pi/env
Environment=PATH=/home/pi/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin
ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'magicmirror-XXXXe' --device_model_id 'magicmirror-XXXXe-mymagicmirror-uotm21'

WorkingDirectory=/home/pi/
StandardOutput=inherit
StandardError=inherit
Restart=always
User=pi

[Install]
WantedBy=multi-user.target
pi@magicmirroros:~/GassistPi/systemd $

shivasiddharth, 2020-10-29:
1. Board and OS details have not been provided.
2. Closing this issue for not sticking to the template.
3. You can re-open the issue once the required details have been filled.

closed 2020-10-29 by shivasiddharth

jlaitila, 2020-10-29:
How can I re-open issue?

pi@magicmirroros:~ $ nano config.js ^C
pi@magicmirroros:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000005f826a19
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@magicmirroros:~ $

jlaitila, 2020-10-29:
pi@magicmirroros:~ $ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@magicmirroros:~ $