shivasiddharth/GassistPi #1070: Issue on systemed service startup

jlaitila, 2020-10-29:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
pi@magicmirroros:~ $ nano config.js ^C
pi@magicmirroros:~ $ cat /proc/cpuinfo
processor : 0
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 76.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 1
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 76.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 2
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 76.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 3
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 76.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : BCM2835
Revision : a02082
Serial : 000000005f826a19
Model : Raspberry Pi 3 Model B Rev 1.2
pi@magicmirroros:~ $

cat /etc/os-release  
```
pi@magicmirroros:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
[... 138 lines cut ...]
Oct 28 07:54:04 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 07:54:04 magicmirroros systemd[1]: Failed to start Google Assistant-Voice.
Oct 28 08:01:22 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 08:01:23 magicmirroros python[1154]: Traceback (most recent call last):
Oct 28 08:01:23 magicmirroros python[1154]: File "/home/pi/GassistPi/src/main.py", line 18, in 
Oct 28 08:01:23 magicmirroros python[1154]: import faulthandler
Oct 28 08:01:23 magicmirroros python[1154]: ImportError: No module named faulthandler
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 1.
Oct 28 08:01:23 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 08:01:23 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 08:01:23 magicmirroros python[1159]: Traceback (most recent call last):
Oct 28 08:01:23 magicmirroros python[1159]: File "/home/pi/GassistPi/src/main.py", line 18, in 
Oct 28 08:01:23 magicmirroros python[1159]: import faulthandler
Oct 28 08:01:23 magicmirroros python[1159]: ImportError: No module named faulthandler
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Main process exited, code=exited, status=1/FAILURE
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Oct 28 08:01:23 magicmirroros systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 2.
Oct 28 08:01:23 magicmirroros systemd[1]: Stopped Google Assistant-Voice.
Oct 28 08:01:23 magicmirroros systemd[1]: Started Google Assistant-Voice.
Oct 28 08:01:23 magicmirroros python[1160]: Traceback (most recent call last):
lines 1-76

shivasiddharth, 2020-10-29:
Couple of questions
1. Is this a modifiied version of Raspbian ?
2. Is this a lite version or full version ?

jlaitila, 2020-10-29:
hmm..I suppose so. Sorry but I'm not so programming oriented. I still try very much and hope I don't bother too much.

This is Magic Mirror image which might be running on some container. I shall try to install raspiOS and manually add everything, if this seems too complicated.

http://unofficialpi.org/Distros/MagicMirrorOS/

jlaitila, 2020-10-30:
I installed Buster release yesterday. It seems to me that Gassip is running, but I can't get GA take any response?

pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ 

pi@raspberrypi:~ $ **sudo systemctl status gassistpi.service** 
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2020-10-30 18:35:16 EET; 6s ago
 Main PID: 3924 (python)
    Tasks: 1 (limit: 2063)
   CGroup: /system.slice/gassistpi.service
           └─3924 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id magicmirror-7264e --device_model_id magicmirror-7264e-mymagicmirror-uotm21

Oct 30 18:35:19 raspberrypi python[3924]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:201: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get
Oct 30 18:35:19 raspberrypi python[3924]:   options=None, file=DESCRIPTOR),
Oct 30 18:35:19 raspberrypi python[3924]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get
Oct 30 18:35:19 raspberrypi python[3924]:   index=0, containing_type=None, fields=[]),
Oct 30 18:35:19 raspberrypi python[3924]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find
Oct 30 18:35:19 raspberrypi python[3924]:   serialized_end=727,
Oct 30 18:35:19 raspberrypi python[3924]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get
Oct 30 18:35:19 raspberrypi python[3924]:   options=None, file=DESCRIPTOR),
Oct 30 18:35:20 raspberrypi python[3924]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working
Oct 30 18:35:20 raspberrypi python[3924]:   from collections import Mapping, Container
lines 1-18/18 (END)

**Running this on cmd line make it say hi at the end**
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id magicmirror-7264e --device_model_id magicmirror-7264e-mymagicmirror-uotm21

pi@raspberrypi:~ $ **aplay -L**
null
    Discard all samples (playback) or generate zero samples (capture)
jack
    JACK Audio Connection Kit
pulse
    PulseAudio Sound Server
default
    Playback/recording through the PulseAudio sound server
output
sysdefault:CARD=b1
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Default Audio Device
dmix:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample mixing device
dsnoop:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct sample snooping device
hw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Direct hardware device without any conversions
plughw:CARD=b1,DEV=0
    bcm2835 HDMI 1, bcm2835 HDMI 1
    Hardware device with all software conversions
usbstream:CARD=b1
    bcm2835 HDMI 1
    USB Stream Output
sysdefault:CARD=USB
    Razer Kraken USB, USB Audio
    Default Audio Device
front:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    Front speakers
surround21:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    2.1 Surround output to Front and Subwoofer speakers
surround40:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    4.0 Surround output to Front and Rear speakers
surround41:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    4.1 Surround output to Front, Rear and Subwoofer speakers
surround50:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    5.0 Surround output to Front, Center and Rear speakers
surround51:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    5.1 Surround output to Front, Center, Rear and Subwoofer speakers
surround71:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    7.1 Surround output to Front, Center, Side, Rear and Woofer speakers
iec958:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    IEC958 (S/PDIF) Digital Audio Output
dmix:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    Direct sample mixing device
dsnoop:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    Direct sample snooping device
hw:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    Direct hardware device without any conversions
plughw:CARD=USB,DEV=0
    Razer Kraken USB, USB Audio
    Hardware device with all software conversions
usbstream:CARD=USB
    Razer Kraken USB
    USB Stream Output
sysdefault:CARD=Headphones
    bcm2835 Headphones, bcm2835 Headphones
    Default Audio Device
dmix:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample mixing device
dsnoop:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct sample snooping device
hw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Direct hardware device without any conversions
plughw:CARD=Headphones,DEV=0
    bcm2835 Headphones, bcm2835 Headphones
    Hardware device with all software conversions
usbstream:CARD=Headphones
    bcm2835 Headphones
    USB Stream Output
pi@raspberrypi:~ $ 

**/etc/asound.conf is missing instead around.conf.bak is locate**d

Thanks!

jlaitila, 2020-10-30:
Trying to provide info as much if it helps

pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: USB [Razer Kraken USB], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $

jlaitila, 2020-10-30:
pi@raspberrypi:~ $ **/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'magicmirror-7264e-mymagicmirror-uotm21' --project_id 'magicmirror-7264e'**

/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  serialized_end=121,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:245: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
/home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
  file=DESCRIPTOR,
[... cut ...]
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: magicmirror-7264e-mymagicmirror-uotm21
device_id: 0713C8AF20XXXXXXXE9DDA1032361E

Fatal Python error: Segmentation fault

Thread 0x76f5fad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1043 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
Segmentation fault

jlaitila, 2020-10-30:
After that error I hear som plinging and HI!

shivasiddharth, 2020-10-31:
It means that the assistant is working.

jlaitila, 2020-10-31:
Okay but I can get no response from the assistant. maybe I have something wrong in my sound/mic setttings. Are all those python warnings etc. normal above? 

/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-10-30 21:10:53,656 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
INFO:gmusicapi.Mobileclient1:failed to authenticate
USB-MIC-JACK

shivasiddharth, 2020-10-31:
Thats normal. The first command should be "Hey google, set volume to maximum". Do you get a "pling" tone when you say "OK, Google" ? If not go through the last couple of closed issues. I have discussed the audio thing there.

closed 2020-10-31 by shivasiddharth