shivasiddharth/GassistPi #1071: Fatal Python error: Segmentation fault

Welsyntoffie, 2020-11-01:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 00000000385c2fe1
Model		: Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
[... 382 lines cut ...]

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": ""}

Fatal Python error: Segmentation fault

Thread 0x1e78e460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 480 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x1f790460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 253 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f48ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1043 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
Segmentation fault
```

shivasiddharth, 2020-11-01:
Can you give a couple of commands and post the output with domoticz enabled. Like request time etc.

Welsyntoffie, 2020-11-01:
It does not even respond, same error.

shivasiddharth, 2020-11-01:
Can you attach your config file here. Do not mask or change anything

Welsyntoffie, 2020-11-01:
```#Config file for declaring custom actions'parameters. The parameters have been listed here in alphabetical order (A-Z).

#Adafruit_IO. Sending commands from Genuine Google Home/Assistant Devices to control Domoticz Tasmota etc.
ADAFRUIT_IO:
  ADAFRUIT_IO_CONTROL: 'Disabled'
  ADAFRUIT_IO_USERNAME: 'ENTER YOUR ADAFRUIT_IO USERNAME HERE'
  ADAFRUIT_IO_KEY: 'ENTER YOUR ADAFRUIT_IO KEY HERE'
  FEEDNAME: 'ENTER YOUR ADAFRUIT_IO FEEDNAME THAT YOU WANT TO MONITOR HERE'

#Custom conversation. Add your queries and expected assistant response. Do not add symbols or special charecters.
Conversation:
  Conversation_Control: 'Enabled'
  question:
    1:
      - 'Who are you'
    2:
      - 'Do you have a family'
  answer:
    1:
      - 'I am not the one you expect to be'
      - 'Who do you think I am'
      - 'What bothers you'
    2:
      - 'Nope i am forever alone'
      - 'The whole world is my family'

#Send sms with clickatell
Clickatell:
  Clickatell_Control: 'Disabled'
  Clickatell_API: 'ENTER_YOUR_CLICKATELL_API'
  Name:
    - 'Bob'
    - 'Adam'
  Reciever:
    - '+48123456789'
    - '+46123456789'

# Deezer user credentials
Deezer:
  Deezer_Control: 'Enabled'
  User_id: '2247762504'

# DIY-HUE declarations
DIYHUE:
  DIYHUE_Control: 'Enabled'

# Domoticz declarations
Domoticz:
  # Set this to 'Enabled' to control devices registered to Domoticz server.
  Domoticz_Control: 'Enabled'
  Server_IP:
    # If you are running the server on the Pi itself, then change the ip to 'localhost'
    - '172.22.2.112'
  Server_port:
    - '8081'
  Devices:
    #The number of names and number of ids should be equal
    #List the device names that you want to control
    Name:
      - 'Relay 1'
      - 'Relay 2'
    #Assign the device ids in the same order as the devices listed above
    Id:
      - '1'
      - '2'

#ESP control declarations. Follow a similar pattern to add your own list.
ESP:
  ESP_Control: 'Disabled'
  IP: 'xxxxxxxxxxxx' #Enter your ESP's IP here.
[... 195 lines cut ...]
  Voice_Gender: 'Female'

#Custom wakeword On/Off.
#To enable custom wakeword, change status to Enabled and restart the assistant.
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  Custom_wakeword_models:
    - '/home/pi/GassistPi/src/resources/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy.umdl'

#Wemo/Emulated Wemo devices
Wemo:
  Wemo_Control: 'Disabled'
  Wemo_Devices:
    IP_Address:
      - 'Enter wemo device 1 ip here'
      - 'Enter wemo device 2 ip here'
    Device_Names:
      - 'Enter wemo device 1 name here'
      - 'Enter wemo device 1 name here'

#Youtube music streaming
YouTube:
  YouTube_Control: 'Enabled'''

shivasiddharth, 2020-11-01:
With the domoticz enabled, is the assistant not taking even time query ? Blank responses recorded has nothing to do with domoticz.

shivasiddharth, 2020-11-01:
Also can you try setting the port to 443

Welsyntoffie, 2020-11-01:
Response with Domoticz Disabled
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the time"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

ON_RESPONDING_FINISHED

ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Welsyntoffie, 2020-11-01:
Response with Domoticz Enabled

```ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "what's the time"}

Fatal Python error: Segmentation fault

Thread 0x1ddfc460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 75 in RunDetection
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 480 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x1b7ff460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 253 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f78ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1043 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
Segmentation fault
```

shivasiddharth, 2020-11-01:
Did you try changing the port in config.yaml to 443 ?

Welsyntoffie, 2020-11-01:
I changed like this, then rebooted my pi.
No it is stable, thank you for that simple suggestion. I thought http would be good enough.

```  # If you are running the server on the Pi itself, then change the ip to 'localhost'
    - 'localhost'
  Server_port:
    - '443 ```

Welsyntoffie, 2020-11-01:
It is working once after I command a relay to switch on, then I get the previous error...
Also the Speech response is long after the relay has been set.

```/home/pi/env/lib/python3.7/site-packages/urllib3/connectionpool.py:988: InsecureRequestWarning: Unverified HTTPS request is being made to host 'localhost'. Adding certificate verification is strongly advised. See: https://urllib3.readthedocs.io/en/latest/advanced-usage.html#ssl-warnings
  InsecureRequestWarning,
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:02] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Welsyntoffie, 2020-11-01:
While I'm here. Can I change the prefix "Custom" when using domoticz to something else.
eg. Hey Google, please turn on PC room light?

shivasiddharth, 2020-11-02:
> It is working once after I command a relay to switch on, then I get the previous error...

I really dont understand what you mean by this. It would be better if you can share the full terminal content to report.

To change the order, change the oder of "say and "devreq" in these lines. https://github.com/shivasiddharth/GassistPi/blob/8ed4af25aaf772a1c9d417df0b924fc07ffbc2e4/src/actions.py#L1541

I dont understand what you mean by "Custom" prefix. There is no trigger command for domoticz devices. The custom prefix is only for MQTT and IR commands. Please go through the README.

Welsyntoffie, 2020-11-02:
> I dont understand what you mean by "Custom" prefix. There is no trigger command for domoticz devices. The custom prefix is only for MQTT and IR commands. Please go through the README.

That was my mistake, I am sorry, I did read the section at MQTT point number 5 and thought that would be the same through out the program.
Regarding the error, I can give command, and the relay would switch, this could happen multiple times, then randomly I get the error and the service stops.
```
Fatal Python error: Segmentation fault

Thread 0x1ddfc460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 75 in RunDetection
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 480 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x1b7ff460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 253 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f78ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1043 in main
  File "/home/pi/GassistPi/src/main.py", line 1051 in <module>
Segmentation fault
```

shivasiddharth, 2020-11-02:
Random crashes is beyond me. This is a DIY thing and crashes are a part of it. Since the issue has been solved closing this .

closed 2020-11-02 by shivasiddharth

Welsyntoffie, 2020-11-02:
Alright.Thank you

Welsyntoffie, 2020-11-02:
Change web ports seems to fix the issue.
HTTP port to 8008
HTTPS port 1024
Not crashing anymore