shivasiddharth/GassistPi #1072: Error  AttributeError: module 'enum' has no attribute 'IntFlag'

e-budianto, 2020-11-03:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master or OSMC (both version tried)

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  

```
Paste your Board and OS details here.   
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a020d3
Serial          : 0000000064e1f340
Model           : Raspberry Pi 3 Model B Plus Rev 1.3

PRETTY_NAME="Open Source Media Center"
NAME="OSMC"
VERSION="June 2020"
VERSION_ID="2020.06-1"
ID=osmc
ID_LIKE=debian

ANSI_COLOR="1;31"
HOME_URL="https://www.osmc.tv"
[... 1210 lines cut ...]

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
OSMC branch:
 /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --device_model_id 'replaced by my model id'

```

### Paste the contents of the terminal below
```
OSMC branch:

Traceback (most recent call last):
  File "/home/osmc/GassistPi/src/main.py", line 32, in <module>
    import snowboydecoder
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/osmc/GassistPi/src/snowboydetect.py", line 15, in <module>
    import _snowboydetect
ImportError: /home/osmc/GassistPi/src/_snowboydetect.so: unexpected reloc type 0x03

```

shivasiddharth, 2020-11-03:
Master branch will not work with OSMC or you can say that I am not sure as I have not tested it. Can you please just post the errors related to the OSMC branch as I am lost with details from two branches.

e-budianto, 2020-11-03:
I'm sorry to make you confused,
here's my error message from osmc branch only:
I have problem during instalation after i enter google challenge code, here's the error message:
`Testing the installed google assistant. Make a note of the generated Device-Id
./GassistPi/scripts/gassist-installer.sh: line 186: googlesamples-assistant-hotword: command not found
`
and when i manually start the program:
`Traceback (most recent call last):
  File "/home/osmc/GassistPi/src/main.py", line 32, in <module>
    import snowboydecoder
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
  File "/home/osmc/GassistPi/src/snowboydetect.py", line 15, in <module>
    import _snowboydetect
ImportError: /home/osmc/GassistPi/src/_snowboydetect.so: unexpected reloc type 0x03`

shivasiddharth, 2020-11-03:
Can you share the terminal contents of the installation ? Need to check if Snowboy was compiled properly.

e-budianto, 2020-11-03:
Sure, I've edited in my post above.

shivasiddharth, 2020-11-04:
you will have two files in /GassistPi/src/resources/Stretch-wrapper folder copy those files and paste it in /GassistPi/src/ folder. It may ask you to replace the files. Replace the files and try starting the assistant.

e-budianto, 2020-11-04:
Great! :+1: , now it's working, however, not for long the program crashed. here's the error: 

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "how are you"}

ON_RENDER_RESPONSE:
{
  "text": "I've been busy staying up to date on the latest election news. Let me know if there's anything I can do for you",
  "type": 0
}

{'text': "I've been busy staying up to date on the latest election news. Let me know if there's anything I can do for you", 'type': <RenderResponseType.TEXT: 0>}
ON_RESPONDING_STARTED:
  {"is_error_response": false}

{'is_error_response': False}
ON_RESPONDING_FINISHED

None
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Fatal Python error: Segmentation fault

Thread 0x490f9460 (most recent call first):
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/osmc/GassistPi/src/main.py", line 246 in start_detector
  File "/usr/local/lib/python3.6/threading.py", line 864 in run
  File "/usr/local/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/local/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x76f33010 (most recent call first):
  File "/home/osmc/env/lib/python3.6/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/osmc/GassistPi/src/main.py", line 345 in main
  File "/home/osmc/GassistPi/src/main.py", line 353 in <module>
Segmentation fault
osmc@osmc:~$
```

Could you give me idea what's wrong?

shivasiddharth, 2020-11-04:
Can you give multiple commands and post the output.

e-budianto, 2020-11-04:
Here it is:
```
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/osmc/env/lib/python3.6/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: osmc-xxxxxxxxxxx
device_id: 87Fxxxxxxxxxxxxxxxxxxxx

ON_MUTED_CHANGED:
  {"is_muted": false}

Mic mute is set to: False
ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "how's the weather today"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

{'is_error_response': False}
ON_RESPONDING_FINISHED

None
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": true}

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Fatal Python error: Segmentation fault

Thread 0x46efd460 (most recent call first):
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 32 in get
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 157 in start
  File "/home/osmc/GassistPi/src/main.py", line 246 in start_detector
  File "/usr/local/lib/python3.6/threading.py", line 864 in run
  File "/usr/local/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/local/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x76f04010 (most recent call first):
  File "/home/osmc/env/lib/python3.6/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/osmc/GassistPi/src/main.py", line 345 in main
  File "/home/osmc/GassistPi/src/main.py", line 353 in <module>
Segmentation fault
osmc@osmc:~$
```

another one :
```
ON_MUTED_CHANGED:
  {"is_muted": false}

Mic mute is set to: False
ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "sing a song for me"}

Fatal Python error: Segmentation fault

Thread 0x48fff460 (most recent call first):
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/osmc/GassistPi/src/main.py", line 246 in start_detector
  File "/usr/local/lib/python3.6/threading.py", line 864 in run
  File "/usr/local/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/local/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x76fef010 (most recent call first):
  File "/home/osmc/env/lib/python3.6/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/osmc/GassistPi/src/main.py", line 345 in main
  File "/home/osmc/GassistPi/src/main.py", line 353 in <module>
Segmentation fault
osmc@osmc:~$
```

and another one:
```
ON_MUTED_CHANGED:
  {"is_muted": false}

Mic mute is set to: False
ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

Kodi TV box not online
ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "tell me about today"}

ON_RESPONDING_STARTED:
  {"is_error_response": false}

{'is_error_response': False}
ON_RESPONDING_FINISHED

None
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

Fatal Python error: Segmentation fault

Thread 0x491f9460 (most recent call first):
  File "/home/osmc/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/osmc/GassistPi/src/main.py", line 246 in start_detector
  File "/usr/local/lib/python3.6/threading.py", line 864 in run
  File "/usr/local/lib/python3.6/threading.py", line 916 in _bootstrap_inner
  File "/usr/local/lib/python3.6/threading.py", line 884 in _bootstrap

Thread 0x76f68010 (most recent call first):
  File "/home/osmc/env/lib/python3.6/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/osmc/GassistPi/src/main.py", line 345 in main
  File "/home/osmc/GassistPi/src/main.py", line 353 in <module>
Segmentation fault
osmc@osmc:~$
```

It hardly running past 3 min without error, except when there is no "Ok Google" hot word triggering.

shivasiddharth, 2020-11-04:
In the config.yaml can you disable Kodi control and see as to whats happening ? Random crashes are beyond my control.

e-budianto, 2020-11-04:
I believe it is not random crashes, but problem in snowboydecoder.py when it handles speech recognition in time.sleep procedure, bug probably. 
Just now i press ^C right after i start the main.py manually, without event saying a hotword, and it will crash immediately.

shivasiddharth, 2020-11-04:
Ctrl +C is meant to close or exit the program abruptly. Abrupt closure of assistant will obviously raise the segmentation fault error.

e-budianto, 2020-11-05:
just want to report i tried install master branch in freshly installed osmc image (OSMC_TGT_rbp2_20200402.img), exactly the same segmentation fault location. It needs a little adjustment though to some hardcoded home folder, but it works (minus segfault of course), plus i think more features than osmc branch. I think this segmentation fault is a bug from python in snowboydecoder, you can google the bug due to time.sleep function, some made different workaround logic.  

Many thanks for such an effort to make this excellent project. :+1:  I think this issue can be closed for now.

closed 2020-11-07 by shivasiddharth