shivasiddharth/GassistPi #1076: Google Assistant not starting automatically after reboot

vermillion191, 2020-11-23:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Master Branch 

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03112
Serial		: 10000000ff9679db
Model		: Raspberry Pi 4 Model B Rev 1.2

### Describe the bug:
```
The GA does not start's automatically even after completing the steps for auto-start
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes
[... 327 lines cut ...]
ALSA lib pcm_a52.c:823:(_snd_pcm_a52_open) a52 is only for playback
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_hdmi.pcm.iec958.0:CARD=0,AES0=6,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958:{AES0 0x6 AES1 0x82 AES2 0x0 AES3 0x2  CARD 0}
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
ALSA lib pcm_usb_stream.c:486:(_snd_pcm_usb_stream_open) Invalid type for card
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: assist-61a5d-myassistant-wrnkez
device_id: 9F5157E4ECAC3488172CBCA796A71A2A

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_MEDIA_STATE_IDLE

ON_START_FINISHED

```

shivasiddharth, 2020-11-23:
Is the manual start working fine ? Does GA reply to queries ?

What is the output of 
sudo systemctl status gassistpi.service

gdagen, 2020-12-06:
Same problem here.  I just did a fresh install of Raspbian Dec 2020 and GassistPi using AIY Hat.  Everything works perfectly when running manually, but the service isn't working on reboot. (no into jingle and "Hi".  Does not respond to commands)

Checking status after reboot:

sudo systemctl status gassistpi.service
```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-12-06 11:22:31 PST; 161ms ago
 Main PID: 915 (python)
    Tasks: 1 (limit: 2063)
   CGroup: /system.slice/gassistpi.service
           └─915 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassistpiv2 --device_model_id gassistpiv2-myassistant-v7djpn

Dec 06 11:22:31 raspberrypi systemd[1]: Started Google Assistant-Voice.

```

checking status again after a minute
```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2020-12-06 11:23:07 PST; 5s ago
 Main PID: 1008 (python)
    Tasks: 1 (limit: 2063)
   CGroup: /system.slice/gassistpi.service
           └─1008 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassistpiv2 --device_model_id gassistpiv2-myassistant-v7djpn

Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:201: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Dec 06 11:23:10 raspberrypi python[1008]:   options=None, file=DESCRIPTOR),
Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors
Dec 06 11:23:10 raspberrypi python[1008]:   index=0, containing_type=None, fields=[]),
Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is go
Dec 06 11:23:10 raspberrypi python[1008]:   serialized_end=727,
Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
Dec 06 11:23:10 raspberrypi python[1008]:   options=None, file=DESCRIPTOR),
Dec 06 11:23:11 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will
Dec 06 11:23:11 raspberrypi python[1008]:   from collections import Mapping, Container
```

I tried "OK Google Set volume to maximum".  Still nothing.

Everything works perfectly when running manually using:
```
sudo systemctl stop gassistpi.service
export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'gassistpiv2' --device_model_id 'gassistpiv2-myassistant-v7djpn'
```
I hear the intro jingle and "HI".  Speaker is LOUD and Assistant responds to commands.  I then reboot and get nothing.

I also made sure "Wait for network" on boot was checked.

shivasiddharth, 2020-12-08:
I think you guys have installed the Google Assistant while I was updating the project for Dec release. I clearly had a note in the git page asking users not to jump to the Dec release yet.

shivasiddharth, 2020-12-08:
Do: 
sudo nano lib/systemd/system/gassistPi.service

There change the Restart from always to never. Now reboot the Pi . Now run

journalctl -u gassistpi.service -f

See what the terminal says

aandroide, 2020-12-08:
journalctl -u gassistpi.service -f
result:
dic 08 16:20:03 pi python[1868]:   type=None,
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:93: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   type=None,
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:100: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   type=None,
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:103: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   name="ABORTED", index=11, number=10, serialized_options=None, type=None
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:106: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   name="OUT_OF_RANGE", index=12, number=11, serialized_options=None, type=None
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:113: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   type=None,
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:116: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   name="INTERNAL", index=14, number=13, serialized_options=None, type=None
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:119: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   name="UNAVAILABLE", index=15, number=14, serialized_options=None, type=None
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:122: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   name="DATA_LOSS", index=16, number=15, serialized_options=None, type=None
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/rpc/code_pb2.py:128: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   serialized_end=349,
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:32: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   dependencies=[google_dot_api_dot_annotations__pb2.DESCRIPTOR,google_dot_longrunning_dot_operations__pb2.DESCRIPTOR,google_dot_protobuf_dot_any__pb2.DESCRIPTOR,google_dot_protobuf_dot_duration__pb2.DESCRIPTOR,google_dot_protobuf_dot_empty__pb2.DESCRIPTOR,google_dot_protobuf_dot_timestamp__pb2.DESCRIPTOR,google_dot_rpc_dot_status__pb2.DESCRIPTOR,])
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:45: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   type=None),
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:49: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:03 pi python[1868]:   type=None),
dic 08 16:20:03 pi python[1868]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:53: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the  [... cut ...]
dic 08 16:20:16 pi python[1930]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 08 16:20:16 pi python[1930]:   options=None, file=DESCRIPTOR),
dic 08 16:20:16 pi systemd[1]: Stopping Google Assistant-Voice...
dic 08 16:20:16 pi systemd[1]: gassistpi.service: Main process exited, code=killed, status=15/TERM
dic 08 16:20:16 pi systemd[1]: gassistpi.service: Succeeded.
dic 08 16:20:16 pi systemd[1]: Stopped Google Assistant-Voice.

shivasiddharth, 2020-12-08:
@aandroide , you already have a issue open. If you are going to take up this issue, you can close the other one.  I will loose track shifting between issues.

What does the version file on the GassistPi say ?
Post output of aplay --version

rminformaticos, 2020-12-09:
Same problem here, today install all, raspbian lastest and gassistpi lastest

start gassistpi manually ok, with service nothing (install service, reboot and nothing)

aplay: version 1.1.8 by Jaroslav Kysela <perex@perex.cz>

shivasiddharth, 2020-12-09:
@rminformaticos  anyone adding to this thread please report https://github.com/shivasiddharth/GassistPi/issues/1076#issuecomment-740269430

shivasiddharth, 2020-12-09:
I also need to know if it's Raspberry Pi OS lite or full desktop version

aandroide, 2020-12-09:
For my , raspberry pi version desktop

rminformaticos, 2020-12-09:
full desktop version: 2020-12-02-raspios-buster-armhf-full.zip

journalctl -u gassistpi.service -f

-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
dic 09 14:58:01 raspberrypi python[464]:   File "/home/pi/GassistPi/src/main.py", line 218, in __init__
dic 09 14:58:01 raspberrypi python[464]:     self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
dic 09 14:58:01 raspberrypi python[464]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
dic 09 14:58:01 raspberrypi python[464]:     stream_callback=audio_callback)
dic 09 14:58:01 raspberrypi python[464]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
dic 09 14:58:01 raspberrypi python[464]:     stream = Stream(self, *args, **kwargs)
dic 09 14:58:01 raspberrypi python[464]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
dic 09 14:58:01 raspberrypi python[464]:     self._stream = pa.open(**arguments)
dic 09 14:58:01 raspberrypi python[464]: OSError: [Errno -9997] Invalid sample rate
dic 09 14:58:02 raspberrypi systemd[1]: gassistpi.service: Succeeded.

shivasiddharth, 2020-12-09:
@aandroide your journal looks unusually long. Looks like you have pasted the same thing over and again. Or you have not changed the restart in the systemd file as I have mentioned.

@rminformaticos what is your audio config or audio setup ?
Post output of: 
```
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
```

rminformaticos, 2020-12-09:
sudo nano /etc/asound.conf
sudo nano ./.asoundrc

Both files was empty

shivasiddharth, 2020-12-09:
In the config.yaml, try changing the custom wakeword engine to Picovoice and see what the journal says.

rminformaticos, 2020-12-09:
Changed file: GassistPi\src\config.yaml
Reboot

journalctl -u gassistpi.service -f

`-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
dic 09 15:30:23 raspberrypi python[468]:   File "/home/pi/GassistPi/src/main.py", line 218, in __init__
dic 09 15:30:23 raspberrypi python[468]:     self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
dic 09 15:30:23 raspberrypi python[468]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
dic 09 15:30:23 raspberrypi python[468]:     stream_callback=audio_callback)
dic 09 15:30:23 raspberrypi python[468]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
dic 09 15:30:23 raspberrypi python[468]:     stream = Stream(self, *args, **kwargs)
dic 09 15:30:23 raspberrypi python[468]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
dic 09 15:30:23 raspberrypi python[468]:     self._stream = pa.open(**arguments)
dic 09 15:30:23 raspberrypi python[468]: OSError: [Errno -9997] Invalid sample rate
dic 09 15:30:24 raspberrypi systemd[1]: gassistpi.service: Succeeded.

`

aandroide, 2020-12-09:
a curiosity, have you tried to see if your microphone records? with the arecord command

shivasiddharth, 2020-12-09:
I am really sick and I don't have access to my Pi right now. That is why I am asking you to try diff things.

Restart your pi. And do:
sudo systemctl stop gassitpi.service

Give a 2 min gap
sudo systemctl restart gassistpi.service

See what happens or journalctl says.

rminformaticos, 2020-12-09:
for me, microphone records, with arecord, and i can control gassist pi correctly if i execute manually.

Tests:
restart pi
pi@raspberrypi:~ $ sudo systemctl stop gassitpi.service
Failed to stop gassitpi.service: Unit gassitpi.service not loaded.
pi@raspberrypi:~ $ sudo systemctl restart gassistpi.service

journalctl -u gassistpi.service -f

-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
dic 09 15:47:59 raspberrypi python[1004]:   File "/home/pi/GassistPi/src/main.py", line 218, in __init__
dic 09 15:47:59 raspberrypi python[1004]:     self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
dic 09 15:47:59 raspberrypi python[1004]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
dic 09 15:47:59 raspberrypi python[1004]:     stream_callback=audio_callback)
dic 09 15:47:59 raspberrypi python[1004]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
dic 09 15:47:59 raspberrypi python[1004]:     stream = Stream(self, *args, **kwargs)
dic 09 15:47:59 raspberrypi python[1004]:   File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
dic 09 15:47:59 raspberrypi python[1004]:     self._stream = pa.open(**arguments)
dic 09 15:47:59 raspberrypi python[1004]: OSError: [Errno -9997] Invalid sample rate
dic 09 15:48:00 raspberrypi systemd[1]: gassistpi.service: Succeeded.

shivasiddharth, 2020-12-09:
Do 
sudo nano /etc/pulse/client.conf.d/00-disable-autospawn.conf

Add a # symbol infront of autospawn. Save the file. Restart the Pi and see what the journalctl says.

If it does not work, you guys need to wait untill I figure out the issue and fix. As mentioned earlier, I am really sick 😷🤢 right now.

rminformaticos, 2020-12-09:
Added # to autospawn=on
Reboot

Raspberry pi says "Hi"

Test say: "Ok Google" but nothing happens

pi@raspberrypi:~ $ journalctl -u gassistpi.service -f
-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
dic 09 15:59:54 raspberrypi python[467]: mmap() failed: Cannot allocate memory
dic 09 15:59:54 raspberrypi python[467]: Failed to create permanent mapping for memfd region with ID = 2997306824
dic 09 15:59:54 raspberrypi python[467]: Ignoring received block reference with non-registered memfd ID = 2997306824
dic 09 15:59:54 raspberrypi python[467]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Protocol error
dic 09 15:59:54 raspberrypi python[467]: [ERROR:audio_output_stream.cc(72)] Device could not be opened: default
dic 09 15:59:54 raspberrypi python[467]: ON_RESPONDING_FINISHED
dic 09 15:59:54 raspberrypi python[467]: ON_RENDER_RESPONSE:
dic 09 15:59:54 raspberrypi python[467]:   {"text": "Hola, ¿en qué puedo ayudarte? 😃", "type": 0}
dic 09 15:59:54 raspberrypi python[467]: ON_CONVERSATION_TURN_FINISHED:
dic 09 15:59:54 raspberrypi python[467]:   {"with_follow_on_turn": false}

shivasiddharth, 2020-12-09:
Ok we are making progress. 
Are you using 1gb version of a Pi board ?

rminformaticos, 2020-12-09:
I supose, i have a Raspberry Pi 3 Model B.

How can i check it

shivasiddharth, 2020-12-09:
No need to check Pi 3 models come with only 1Gb. Open a terminal and run

`echo 20 > /proc/sys/vm/nr_hugepages`

Restart after this. If this does not work then you need to change the OS from full fledged version to limited Desktop version. Or you can try creating a swapfile.  Get back on this. I will provide instructions for creating a swapfile.

rminformaticos, 2020-12-09:
pi@raspberrypi:~ $ echo 20 > /proc/sys/vm/nr_hugepages
bash: /proc/sys/vm/nr_hugepages: No such file or directory

My sd was 32 gb

I downloaded this morning: Raspberry Pi OS with desktop and recommended software
I must downoaded this??? Raspberry Pi OS with desktop

from https://www.raspberrypi.org/software/operating-systems/

I updated OS today, i had older Raspbian version.

shivasiddharth, 2020-12-09:
Ok try this:
```
sudo dd if=/dev/zero of=/var/swap2 bs=1024 count=1000000
sudo chmod 600 /var/swap2
sudo mkswap /var/swap2
sudo swapon /var/swap2
sudo swapon
```

Reboot.

rminformaticos, 2020-12-09:
pi@raspberrypi:~ $ sudo dd if=/dev/zero of=/var/swap2 bs=1024 count=1000000
1000000+0 registros leídos
1000000+0 registros escritos
1024000000 bytes (1,0 GB, 977 MiB) copied, 77,2522 s, 13,3 MB/s
pi@raspberrypi:~ $ sudo chmod 600 /var/swap2
pi@raspberrypi:~ $ sudo mkswap /var/swap2
Setting up swapspace version 1, size = 976,6 MiB (1023995904 bytes)
no label, UUID=8e224e89-20f3-4bf2-9ed6-e306e36b6de4
pi@raspberrypi:~ $ sudo swapon /var/swap2
pi@raspberrypi:~ $ sudo swapon
NAME       TYPE   SIZE USED PRIO
/var/swap  file   100M 256K   -2
/var/swap2 file 976,6M   0B   -3

---- REBOOT ------

-- Logs begin at Thu 2019-02-14 11:11:59 CET. --
dic 09 16:57:13 raspberrypi python[465]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
dic 09 16:57:14 raspberrypi python[465]: device_model_id: homepi-aries
dic 09 16:57:14 raspberrypi python[465]: device_id: A2FF527635060FB40D63687AB055BEFC
dic 09 16:57:14 raspberrypi python[465]: ON_MUTED_CHANGED:
dic 09 16:57:14 raspberrypi python[465]:   {"is_muted": false}
dic 09 16:57:18 raspberrypi python[465]: ON_START_FINISHED
dic 09 16:57:18 raspberrypi python[465]: ON_MEDIA_STATE_IDLE
dic 09 16:57:36 raspberrypi pulseaudio[922]: [alsa-source-USB Audio] alsa-source.c: ALSA woke us up to read new data from the device, but there was actually nothing to read.
dic 09 16:57:36 raspberrypi pulseaudio[922]: [alsa-source-USB Audio] alsa-source.c: Most likely this is a bug in the ALSA driver 'snd_usb_audio'. Please report this issue to the ALSA developers.
dic 09 16:57:36 raspberrypi pulseaudio[922]: [alsa-source-USB Audio] alsa-source.c: We were woken up with POLLIN set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
dic 09 16:58:00 raspberrypi python[465]: ON_CONVERSATION_TURN_STARTED
dic 09 16:58:08 raspberrypi python[465]: ON_END_OF_UTTERANCE
dic 09 16:58:08 raspberrypi python[465]: ON_END_OF_UTTERANCE
dic 09 16:58:08 raspberrypi python[465]: ON_RECOGNIZING_SPEECH_FINISHED:
dic 09 16:58:08 raspberrypi python[465]:   {"text": "hola"}
dic 09 16:58:08 raspberrypi python[465]: ON_RENDER_RESPONSE:
dic 09 16:58:08 raspberrypi python[465]:   {"text": "Echaba de menos tu voz.", "type": 0}
dic 09 16:58:09 raspberrypi python[465]: ON_RESPONDING_STARTED:
dic 09 16:58:09 raspberrypi python[465]:   {"is_error_response": false}
dic 09 16:58:09 raspberrypi python[465]: mmap() failed: Cannot allocate memory
dic 09 16:58:09 raspberrypi python[465]: Failed to create permanent mapping for memfd region with ID = 4168094882
dic 09 16:58:09 raspberrypi python[465]: Ignoring received block reference with non-registered memfd ID = 4168094882
dic 09 16:58:09 raspberrypi python[465]: ALSA lib pulse.c:243:(pulse_connect) PulseAudio: Unable to connect: Protocol error
dic 09 16:58:09 raspberrypi python[465]: [ERROR:audio_output_stream.cc(72)] Device could not be opened: default
dic 09 16:58:09 raspberrypi python[465]: ON_RENDER_RESPONSE:
dic 09 16:58:09 raspberrypi python[465]:   {"text": "¿Puedo ayudarte?", "type": 0}
dic 09 16:58:09 raspberrypi python[465]: ON_RESPONDING_FINISHED
dic 09 16:58:09 raspberrypi python[465]: ON_CONVERSATION_TURN_FINISHED:
dic 09 16:58:09 raspberrypi python[465]:   {"with_follow_on_turn": true}
dic 09 16:58:09 raspberrypi python[465]: ON_CONVERSATION_TURN_STARTED
dic 09 16:58:17 raspberrypi python[465]: ON_END_OF_UTTERANCE
dic 09 16:58:17 raspberrypi python[465]: ON_END_OF_UTTERANCE
dic 09 16:58:17 raspberrypi python[465]: ON_RECOGNIZING_SPEECH_FINISHED:
dic 09 16:58:17 raspberrypi python[465]:   {"text": ""}
dic 09 16:58:17 raspberrypi python[465]: ON_NO_RESPONSE
dic 09 16:58:17 raspberrypi python[465]: ON_CONVERSATION_TURN_FINISHED:
dic 09 16:58:17 raspberrypi python[465]:   {"with_follow_on_turn": false}

-------

Was the same, gassistpi loads buts nothing sound

shivasiddharth, 2020-12-09:
@rminformaticos  are you interested in all the features of GassistPi or you need just the Google Assistant ?

rminformaticos, 2020-12-09:
Google Assistant was most important, if i can play radio stations and custom scripts Well

Yesterday works correctly with older raspbian version, the problem was when this morning update to the new version today

shivasiddharth, 2020-12-09:
Did you do a sudo upgrade instead of starting from a fresh image ? I would suggest you to use the minimal desktop version that is around 1.7 GB

aandroide, 2020-12-09:
guys i wanted to try to start the assistant automatically and strangely it doesn't restart anymore.
It only restarts when I run custom commands but it also did it with the old raspbian version for about 3 weeks. and that's why I wanted to update the image.
Since my mistakes are different from this discussion, I will open a new discussion.
do you agree shivasiddhart?

shivasiddharth, 2020-12-09:
@aandroide check the gitter general discussion. The custom command restarts are due to gtts fault. You need to update the gtts to the latest version.

aandroide, 2020-12-09:
@shivasiddharth  
everything is up to date, both the system image and the entire gassistpi project

shivasiddharth, 2020-12-09:
@aandroide do manual start give custom conversation command see what is causing the crash, if the reason is already not discussed only then open an issue.

aandroide, 2020-12-09:
I'll do the test in a moment but my eye fell on this. should we worry? https://developers.google.com/assistant/sdk/guides/library/python/embed/run-sample
[image]

aandroide, 2020-12-09:
> Do:
> sudo nano lib/systemd/system/gassistPi.service
> 
> There change the Restart from always to never. Now reboot the Pi . Now run
> 
> journalctl -u gassistpi.service -f
> 
> See what the terminal says

pi@pi:~ $ journalctl -u gassistpi.service -f
-- Logs begin at Wed 2020-12-09 20:43:41 CET. --
dic 09 20:51:10 pi python[1335]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
dic 09 20:51:10 pi python[1335]: Thread 0x76fd6ad0 (most recent call first):
dic 09 20:51:10 pi python[1335]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
dic 09 20:51:10 pi python[1335]:   File "/usr/lib/python3.7/queue.py", line 179 in get
dic 09 20:51:10 pi python[1335]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
dic 09 20:51:10 pi python[1335]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
dic 09 20:51:10 pi python[1335]:   File "/home/pi/GassistPi/src/main.py", line 1110 in main
dic 09 20:51:10 pi python[1335]:   File "/home/pi/GassistPi/src/main.py", line 1121 in <module>
dic 09 20:51:10 pi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
dic 09 20:51:10 pi systemd[1]: gassistpi.service: Failed with result 'signal'.
dic 09 20:52:27 pi systemd[1]: Started Google Assistant-Voice.
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   serialized_end=121,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
dic 09 20:52:30 pi python[1419]:   file=DESCRIPTOR,
dic 09 20:52:30 pi python[1419]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated cod [... cut ...]
dic 09 20:52:38 pi python[1419]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
dic 09 20:52:38 pi python[1419]:   File "/home/pi/GassistPi/src/main.py", line 1110 in main
dic 09 20:52:38 pi python[1419]:   File "/home/pi/GassistPi/src/main.py", line 1121 in <module>
dic 09 20:52:38 pi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
dic 09 20:52:38 pi systemd[1]: gassistpi.service: Failed with result 'signal'.

aandroide, 2020-12-09:
nothing to do i reinstalled gassistpi and the restart problem came back. i changed system file from always to never and i get these errors

aandroide, 2020-12-09:
@vermillion191 @rminformaticos  
guys i think i found out so it does reboot our system.
Please follow my procedure and tell me if you have solved the problem.
open the config.yaml file and change this line Custom_Wakeword: 'Enabled' to Custom_Wakeword: 'Disabled'
to me it solved the restart problem. if he does the same to you, sid will know better what to work on to correct the problem.

shivasiddharth, 2020-12-10:
@aandroide , your issue is different from the others'. You are confusing others and me. Your problem is that the assistant is restarting. The main issue reported here is diff. Open up a seperate issue I will deal with you there.

shivasiddharth, 2020-12-10:
@vermillion191  and @gdagen  please read through my comments for @rminformaticos . That should solve your issue.

gdagen, 2020-12-10:
I originally installed the latest Raspbian minimal desktop version (1.7GB) on Dec 7 and then Gassistpi (following all instructions).  I did not upgrade.  What else should I try?

BTW, I tried flashing a new micro SD card with the older Raspbian version and everything works, but it automatically restarts Gassist Pi a few times (I hear the intro jingle and "Hi" multiple times) before it stays and everything works fine.  Very strange,  I'm using the AIY Hat and the appropriate drivers you provided (based on Raspbian version).

shivasiddharth, 2020-12-10:
@gdagen  Did you try this. https://github.com/shivasiddharth/GassistPi/issues/1076#issuecomment-741824052  this is what is causing multiple restarts before stabilization.

shivasiddharth, 2020-12-10:
@gdagen see if you have mpg123 folder in the /home/pi directory. If you have, then it means that you are on the older version. 
Also to double check do 
aplay --version
You should get 1.1.8 if you get 1.1.3 then it means that you are on the older version.

aandroide, 2020-12-10:
> Stesso problema qui. Ho appena eseguito una nuova installazione di Raspbian Desktop OS Dec 2020 e GassistPi utilizzando AIY Hat. Tutto funziona perfettamente quando viene eseguito manualmente, ma il servizio non funziona al riavvio. (nessun jingle introduttivo e "Ciao". Non risponde ai comandi)
> 
> Verifica dello stato dopo il riavvio:
> 
> sudo systemctl status gassistpi.service
> 
> ```
> ● gassistpi.service - Google Assistant-Voice
>    Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2020-12-06 11:22:31 PST; 161ms ago
>  Main PID: 915 (python)
>     Tasks: 1 (limit: 2063)
>    CGroup: /system.slice/gassistpi.service
>            └─915 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassistpiv2 --device_model_id gassistpiv2-myassistant-v7djpn
> 
> Dec 06 11:22:31 raspberrypi systemd[1]: Started Google Assistant-Voice.
> ```
> 
> controllando nuovamente lo stato dopo un minuto
> 
> ```
> ● gassistpi.service - Google Assistant-Voice
>    Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2020-12-06 11:23:07 PST; 5s ago
>  Main PID: 1008 (python)
>     Tasks: 1 (limit: 2063)
>    CGroup: /system.slice/gassistpi.service
>            └─1008 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassistpiv2 --device_model_id gassistpiv2-myassistant-v7djpn
> 
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:201: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
> Dec 06 11:23:10 raspberrypi python[1008]:   options=None, file=DESCRIPTOR),
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors
> Dec 06 11:23:10 raspberrypi python[1008]:   index=0, containing_type=None, fields=[]),
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is go
> Dec 06 11:23:10 raspberrypi python[1008]:   serialized_end=727,
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors
> Dec 06 11:23:10 raspberrypi python[1008]:   options=None, file=DESCRIPTOR),
> Dec 06 11:23:11 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/validictory/validator.py:7: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will
> Dec 06 11:23:11 raspberrypi python[1008]:   from collections import Mapping, Container
> ```
> 
> Ho provato "OK Google Imposta il volume al massimo". Ancora niente.
> 
> Tutto funziona perfettamente quando si esegue manualmente utilizzando:
> 
> ```
> sudo systemctl stop gassistpi.service
> export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs
> /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'gassistpiv2' --device_model_id 'gassistpiv2-myassistant-v7djpn'
> ```
> 
> Sento il jingle dell'intro e "HI". L'altoparlante è FORTE e l'assistente risponde ai comandi. Quindi riavvio e non ottengo nulla.
> 
> Mi sono anche assicurato che "Attendi rete" all'avvio fosse selezionato.

> Stesso problema qui. Ho appena eseguito una nuova installazione di Raspbian Desktop OS Dec 2020 e GassistPi utilizzando AIY Hat. Tutto funziona perfettamente quando viene eseguito manualmente, ma il servizio non funziona al riavvio. (nessun jingle introduttivo e "Ciao". Non risponde ai comandi)
> 
> Verifica dello stato dopo il riavvio:
> 
> sudo systemctl status gassistpi.service
> 
> ```
> ● gassistpi.service - Google Assistant-Voice
>    Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
>    Active: active (running) since Sun 2020-12-06 11:22:31 PST; 161ms ago
>  Main PID: 915 (python)
>     Tasks: 1 (limit: 2063)
>    CGroup: /system.slice/gassistpi.service
>            └─915 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id gassistpiv2 --device_model_id gassistpiv2-myassistant-v7djpn
> 
[... 16 lines cut ...]
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:215: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors
> Dec 06 11:23:10 raspberrypi python[1008]:   index=0, containing_type=None, fields=[]),
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:218: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is go
> Dec 06 11:23:10 raspberrypi python[1008]:   serialized_end=727,
> Dec 06 11:23:10 raspberrypi python[1008]: /home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/proto/cloud_speech_pb2.py:235: DeprecationWarning: Call to deprecated create function FieldD [... cut ...]
> 
> Sento il jingle dell'intro e "HI". L'altoparlante è FORTE e l'assistente risponde ai comandi. Quindi riavvio e non ottengo nulla.
> 
> Mi sono anche assicurato che "Attendi rete" all'avvio fosse selezionato.

the message sid eliminated was for this user I didn't want to confuse anyone but his problem was also identical and I solved it by disabling snowboy in config.yaml and giving the command gassistpi start it never restarted. sorry if i insist sid but i have already dealt with his problem.

rminformaticos, 2020-12-10:
@shivasiddharth 

New tests: do it a fresh install with 2020-12-02-raspios-buster-armhf
Install gassistpi, disabled autospawn, reboot and starts well with service

But GA's volume is to low, i say: "Ok Google: sube el volumen al máximo" (i test in english, and diferent words in spanish, but nothing do) Says "OK" and nothing change, volume is the same, low.
I say "Ok Google cual es el volumen": Says: "Volume is at 100%" (in spanish)

Yesterday with full OS volume increases and listens good

Any ideas????

Thanks

shivasiddharth, 2020-12-10:
Open the desktop and see if the volume is full on the speaker icon on top.

rminformaticos, 2020-12-10:
[image]

shivasiddharth, 2020-12-10:
Do:
sudo apt-get install pavucontrol

Open a terminal and enter
pavucontrol

That will open up a window. Check for the volume levels there.

Also open a terminal and run 
speaker-test

If the speaker-test volume is not high then you need to check the speaker

rminformaticos, 2020-12-10:
Installed but i dont see nothing python

[image]

[image]

All volumes was at 100 %

rminformaticos, 2020-12-10:
Well i change in first tab: "system sounds" and increase to 153% and GA's volume increases and sounds good.

[image]

aandroide, 2020-12-10:
ok I try to get back in line with your mistakes, I re-enabled Custom_Wakeword.
I followed comment #1076 (comment) and I also confirm that it doesn't restart anymore.
i apologize to @shivasiddharth i was convinced it was a mistake in snowboy.

gdagen, 2020-12-10:
For what it's worth. I did not have a mpg123 folder and am running aplay version 1.1.8.

I commented out autospawn as specified in [#1076 (comment)](https://github.com/shivasiddharth/GassistPi/issues/1076#issuecomment-741824052), rebooted and now the service loads on boot and everything works.  Thanks @shivasiddharth !

aandroide, 2020-12-10:
@gdagen 
if you ask your voice assistant Do you have a family. does he answer you? or does it restart?

gdagen, 2020-12-10:
@aandroide 
If I say "Do you have a family".   I hear back, "The whole world is my family".  Everything works perfectly.  Did you modify your config.yaml file?  If so, there may be a typo in there.  For instance, you cannot have any punctuation within the Custom Conversation section.  Including periods and apostrophes.   When in doubt, replace your config.yaml file with the original one for testing.

aandroide, 2020-12-10:
@gdagen  
thanks for the test I'm trying to understand if mine is an isolated case. one last question use gassistpi updated in december? and is the raspbian system image still the old one or the new one from December?

gdagen, 2020-12-10:
@aandroide it looks like the winning combination (for me anyway) is:

1) Flash micro sd card with the latest Raspberry Pi OS with desktop (Dec 2, 2020 - 1.17GB version)
2) Install GassistPi following all instructions (including setting up Google Assistant to auto start on boot)
3) sudo nano /etc/pulse/client.conf.d/00-disable-autospawn.conf    (add a # in front of autospawn=on)

reboot and everything works.

shivasiddharth, 2020-12-10:
This thread's OP @vermillion191 has never bothered to return and respond since opening the issue.
From the positive responses of others I believe we have fixed this issue.
I will automate the autospawn thingy in the next update.

aandroide, 2020-12-10:
@gdagen we just found out that my problems show up when setting the it-IT language to en-US everything works fine. I opened a thread other than this for my problem because as shivasiddharth says the initial problem has been solved by this thread. Thanks again for your test.

commit 9c0b3ecb (2020-12-14): Bug fixes

Fixes #1087  Fixes #1085  Fixes #1076
closed 2020-12-14 by shivasiddharth