shivasiddharth/GassistPi #1077: Script won't Autoboot. Status showing active (running)..

Zwillyzz, 2020-11-26:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
pi@raspberrypi:~ $ cat /proc/cpuinfo 
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 000000007a3f9a99
Model		: Raspberry Pi 3 Model B Rev 1.2

pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
[... 383 lines cut ...]
Yes/No
Yes.
### 4. Is the assistant service restarting automatically?
Yes/No
No.
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
Same above. I can run the assistant and it works. Youtube works, but it's a hit and miss at this point but I think it's the cloud having issues.
```

### Paste the contents of the terminal below
```
Same as above.
```
  
Everything works as intended but I can't get the script to boot automatically. Ran sudo systemctl status gassistpi.service as explained above.

shivasiddharth, 2020-11-26:
Your systemctl status output is truncated can you provide the full output ? Also, please go through the existing closed and open issues with autoboot not working. Your issue may have already been covered.

Zwillyzz, 2020-11-26:
I've gone through a few already. I can't find a solid lead.

I've been trying to figure how hot to view full output. Could you tell me how? Thanks

Also ran:
`sudo /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'assistpi-a61af' --device_model_id 'assistpi-a61af-googleassistpi-tl3jod'`

Script runs the same as manually BUT ends with this: 
```
usage: main.py [-h] [--device-model-id DEVICE_MODEL_ID]
               [--project-id PROJECT_ID] [--nickname NICKNAME]
               [--device-config DEVICE_CONFIG_FILE]
               [--credentials OAUTH2_CREDENTIALS_FILE] [--query QUERY] [-v]
main.py: error: argument --credentials: No such file or directory: "/root/.config/google-oauthlib-tool/credentials.json
```

shivasiddharth, 2020-11-26:
I have replied to you regarding that in youtube. You should not use sudo. 
Use `sudo status gassistpi.service -l` to view full logs.

Zwillyzz, 2020-11-26:
That code didn't work. Got
`pi@raspberrypi:~ $ sudo status gassistpi.service -l
sudo: status: command not found
`

Zwillyzz, 2020-11-26:
I think systemctl was forgotten? I added that I got the following:

```
`pi@raspberrypi:~ $ sudo systemctl status gassistpi.service -l`

● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Wed 2020-11-25 19:07:05 PST; 778ms ago
 Main PID: 6618 (python)
    Tasks: 1 (limit: 1939)
   CGroup: /system.slice/gassistpi.service
           └─6618 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Nov 25 19:07:05 raspberrypi systemd[1]: Started Google Assistant-Voice.
lines 1-9/9 (END)

```

Zwillyzz, 2020-11-26:
Also ran:

`sudo journalctl -u gassistpi.service`

Got the following:

```
pi@raspberrypi:~ $ sudo journalctl -u gassistpi.service
-- Logs begin at Wed 2020-11-25 18:42:28 PST, end at Wed 2020-11-25 19:03:35 PST
Nov 25 18:47:35 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   serialized_pb=b'\n\x15google/api/htt
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   serialized_end=121,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packag
lines 1-23...skipping...
-- Logs begin at Wed 2020-11-25 18:42:28 PST, end at Wed 2020-11-25 19:03:35 PST. --
Nov 25 18:47:35 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. 
Nov 25 18:47:38 raspberrypi python[1208]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away.
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Plea
Nov 25 18:47:38 raspberrypi python[1208]:   serialized_end=121,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:38 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away
Nov 25 18:47:38 raspberrypi python[1208]:   file=DESCRIPTOR,
[... cut ...]
Nov 25 18:47:39 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:70: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors 
Nov 25 18:47:39 raspberrypi python[1208]:   file=DESCRIPTOR,
Nov 25 18:47:39 raspberrypi python[1208]: /home/pi/env/lib/python3.7/site-packages/google/longrunning/operations_proto_pb2.py:97: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors 
Nov 25 18:47:39 raspberrypi python[1208]:   file=DESCRIPTOR,
lines 1-60

```

shivasiddharth, 2020-11-26:
Do you hear the "Hi" chime ?
Post the output of:
```
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
aplay -l
arecord -l
```

Zwillyzz, 2020-11-26:
Yes I hear the "Hi" chime when I execute it manually.

`sudo nano /etc/asound.conf`

`pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
`

`sudo nano ./.asoundrc`

`pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
`
`aplay -l`

`**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
`

`arecord -l`

`**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
`

Also...executing `sudo systemctl status gassistpi.service -l` is yielding different results too. Ran it a second time and got this:

`pi@raspberrypi:~ $ sudo systemctl -l status gassistpi
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset
   Active: active (running) since Wed 2020-11-25 19:16:49 PST; 5s ago
 Main PID: 1112 (python)
    Tasks: 1 (limit: 1939)
   CGroup: /system.slice/gassistpi.service
           └─1112 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --pr

Nov 25 19:16:51 raspberrypi python[1112]: /home/pi/env/lib/python3.7/site-packag
Nov 25 19:16:51 raspberrypi python[1112]:   options=None, file=DESCRIPTOR),
Nov 25 19:16:51 raspberrypi python[1112]: /home/pi/env/lib/python3.7/site-packag
Nov 25 19:16:51 raspberrypi python[1112]:   index=0, containing_type=None, field
Nov 25 19:16:51 raspberrypi python[1112]: /home/pi/env/lib/python3.7/site-packag
Nov 25 19:16:51 raspberrypi python[1112]:   serialized_end=727,
Nov 25 19:16:51 raspberrypi python[1112]: /home/pi/env/lib/python3.7/site-packag
Nov 25 19:16:51 raspberrypi python[1112]:   options=None, file=DESCRIPTOR),
Nov 25 19:16:52 raspberrypi python[1112]: /home/pi/env/lib/python3.7/site-packag
Nov 25 19:16:52 raspberrypi python[1112]:   from collections import Mapping, Con
lines 1-18/18 (END)

`

shivasiddharth, 2020-11-26:
Can you please paste the details carefully ? Both .asoundrc and asound.conf  are truncated. I wanted to know if you heard a "Hi" on autostart ?

Zwillyzz, 2020-11-26:
Sorry. Trying to figure out how to select all the text in terminal.

No hi chime on autostart.

Zwillyzz, 2020-11-26:
However for both of those .asoundrc and asound.conf that's all I see when I run the command in terminal.

Zwillyzz, 2020-11-26:
[asound.txt](https://github.com/shivasiddharth/GassistPi/files/5600311/asound.txt)

Here is the actual asound.conf.

shivasiddharth, 2020-11-26:
If a service start fails, it will try restarting itself a number of times and eventually stop. You can find the full logs only after that. Else everytime you try and get the log, you will get only a chunck. Boot the Pi, leave it alone for 5-10 mins and then try to get the systemctl status and the journalctl status.

shivasiddharth, 2020-11-26:
Or do this:
```
sudo systemctl stop gassistpi.service
sudo systemctl start gassistpi.service
journalctl -u gassistpi.service -f
```

Zwillyzz, 2020-11-26:
That opened up some perspective with the status. It seems to be looping. I will post the full terminal soon but this caught my attention:

`Nov 25 19:55:18 raspberrypi systemd[1]: gassistpi.service: Succeeded.
Nov 25 19:55:18 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Nov 25 19:55:18 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 25.
Nov 25 19:55:18 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 25 19:55:18 raspberrypi systemd[1]: Started Google Assistant-Voice.`

That restart counter is gradually increasing.

Zwillyzz, 2020-11-26:
Managed to get one cycle. It's doing more or less of the same. 

`Nov 25 20:19:41 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Nov 25 20:19:41 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 184.
Nov 25 20:19:41 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 25 20:19:41 raspberrypi systemd[1]: Started Google Assistant-Voice.
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:35: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   serialized_pb=b'\n\x15google/api/http.proto\x12\ngoogle.api"T\n\x04Http\x12#\n\x05rules\x18\x01 \x03(\x0b\x32\x14.google.api.HttpRule\x12\'\n\x1f\x66ully_decode_reserved_expansion\x18\x02 \x01(\x08"\x81\x02\n\x08HttpRule\x12\x10\n\x08selector\x18\x01 \x01(\t\x12\r\n\x03get\x18\x02 \x01(\tH\x00\x12\r\n\x03put\x18\x03 \x01(\tH\x00\x12\x0e\n\x04post\x18\x04 \x01(\tH\x00\x12\x10\n\x06\x64\x65lete\x18\x05 \x01(\tH\x00\x12\x0f\n\x05patch\x18\x06 \x01(\tH\x00\x12/\n\x06\x63ustom\x18\x08 \x01(\x0b\x32\x1d.google.api.CustomHttpPatternH\x00\x12\x0c\n\x04\x62ody\x18\x07 \x01(\t\x12\x15\n\rresponse_body\x18\x0c \x01(\t\x12\x31\n\x13\x61\x64\x64itional_bindings\x18\x0b \x03(\x0b\x32\x14.google.api.HttpRuleB\t\n\x07pattern"/\n\x11\x43ustomHttpPattern\x12\x0c\n\x04kind\x18\x01 \x01(\t\x12\x0c\n\x04path\x18\x02 \x01(\tBj\n\x0e\x63om.google.apiB\tHttpProtoP\x01ZAgoogle.golang.org/genproto/googleapis/api/annotations;annotations\xf8\x01\x01\xa2\x02\x04GAPIb\x06proto3',
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:62: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:80: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:92: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   serialized_end=121,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:119: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:137: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:155: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:173: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:191: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:209: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 25 20:19:43 raspberrypi python[9488]:   file=DESCRIPTOR,
Nov 25 20:19:43 raspberrypi python[9488]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:227: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or qu [... cut ...]
Nov 25 20:19:50 raspberrypi python[9488]: OSError: [Errno -9996] Invalid input device (no default output device)
Nov 25 20:19:51 raspberrypi systemd[1]: gassistpi.service: Succeeded.
Nov 25 20:19:51 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Nov 25 20:19:51 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 185.
Nov 25 20:19:51 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 25 20:19:51 raspberrypi systemd[1]: Started Google Assistant-Voice.
`

shivasiddharth, 2020-11-26:
Which Raspberry Pi image are you using ?

Zwillyzz, 2020-11-26:
Using the AIY Kits Release 2019-11-13 on https://github.com/google/aiyprojects-raspbian/releases

It's based on buster though.

shivasiddharth, 2020-11-26:
Can you test this on the Buster from the official raspberry pi downloads section ?

Zwillyzz, 2020-11-26:
Ok I will try Raspberry Pi OS with desktop and recommended software
Release date: August 20th 2020

at https://www.raspberrypi.org/software/operating-systems/

and report back.

Zwillyzz, 2020-11-26:
Real quick. The youtube function is running weird because there's an issue with gTTS token? Requesting a song is super slow and it's a hit or miss. It works sometimes, and if not, the script crashes giving a segmentation error. I re-run the script and pray it works the 2nd time.

Zwillyzz, 2020-11-26:
Ok it worked on a fresh image but it restarted 10 times or so before saying "Hi". Would hear the chime each time, finally it chime and said Hi and it started to work.

```
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:263: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:281: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:297: DeprecationWarning: Call to deprecated create function OneofDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   fields=[],
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:301: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   serialized_end=381,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:328: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:346: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/http_pb2.py:358: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   serialized_end=430,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/annotations_pb2.py:42: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   google_dot_protobuf_dot_descriptor__pb2.DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/api/annotations_pb2.py:64: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:39: DeprecationWarning: Call to deprecated create function FileDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   dependencies=[google_dot_protobuf_dot_any__pb2.DESCRIPTOR],
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:66: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:84: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:102: DeprecationWarning: Call to deprecated create function FieldDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
Nov 26 14:40:08 raspberrypi python[1422]:   file=DESCRIPTOR,
Nov 26 14:40:08 raspberrypi python[1422]: /home/pi/env/lib/python3.7/site-packages/google/rpc/status_pb2.py:114: DeprecationWarning: Call to deprecated create function Descriptor(). Note: Create unlinked d [... cut ...]
Nov 26 14:40:30 raspberrypi python[1484]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Nov 26 14:40:30 raspberrypi python[1484]: device_model_id: assistpi-a61af-googleassistpi-tl3jod
Nov 26 14:40:30 raspberrypi python[1484]: device_id: 08E3842816CA4B9B88898CB372941291
Nov 26 14:40:31 raspberrypi python[1484]: ON_MUTED_CHANGED:
Nov 26 14:40:31 raspberrypi python[1484]:   {"is_muted": false}
Nov 26 14:40:31 raspberrypi python[1484]: ON_MEDIA_STATE_IDLE
Nov 26 14:40:31 raspberrypi python[1484]: ON_START_FINISHED
```

Zwillyzz, 2020-11-26:
Here's what caught my eye:

```
Nov 26 14:52:02 raspberrypi python[3250]: ON_MUTED_CHANGED:
Nov 26 14:52:02 raspberrypi python[3250]:   {"is_muted": false}
Nov 26 14:52:06 raspberrypi python[3250]: ON_START_FINISHED
Nov 26 14:52:06 raspberrypi python[3250]: ON_MEDIA_STATE_IDLE
Nov 26 14:52:06 raspberrypi python[3250]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Nov 26 14:52:06 raspberrypi python[3250]: Fatal Python error: Aborted
Nov 26 14:52:06 raspberrypi python[3250]: Thread 0x5c6fc460 (most recent call first):
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 32 in get
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/snowboydecoder.py", line 157 in start
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/main.py", line 504 in start_detector
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 865 in run
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
Nov 26 14:52:06 raspberrypi python[3250]: Thread 0x61eff460 (most recent call first):
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/main.py", line 277 in pushbutton
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 865 in run
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
Nov 26 14:52:06 raspberrypi python[3250]: Thread 0x66dff460 (most recent call first):
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 552 in wait
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 1164 in run
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap
Nov 26 14:52:06 raspberrypi python[3250]: Thread 0x76f36ad0 (most recent call first):
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/threading.py", line 300 in wait
Nov 26 14:52:06 raspberrypi python[3250]:   File "/usr/lib/python3.7/queue.py", line 179 in get
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/main.py", line 1110 in main
Nov 26 14:52:06 raspberrypi python[3250]:   File "/home/pi/GassistPi/src/main.py", line 1121 in <module>
Nov 26 14:52:06 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Nov 26 14:52:06 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Nov 26 14:52:06 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Nov 26 14:52:06 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 19.

```

Zwillyzz, 2020-11-27:
Also Youtube plays wayyy better, faster. Should add a disclaimer in the READ ME, not to use the AIY Image! That kit lead me to your project indirectly, and I already had the image installed.

Just need help to figure out why gassistpi.service takes to many restarts to work. Sometimes it's after 5 times, sometimes it takes 20 times. Again, I would hear the chime with no "Hi" and it would loop and loop until I hear the chime and "Hi".

Zwillyzz, 2020-11-27:
Solved. I took an educational assumption based on the error that the issue was hardware related. Switched power supply and it runs flawlessly now.

Could you recommend me a power supply with an on/off switch?

shivasiddharth, 2020-11-27:
What does the numbers within the version file say ?

Zwillyzz, 2020-11-27:
I think I got it working. Problem: Poor power supply. So far, so good. Other problem: Google AIY image.

Google cloud however, is a bit bumpy (inconsistent at times). Also learned there is a 500 call limit daily? Why? 

I was getting the ON_ASSISTANT_ERROR earlier this evening. Was it because of that limit?

Also, is there a way to increase the input volume for the microphone? I have to speak to it pretty close. I don't want to have to yell.

shivasiddharth, 2020-11-27:
> python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed. 

Is an audio problem. Do:
`sudo apt-get install pulseaudio`
to solve that. 

If you hit the 500 limit, it will say that something went wrong or google assistant cant be reached. You need to increase the volume of the mic from alsamixer. Type alsamixer in terminal and change the volume.

Zwillyzz, 2020-11-27:
Is there a command to uninstall the current pulseaudio installed to re-install? It's saying it's the latest version

Playback and capture all at 100% lol.

shivasiddharth, 2020-11-28:
I have fixed all of the audio issues in the last update. If you want, you try from scratch again with the last update. Or you can just do a `sudo apt-get purge pulseaudio` and then `sudo apt-get install pulseaudio` and `sudo apt-get install -f`

Zwillyzz, 2020-11-29:
Still does it. Tried multiple power supplies already. It gets to the chime, I hear no "Hi" and it restarts. 

However, after certain restarts it works eventually, but sometimes (on occasion) it randomly crashes when requesting a YouTube link. 

I thought it was a power issue but it doesn't seem to be the case. I bought a power supply today and I no longer receive low voltage notification.

shivasiddharth, 2020-11-29:
What does your version number read ?

Zwillyzz, 2020-11-29:
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 000000007a3f9a99
Model		: Raspberry Pi 3 Model B Rev 1.2
```

```
pi@raspberrypi:~ $ sudo cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

shivasiddharth, 2020-11-29:
Numbers in the version file in the GassistPi folder

Zwillyzz, 2020-11-29:
```
{
  "Version":{
    "SDK":060,
    "Features":038,
    "Revision":001
  }
}
```

shivasiddharth, 2020-11-29:
Can you post the output of aplay --version ?

Zwillyzz, 2020-11-29:
```
pi@raspberrypi:~ $ aplay --version
aplay: version 1.1.3 by Jaroslav Kysela <perex@perex.cz>
```

shivasiddharth, 2020-11-29:
I tested the latest build on the latest buster without any issues. I am unable to replicate your issue.

shivasiddharth, 2020-11-29:
I will leave this issue open to see if others report of multiple restarts.

Zwillyzz, 2020-11-29:
What Raspberry Pi do you use? I'm thinking of upgrading to the 3B+. Perhaps it's a performance issue? Or thinking of overclocking the CPU a bit to test.

shivasiddharth, 2020-11-29:
I have Pi3 and Pi4. Its not a performance issue. Both offer similar performance with respect to GA.

shivasiddharth, 2020-11-29:
Pi3B+ is already couple of years old. Get the Pi4 if you want.

Zwillyzz, 2020-11-29:
Will the AIY Hat be compatible with the Pi4 though?

shivasiddharth, 2020-11-29:
Which version of AIY you have ? Will test it out and let you know.

Zwillyzz, 2020-11-29:
I bought the V1 kit.

Okay thanks!

Zwillyzz, 2020-11-29:
Did a fresh build last buster and latest GassistPi and same issue persists with a different memory card. Overclocking didn't help either, so I will just pass on upgrading the pi. FYI, I have no issues with the standard google assistant demo included with the kit, but it has no Youtube :).

Anyway, I can start it manually 10 times in a row and have zero hiccups. The issue is auto start. Could there be a different way to code auto start?

Zwillyzz, 2020-11-29:
Found a significant piece of info. Explained in the edits above ^.

ymaj42, 2020-12-05:
Since no one else is chiming in, for the record I am also having this issue. It starts eventually but never on the first try. I am also using the AIY hat on the Pi4.

shivasiddharth, 2020-12-05:
Immediately after booting run this on the terminal.
journalctl -u gassistpi.service -f

this should point out the cause for the restarts.

shivasiddharth, 2020-12-06:
The audio related issues have been fixed. Please re-open the issue if the problem persists.

closed 2020-12-06 by shivasiddharth