shivasiddharth/GassistPi #1079: After update, GassistPi fails to load with "Invalid input device" Error in pyaudio.py

gdagen, 2020-11-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
```
```
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : a02082
Serial          : 000000007bd2e657
Model           : Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
[... 307 lines cut ...]
type hw
card 1
}
ctl.!default {
type hw
card 1
}

$cat /etc/asound.conf
options snd-usb-audio index=1
pcm.!default {
type hw
card 1
}
ctl.!default {
type hw
card 1
}

Within /usr/share/alsa/alsa.conf

defaults.ctl.card 1
defaults.pcm.card 1

```

shivasiddharth, 2020-11-27:
You have not used the audio configurations that I have provided. Check the project specific audio configs.

gdagen, 2020-11-27:
I did originally, but when it didn't work I tried to simplify things.  OK, I just reinstalled the Audio Config:

```
pi@raspberrypi:~ $ sudo chmod +x ./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh

pi@raspberrypi:~ $ sudo ./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh
Renaming /home/pi/.asoundrc to /home/pi/.asoundrc.bak...
Renaming /etc/asound.conf to /etc/asound.conf.bak...
Installing USB DAC config

pi@raspberrypi:~ $ cat ~/.asoundrc
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

pi@raspberrypi:~ $ cat /etc/asound.conf
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}

```
I then rebooted:
```
$ sudo reboot
```

Once rebooted, I run the following command:
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'voice-api-xxxxxx' --device_model_id 'xxxxxx' (actual ids hidden here)
```
and I get the same error
```
2020-11-27 10:47:14,281 : ERROR : [Errno -9996] Invalid input device (no default output device)
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 218, in __init__
    self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9996] Invalid input device (no default output device)

```

shivasiddharth, 2020-11-27:
Something is not right about your aplay -l. Where are the HMDI and Jack cards ? Why are they missing ?
Does the Samson Go have a speaker ? Can it be used as a laptop speaker ?

gdagen, 2020-11-27:
I dunno about the HDMI etc. being missing.  It's a standard raspberry pi 3 B.  The Samson GoMic has a headphone jack that I use as the "speaker".  I should mention that everything was working perfectly before updating gassistpi.  

[image]

shivasiddharth, 2020-11-28:
Did you use Noobs ? Is that a Lite version or a Desktop version of the OS ?

gdagen, 2020-11-28:
It's the Raspbian Desktop version for Google "AIY Kits Release 2019-11-13".  Installed from the img file.  Got the image from here:
https://github.com/google/aiyprojects-raspbian/releases

Got the whole thing working using the Google AIY voice kit, but wanted to install GassistsPi for all the additional features.  I did, and it worked great.  I later went back and installed the latest version of GassistPi and it stopped working :-(

Should I try installing from a fresh Raspbian Desktop (Buster) not the google AIY version?

shivasiddharth, 2020-11-28:
Yup. I dont recommend AIY image.

gdagen, 2020-11-28:
OK, I flashed the SD card with the latest version of Raspbin Desktop (Aug2020), updated all software and rebooted.  I then installed GassistPi and when I got to the audio drivers, I installed the USB-DAC drivers.  The speaker-test didn't work (just got silence).  I then attempted to install the drivers for the USB MIC and Audio Jack (using your instructions about copying the files from the Extras folder).

Still silence when running:  $ speaker-test

I tried both the audio jack on the Raspberry pi and the audio jack on the GoMic.  Neither worked.

$ aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: GoMic [Samson GoMic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 2: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
```
$ arecord -l
```
**** List of CAPTURE Hardware Devices ****
card 1: GoMic [Samson GoMic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
$ cat /home/pi/.asoundrc
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
$ cat /etc/asound.conf
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:1,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
When in the Raspbian Desktop environment, I can see the speaker icon in the taskbar, but it is greyed out and has a red X.  No other audio options or mixers are available.

gdagen, 2020-11-28:
I just tried changing the /home/pi/.asoundrc  and /etc/asound.conf files back to what I originally had with the old img, and now the audio tests work.

$ cat /home/pi/.asoundrc
```
options snd-usb-audio index=1
pcm.!default {
type hw
card 1
}
ctl.!default {
type hw
card 1
}
```
$ cat /etc/asound.conf
```
options snd-usb-audio index=1
pcm.!default {
type hw
card 1
}
ctl.!default {
type hw
card 1
}
```
is it safe to continue with the install?  or do I need to add something for the "dsnoop" and "dnsooper"?

gdagen, 2020-11-28:
I also tried changing the audio settings using the following and this also worked!  So I only need to know if I can proceed with the install or if I need to change something to make sure it works with GassistPi.

$ cat /home/pi/.asoundrc
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
```
$ cat /etc/asound.conf
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
```

gdagen, 2020-11-29:
OK, I figured out that my audio device # kept changing every time I rebooted, so I unplugged the HDMI.  
Every time I used the USB-MIC-JACK drivers, it never worked.  I tried creating a hybrid of the settings that worked and the ones you've provided and I ended up with this:

$ cat /home/pi/.asoundrc
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "dsnooper"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:0,0"
  }
}
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
```
$ cat /etc/asound.conf
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "dsnooper"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:0,0"
  }
}
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
```
I rebooted and now all of the speaker-tests work!  I can hear audio finally.

so now, I go to manually launch GassistPi using:

$ export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs
$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'voice-api-xxx' --device_model_id 'xxx' (ids hidden)

and guess what?  I get the exact same error as originally posted.
```
2020-11-28 16:38:30,013 : ERROR : [Errno -9996] Invalid input device (no default output device)
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 218, in __init__
    self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9996] Invalid input device (no default output device)
```
I was really hoping this was getting somewhere.  :-(

shivasiddharth, 2020-11-29:
You could have just waited for my response. Remeber that I may not be in the same time zone as you. Use the USB-DAC settings and get back with the asound.conf and .asoundrc. After a reboot.

gdagen, 2020-11-29:
I appreciate your help with all this.  I tried the USB-DAC settings already and they didn't work, but I will try again here.

$ sudo ./GassistPi/audio-drivers/USB-DAC/scripts/install-usb-dac.sh
```
Renaming /home/pi/.asoundrc to /home/pi/.asoundrc.bak...
Renaming /etc/asound.conf to /etc/asound.conf.bak...
Installing USB DAC config
```
$ sudo reboot

$ cat /home/pi/.asoundrc
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

$cat /etc/asound.conf
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}
pcm.dmixer {
    type dmix
    ipc_key 1024
    slave {
        pcm "hw:1,0"
        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100
    }

    }
pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "dmixer"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```

$ aplay -l
```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 1: GoMic [Samson GoMic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

$ arecord -l
```
**** List of CAPTURE Hardware Devices ****
card 1: GoMic [Samson GoMic], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

```

shivasiddharth, 2020-11-29:
What happens when you run speaker-test ? Can you share the screenshot of the terminal running speaker-test. Also please make sure that volume is set to maximum using alsamixer

shivasiddharth, 2020-11-29:
If you don't hear anything even after setting the output volume to maximum, remove this part in both asound.conf and .asoundrc

        period_time 0
        period_size 1024
        buffer_size 8192
        rate 44100

gdagen, 2020-11-29:
speaker-test results in no audio.  only silence.  I have to [Ctrl + C] to get out.  alsamixer playback volume is set to maximum for GoMic.  Still no audio for speaker-test.

$ speaker-test 
```
speaker-test 1.1.3

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 192 to 2097152
Period size range from 64 to 699051
Using max buffer size 2097152
Periods = 4
was set period_size = 524288
was set buffer_size = 2097152
 0 - Front Left
Time per period = 12.304836
 0 - Front Left
^C
```
one thing I did notice is that the default sound device in alsamixer is "Pulse Audio" (see attached screenshot).  This seemed strange since it doesn't show up in aplay -l or arecord -l

[image]

[image]

shivasiddharth, 2020-11-29:
Can you do a 
sudo apt-get purge pulseaudio
Reboot and try the speaker test ?

gdagen, 2020-11-30:
It worked!  After removing PulseAudio and rebooting, now I hear the static from the speaker-test. 

For fun, I tried launching gassistpi, but still getting the same error.

$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'voice-api-xxxxxx' --device_model_id 'xxxxxxxxxxxx'  (actual ids hidden here)
```
...
[01b4d5e0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01b91878] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication
2020-11-29 16:13:27,144 : ERROR : [Errno -9996] Invalid input device (no default output device)
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 218, in __init__
    self.detector = snowboydecoder.HotwordDetector(snowboy_models, sensitivity=self._sensitivities)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 115, in __init__
    stream_callback=audio_callback)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/pyaudio.py", line 441, in __init__
    self._stream = pa.open(**arguments)
OSError: [Errno -9996] Invalid input device (no default output device)
```

shivasiddharth, 2020-11-30:
What are the numbers in the version file that you have in the GassistPi folder ?

Can you put back the part of the codes you removed and try again ?

gdagen, 2020-11-30:
$ cat ~/GassistPi/version
```
{
  "Version":{
    "SDK":060,
    "Features":038,
    "Revision":001
  }
}
```
I put the codes back into /etc/asound.conf and /home/pi/.asoundrc files and rebooted.

$ speaker-test  (still works.  I'm hearing audio)

shivasiddharth, 2020-12-01:
In the alsamixer, in the F6 window, can you select the Samson Go and see what happens ?

gdagen, 2020-12-01:
I have already tried that several times.  It looks like this is standard behavior for the alsamixer.  Always defaults to "-".  I tried 

$ sudo alslamixer

and it launched showing the GoMic, but after clicking F6 it shows the same thing (with default at the top next to "-")

shivasiddharth, 2020-12-01:
I have edited the comment please check that.

gdagen, 2020-12-01:
I just installed portaudio19-dev and rebooted.  What did you want me to do now?  Just try to run GassistPi or alsamixer?

shivasiddharth, 2020-12-01:
Try running GassistPi

gdagen, 2020-12-01:
Progress!!

After running GassistPi, I got a little further.  I even heard the google assistant startup wav and the assistant saying "Hi", but it immediately crashed with a new error.  Here are the last lines:

```
[01d97060] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01ddb430] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
USB-DAC
Initializing GPIOs 5 and 6 for assistant activity indication
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
[... cut ...]
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 1099, in main
    device_model_id, device_id, args.nickname)
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/device_helpers.py", line 85, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device NOT_FOUND (404): Requested entity was not found.
pi@raspberrypi:~ $
```

shivasiddharth, 2020-12-01:
Seems like an issue with the ids that you have used or the authentication process was not completed properly

gdagen, 2020-12-01:
was working before.  What do you suggest?  reinstall Gassistpi?

shivasiddharth, 2020-12-01:
Looks like you have used the model_id first and then the project_id. Double check the command that you have used to manually start the assistant. You need to present the terminal contents full including the command that you used to start the assistant. Masking things does not help and no one can do any malice with these ids.

gdagen, 2020-12-01:
I just took a look at the new instructions (Looks great BTW!) and I see that the method to manually start GAssistPi has changed.

Is this the correct way?  It look like the project ID is now missing from the syntax.  I also see that doing a direct copy/paste results in the incorrect apostrophe and an mdash instead of two hyphens.

I'm guessing the correct method is:
```
$ export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'replace this with project id' --device_model_id 'replace this with the model id'
```

[image]

shivasiddharth, 2020-12-01:
Thanks for pointing out. I have fixed that now.

gdagen, 2020-12-01:
I triple checked and my ids are correct.  The new syntax that you are suggesting doesn't work and results in the following:

command:
```
$ export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'voice-api-296023' --device_model_id '679184919539'
```
result:
```
-bash: export: `/home/pi/env/bin/python': not a valid identifier
-bash: export: `-u': not a valid identifier
-bash: export: `/home/pi/GassistPi/src/main.py': not a valid identifier
-bash: export: `--project_id': not a valid identifier
-bash: export: `voice-api-296023': not a valid identifier
-bash: export: `--device_model_id': not a valid identifier
-bash: export: `679184919539': not a valid identifier
```

shivasiddharth, 2020-12-01:
export...... is the first command.
/home/... is the second command. 

You have run  both on the same line.

gdagen, 2020-12-01:
Ah.  That was not clear.  OK, I just did the following commands:
```
export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs
```
and
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'voice-api-296023' --device_model_id '679184919539'
```
and I'm getting the same error when it gets to the "Registering..." part.
```
Registering...2020-12-01 09:09:38,960 : ERROR : Failed to register device NOT_FOUND (404): Requested entity was not found.
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 1099, in main
    device_model_id, device_id, args.nickname)
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/device_helpers.py", line 85, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device NOT_FOUND (404): Requested entity was not found.

```

shivasiddharth, 2020-12-01:
You can do that.

shivasiddharth, 2020-12-01:
Not required actually. Do this.

```
source env/bin/activate
google-oauthlib-tool --scope https://www.googleapis.com/auth/assistant-sdk-prototype \
          --scope https://www.googleapis.com/auth/gcm \
          --save --headless --client-secrets "ADD YOUR PATH TO THE CREDENTIALS FILE HERE"
```

gdagen, 2020-12-01:
I reinstalled GassistPi before I saw your new response.  Authorized successfully.  I had to purge pulseaudio again, rebooted and tried to run again, but getting the same error.

gdagen, 2020-12-01:
I just tried authorizing using your method above.  I got the message:

credentials saved: /home/pi/.config/google-oauthlib-tool/credentials.json

I rebooted and then the following commands:
```
export MPG123_MODDIR=/home/${USER}/mpg123-1.25.13/src/libout123/modules/.libs
```

```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'voice-api-296023' --device_model_id '679184919539'
```
same error
```
Registering...2020-12-01 09:57:38,605 : ERROR : Failed to register device NOT_FOUND (404): Requested entity was not found.
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 1121, in <module>
    Myassistant().main()
  File "/home/pi/GassistPi/src/main.py", line 1099, in main
    device_model_id, device_id, args.nickname)
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/device_helpers.py", line 85, in register_device
    raise RegistrationError(r, device_model_id)
google.assistant.library.device_helpers.RegistrationError: Failed to register device NOT_FOUND (404): Requested entity was not found.

```

ozbotics, 2020-12-01:
I had the same problem. (Just today, solved it just now)
I am using a RPi 3B with fresh Buster OS and an Adafruit Voice Bonnet.
Initially I had the same issue with the 'Invalid input device (no default output device)' problem. Following the advice here I purged PulseAudio, rebooted, installed portaudio19-dev, rebooted - successfully passed the Audio device initialization, but got caught on the 'Failed to register device PERMISSION_DENIED' error.

At that point I abandoned my original google account setup, started again and had success with new credentials.

gdagen, 2020-12-01:
Thanks ozbotics.  That was exactly it for me too.  I just started all over.  

Refresh install of Raspbian OS
purge PulseAudio (reboot)
install portaudio19-dev (reboot)
Start all over with Google credentials and created a new project, etc.

There was a small blip in the process when I couldn't get the authorization link to work, but I logged back into Google Console and created a "test" user using the same gmail email account as my console login and then it worked.  Still had to add an exception in the browser since the app isn't "verified by google", but that worked.  Pasted the authorization code into gassistpi and launched manually.  Everything works perfectly!

closed 2020-12-01 by gdagen

ozbotics, 2020-12-02:
I wonder if GassistPi can work with PulseAudio? 
Although I personally can't see the point of adding another layer of abstraction and introducing latency if its not necessary.
Perhaps the install script (or the installation instructions) should be modified to include the purge PulseAudio and install portaudio19-dev steps.

Glad to have been of some assistance (or at least confirmation). Your conversation with shivasiddharth helped me considerably. 
(nice timing)

shivasiddharth, 2020-12-02:
@ozbotics , There are a number of different audio combinations that people use. It's hard to have a universal installer script. I have tried to automate things to my best. I will try and accommodate the purge commands and port audio installation in the installer script. But if someone chooses to install the assistant and then configure the audio, the patch will obviously not work. 
It's not that hard to seach the issue and find a solution in the git.

ozbotics, 2020-12-02:
Hi @shivasiddharth, 

My comment wasn't a slap at you, you've done an excellent job in what I have found to be a very slippery area.

This problem doesn't appear to be limited to one of a subset of audio devices / OSs. I tried this on both Stretch and Buster, with an Adafruit card (and @gdagen on a USB mic / ?audio). We both discovered that we could setup our audio devices in ~/.asoundrc and /etc/asound.conf with no difficulty. The problem for both of us seems to be that after gassist-installer.sh has been run PulseAudio has been installed and for whatever reason - it breaks the system.

After several man-days of work and half a dozen complete re-installs, I was ecstatic to discover your conversation with @gdagen which led me to the "purge pulseaudio,  install portaudio19-dev, regenerate credentials" solution, without which I would have been entirely boned. I imagine that while the emotional rollercoaster would be less steep, other people suffering from the same problem will significantly benefit from a line or two in the installation guide covering this.

I don't know a lot about Linux Audio, (more today than I did yesterday ;)  I am curious, are there configurations that do work with PulseAudio installed? (I'm happy with my currently working system, but) could it be configured with the right .asoundrc / asound.conf so that these additional steps are not necessary?

shivasiddharth, 2020-12-03:
The issue is that majority of the audio devices like 1. AIY HATs 2. USB Mic and HDMI/JACK 3. Respeaker HATs need pulseaudio. The ALSA plugin is treacherous in Buster builds. Stretch came with ALSA 1.1.3 which was really stable. The buster builds have updated ALSA which caused issues with portaudio. Downgrading ALSA,  compiling portaudio and mpg123 from scratch removed the requirement for pulseaudio. But I always take extra precautionary steps as most of them are new to linux/pi etc. Some projects ask people to run the upgrade command. 

Inadvertently, some upgrades could brake the downgraded ALSA or compiled portaudio. The installed pulseaudio is an added level of measure to prevent crashes in those cases. 

Only instances where the pulseaudio is not working is with USB DAC, MATRIX HAT AND  Respeaker 2 Mic HAT. I have a potential cause and fix in my mind. Need to see if it works.

Agreed that users will panic if they run into some errors. Even I want to avoid that coz it's one less issue for me to deal with here. Lolz..

Try this out.
sudo apt-get install pulseaudio
sudo apt-get install pavucontrol

Now launch the pavucontrol and there choose the USB DAC there. Now see if the GassistPi works by manually launching.

shivasiddharth, 2020-12-03:
Here is the solution for the USB-DAC to work with pulseaudio. Disable the onboard Pi audio in /boot/config.txt. I will incorporate this with the USB-DAC audio configuration installer.

ozbotics, 2020-12-03:
Ok, I try this and let you know.

ozbotics, 2020-12-03:
It turns out that the Adafruit Voice Bonnet is actually a Respeaker 2 Mic HAT. I tried your advice (meant for USB-DAC), but I ended up having the same problem as above. For the moment I have a working system (using the purge pulseaudio ... method). I'll look forward to your potential fix. Hope it works out.

Aside: I notice on https://github.com/respeaker/seeed-voicecard/tree/master/pulseaudio they provide instruction to enable PulseAudio for ReSpeaker 4 Mic Array and 6 Mic Array, but not Respeaker 2 Mic. I wonder what that is about.

shivasiddharth, 2020-12-04:
If you are using HATs like Respeaker, etc. You should use the specific audio config and not the USB-DAC. The fix I have issued is for the general USB-DACs.

gdagen, 2020-12-04:
It's really hard to accommodate every possible configuration people might have, but I'm very happy to say this is now working for me and I've learned a lot about audio on the raspberry pi.  Thank you, shivasiddharth for your hard work on building this amazingly useful tool and keeping it going!!  I will continue to contribute any findings that may be useful to improve it.

shivasiddharth, 2020-12-06:
Dec 2020 Raspberry Pi OS update is awesome. I have updated the project. All audio issues should be ironed out.

ozbotics, 2020-12-06:
Can confirm. Works perfectly on my Pi 3B + Adafruit Voice Bonnet (Respeaker 2 Mic HAT)
Thanks for your great work