shivasiddharth/GassistPi #1085: the say command causes the whole gassistpi system to go haywire

aandroide, 2020-12-10:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

master with the latest update in December

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
pi@pi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000a91ab7c2
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@pi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@pi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage5
pi@pi:~ $ 
  
```

### Describe the bug:
```

I initially thought my custom commands were the problem because they use the say command.
I then tried to disable the say command from my custom command and it ran successfully.
I then tried to ask google to play music from youtube and even there being the say command the service has stopped.
so I really believe that with the latest version of gtts there are changes to the "say" program.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "Riproduci bla bla bla da YouTube"}

stopping vlc
'NoneType' object has no attribute 'group'
Fatal Python error: Segmentation fault

Thread 0x1d4ea460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 505 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x1fcef460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 278 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f8cad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1135 in main
  File "/home/pi/GassistPi/src/main.py", line 1143 in <module>
Errore di segmentazione

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
reboot with any custom command and not using the "say" function
```

### Paste the contents of the terminal below
```

```

aandroide, 2020-12-10:
`pi@pi:~ $ pip list
Package           Version    
----------------- -----------
arandr            0.1.9      
asn1crypto        0.24.0     
automationhat     0.2.0      
blinker           1.4        
blinkt            0.1.2      
buttonshim        0.0.2      
Cap1xxx           0.1.3      
certifi           2018.8.24  
chardet           3.0.4      
Click             7.0        
colorama          0.3.7      
colorzero         1.1        
configparser      3.5.0b2    
cookies           2.2.1      
cryptography      2.6.1      
drumhat           0.1.0      
entrypoints       0.3        
enum34            1.1.6      
envirophat        1.0.0      
ExplorerHAT       0.4.2      
Flask             1.0.2      
fourletterphat    0.1.0      
funcsigs          1.0.2      
gpiozero          1.5.1      
gTTS              2.2.1      
gyp               0.1        
idna              2.6        
ipaddress         1.0.17     
itsdangerous      0.24       
Jinja2            2.10       
keyring           17.1.1     
keyrings.alt      3.1.1      
MarkupSafe        1.1.0      
mcpi              0.1.1      
microdotphat      0.2.1      
mock              2.0.0      
mote              0.0.4      
motephat          0.0.3      
numpy             1.16.2     
oauthlib          2.1.0      
olefile           0.46       
pantilthat        0.0.7      
pbr               4.2.0      
phatbeat          0.1.1      
pianohat          0.1.0      
picamera          1.13       
picraft           1.0        
piglow            1.2.5      
pigpio            1.44       
Pillow            5.4.1      
pip               18.1       
pycairo           1.16.2     
pycrypto          2.6.1      
pyflakes          2.0.0      
pygame            1.9.4.post1
PyGObject         3.30.4     
pyinotify         0.9.6      
PyJWT             1.7.0      
pyOpenSSL         19.0.0     
pyserial          3.4        
pyxdg             0.25       
rainbowhat        0.1.0      
requests          2.21.0     
requests-oauthlib 1.0.0      
responses         0.9.0      
RPi.GPIO          0.7.0      
RTIMULib          7.2.1      
scrollphat        0.0.7      
scrollphathd      1.2.1      
SecretStorage     2.3.1      
sense-emu         1.1        
sense-hat         2.2.0      
setuptools        40.8.0     
simplejson        3.16.0     
six               1.12.0     
skywriter         0.0.7      
sn3218            1.2.7      
spidev            3.4        
touchphat         0.0.1      
twython           3.7.0      
unicornhathd      0.0.4      
urllib3           1.24.1     
Werkzeug          0.14.1     
wheel             0.32.3     
pi@pi:~ $ 
`

aandroide, 2020-12-10:
{
  "Version":{
    "SDK":061,
    "Features":038,
    "Revision":001
  }
}

aandroide, 2020-12-10:
I created an audio test .py file and it works fine.
the file is created and I can play it correctly by opening it manually

#!/usr/bin/env python
from gtts import gTTS
tts = gTTS('questo e un messaggio di prova', lang='it')
tts.save('prova.mp3')

shivasiddharth, 2020-12-10:
Is there a log file generated ? If there is a log file can you please attach that. Did you try the basic custom conversation ? Custom question and answer that is there in the config file

aandroide, 2020-12-10:
gassistpi.log=
2020-12-10 21:57:12,205 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2020-12-10 21:57:12,215 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2020-12-10 21:57:12,479 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2020-12-10 21:57:23,542 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2020-12-10 21:57:23,822 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None
2020-12-10 21:57:24,020 : DEBUG : https://translate.google.com:443 "GET / HTTP/1.1" 200 None

aandroide, 2020-12-10:
yes I asked do you have a family and it exited the program and the gassistpi.log reports the same data posted above

shivasiddharth, 2020-12-10:
Try this:
```
source env/bin/activate
pip3 uninstall gtts-token
```
Restart and try

shivasiddharth, 2020-12-10:
If you have the gender as Male, try setting the gender to Female and check.

aandroide, 2020-12-10:
I followed your instructions and nothing has changed

aandroide, 2020-12-10:
i don't know if it can help, i kept the tmp folder open to see if mp3 file is generated but nothing is generated

shivasiddharth, 2020-12-10:
Share the output of 
```
source env/bin/activate
pip3 freeze
```

aandroide, 2020-12-10:
(env) pi@pi:~ $ pip3 freeze
adafruit-io==2.1
aftership==1.1.0
appdirs==1.4.4
beautifulsoup4==4.9.3
bs4==0.0.1
cachetools==4.1.1
casttube==0.2.1
certifi==2020.12.5
cffi==1.14.4
chardet==3.0.4
click==6.7
colorzero==1.1
coverage==5.3
coveralls==2.2.0
cryptography==3.2
decorator==4.4.2
docopt==0.6.2
enum34==1.1.10
feedparser==6.0.2
futures==3.1.1
gmusicapi==13.0.0
google-api-core==1.23.0
google-api-python-client==1.12.8
google-assistant-grpc==0.2.1
google-assistant-library==1.1.0
google-assistant-sdk==0.6.0
google-auth==1.23.0
google-auth-httplib2==0.0.4
google-auth-oauthlib==0.2.0
google-cloud-speech==0.36.0
google-cloud-texttospeech==0.3.0
googleapis-common-protos==1.52.0
googletrans @ git+https://github.com/shivasiddharth/py-googletrans.git@506ec8e5a161045b1effaffec565b1a704e392c8
gpiozero==1.5.1
gpsoauth==0.4.1
grpcio==1.34.0
gTTS==2.2.1
httplib2==0.18.1
idna==2.10
ifaddr==0.1.7
kodi-json==1.0.0
lxml==4.6.2
MechanicalSoup==0.12.0
mock==3.0.5
mutagen==1.45.1
netifaces==0.10.9
numpy==1.19.4
oauth2client==4.1.3
oauthlib==3.1.0
pafy==0.5.5
paho-mqtt==1.4.0
pathlib2==2.3.5
pkg-resources==0.0.0
proboscis==1.2.6.0
protobuf==3.14.0
psutil==5.7.3
pushbullet.py==0.12.0
pvporcupine==1.8.7
pyasn1==0.4.8
pyasn1-modules==0.2.8
PyAudio==0.2.11
PyChromecast==7.5.1
pycparser==2.20
pycryptodomex==3.7.2
pyOpenSSL==20.0.0
python-dateutil==2.8.1
python-magic==0.4.18
python-vlc==3.0.11115
pytz==2020.4
pywemo==0.4.39
PyYAML==3.13
requests==2.25.0
requests-oauthlib==1.3.0
RPi.GPIO==0.7.0
rsa==4.6
sgmllib3k==1.0.0
six==1.15.0
sounddevice==0.3.15
SoundFile==0.10.3.post1
soupsieve==2.0.1
spidev==3.5
spotipy @ git+https://github.com/plamere/spotipy.git@94e164385e1b64f99e45053c6bdd638770c982e2
tenacity==4.12.0
uritemplate==3.0.1
urllib3==1.26.2
validictory==1.1.2
websocket-client==0.57.0
youtube-dl==2020.12.7
zeroconf==0.28.6
(env) pi@pi:~ $

shivasiddharth, 2020-12-10:
Ok you need to please wait until I sit with my Pi and see as to what is going wrong

aandroide, 2020-12-10:
I found out what causes the error
speaking in the other thread, it made me question whether my config.yaml file might have typos. I then downloaded the original config.yaml file without making any changes and when I asked "do you have a family" he obviously replied in English.
I therefore proceeded to change only the language in it-IT and the errors from before have returned.

shivasiddharth, 2020-12-10:
Attach your config file here

aandroide, 2020-12-10:
[config.txt](https://github.com/shivasiddharth/GassistPi/files/5675933/config.txt)

aandroide, 2020-12-10:
I also tried to use google cloud instead of gtts and it does the same thing, in en-US it works but in it-IT it doesn't work

shivasiddharth, 2020-12-10:
I am not fully well. I am doing whatever troubleshooting I can from my bed. Please wait until I get hands on with the Pi. I will try and see as to what's going wrong. I need 2-3 days time.

aandroide, 2020-12-10:
Don't worry, I understand. indeed I sincerely hope that you will recover soon. Meanwhile, I continue to do research on the case and keep you updated.

aandroide, 2020-12-10:
what I know for sure is that it is not a problem of the gtts library because this code works and reproduces it with both vlc and mpg123. it does not work with alsa and sox.

#!/usr/bin/env python
#This is different from AIY Kit's actions
#Copying and Pasting AIY Kit's actions commands will not work

from gtts import gTTS
import subprocess
tts = gTTS('questo e un messaggio di prova', lang='it')
tts.save('prova.mp3')
subprocess.run(["vlc", "prova.mp3"]

aandroide, 2020-12-10:
in action.py I modified these lines and now it works in Italian.
def say(words,sourcelang=None,destinationlang=None):
    if sourcelang!=None and destinationlang!=None:
        sayword=trans(words,destinationlang,sourcelang)
        gttssay(sayword,destinationlang,'Male')
    else:
        if sourcelang==None:
            sourcelanguage='it'

shivasiddharth, 2020-12-11:
I will fix the issue.

aandroide, 2020-12-11:
I found a new error re-enabling my data in the mqtt section in config.yaml that I copied from the old gassistpi backup
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/home/pi/GassistPi/src/main.py", line 624, in mqtt_start
    client.connect(configuration['MQTT']['IP'], 1883, 60)
  File "/home/pi/env/lib/python3.7/site-packages/paho/mqtt/client.py", line 839, in connect
    return self.reconnect()
  File "/home/pi/env/lib/python3.7/site-packages/paho/mqtt/client.py", line 962, in reconnect
    sock = socket.create_connection((self._host, self._port), source_address=(self._bind_address, 0))
  File "/usr/lib/python3.7/socket.py", line 727, in create_connection
    raise err
  File "/usr/lib/python3.7/socket.py", line 716, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

Connected to Adafruit IO!
Connected to Adafruit IO!  Listening for Pro changes...

aandroide, 2020-12-11:
what can cause the error? adafruit works fine while mqtt closes on startup

aandroide, 2020-12-13:
> I will fix the issue.

I solved the problem. The problem is caused by an error in the googletrans library. I updated the library to
this version: pip install googletrans==3.1.0a0
and the translator is back to working

aandroide, 2020-12-13:
> Ho trovato un nuovo errore riattivando i miei dati nella sezione mqtt in config.yaml che ho copiato dal vecchio backup gassistpi.
> Eccezione nel thread Thread-3:
> Traceback (la chiamata più recente per ultima):
> File "/ usr / lib / python3. 7 / threading.py ", riga 917, in _bootstrap_inner
> self.run ()
> File" /usr/lib/python3.7/threading.py ", riga 865, in esecuzione
> self._target (* self._args, ** self ._kwargs)
> File "/home/pi/GassistPi/src/main.py", riga 624, in mqtt_start
> client.connect (configurazione ['MQTT'] ['IP'], 1883, 60)
> File "/ home / pi /env/lib/python3.7/site-packages/paho/mqtt/client.py ", riga 839, in connect
> return self.reconnect ()
> File" /home/pi/env/lib/python3.7/site- packages / paho / mqtt / client.py ",linea 962, in riconnessione
> sock = socket.create_connection ((self._host, self._port), source_address = (self._bind_address, 0))
> File "/usr/lib/python3.7/socket.py", riga 727, in create_connection
> solleva err
> File "/usr/lib/python3.7/socket.py", riga 716, in
> create_connection sock.connect (sa)
> ConnectionRefusedError: [Errno 111] Connessione rifiutata
> 
> Collegato ad Adafruit IO!
> Collegato ad Adafruit IO! Ascolto di modifiche Pro ...

also fixed this error. i just forgot i was using a new sd card with new image and updated gassistpi and consequently i had to install mosquitto as a broker on the raspberry.

commit 9c0b3ecb (2020-12-14): Bug fixes

Fixes #1087  Fixes #1085  Fixes #1076