shivasiddharth/GassistPi #1089: The general volume of the raspberry cannot be adjusted when running Gassistpi automatically.

aandroide, 2020-12-16:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
latest version of December 15th.
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
pi@pi:~ $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000a91ab7c2
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@pi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@pi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage5
  
```

### Describe the bug:
```
https://drive.google.com/file/d/1RE0jw3EnWJDutgmf_h15VSPT9RobWw0g/view

I created a video where I show the error. pay no attention to the start-up sound I modified with the ac / dc song
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
no crash
```

### Paste the contents of the terminal below
```

no errors logged
```

aandroide, 2020-12-16:
pi@pi:~ $ pip list
Package           Version    
----------------- -----------
arandr            0.1.9      
asn1crypto        0.24.0     
automationhat     0.2.0      
blinker           1.4        
blinkt            0.1.2      
buttonshim        0.0.2      
Cap1xxx           0.1.3      
certifi           2018.8.24  
chardet           3.0.4      
Click             7.0        
colorama          0.3.7      
colorzero         1.1        
configparser      3.5.0b2    
cookies           2.2.1      
cryptography      2.6.1      
drumhat           0.1.0      
entrypoints       0.3        
enum34            1.1.6      
envirophat        1.0.0      
ExplorerHAT       0.4.2      
Flask             1.0.2      
fourletterphat    0.1.0      
funcsigs          1.0.2      
gpiozero          1.5.1      
gTTS              2.2.1      
gyp               0.1        
idna              2.6        
ipaddress         1.0.17     
itsdangerous      0.24       
Jinja2            2.10       
keyring           17.1.1     
keyrings.alt      3.1.1      
MarkupSafe        1.1.0      
mcpi              0.1.1      
microdotphat      0.2.1      
mock              2.0.0      
mote              0.0.4      
motephat          0.0.3      
numpy             1.16.2     
oauthlib          2.1.0      
olefile           0.46       
pantilthat        0.0.7      
pbr               4.2.0      
phatbeat          0.1.1      
pianohat          0.1.0      
picamera          1.13       
picraft           1.0        
piglow            1.2.5      
pigpio            1.44       
Pillow            5.4.1      
pip               18.1       
pycairo           1.16.2     
pycrypto          2.6.1      
pyflakes          2.0.0      
pygame            1.9.4.post1
PyGObject         3.30.4     
pyinotify         0.9.6      
PyJWT             1.7.0      
pyOpenSSL         19.0.0     
pyserial          3.4        
pyxdg             0.25       
rainbowhat        0.1.0      
requests          2.21.0     
requests-oauthlib 1.0.0      
responses         0.9.0      
RPi.GPIO          0.7.0      
RTIMULib          7.2.1      
scrollphat        0.0.7      
scrollphathd      1.2.1      
SecretStorage     2.3.1      
sense-emu         1.1        
sense-hat         2.2.0      
setuptools        40.8.0     
simplejson        3.16.0     
six               1.12.0     
skywriter         0.0.7      
sn3218            1.2.7      
spidev            3.4        
touchphat         0.0.1      
twython           3.7.0      
unicornhathd      0.0.4      
urllib3           1.24.1     
Werkzeug          0.14.1     
wheel             0.32.3     

driver for audio setup=AIY-HAT
image system: 2020-12-02-raspios-buster-armhf-full
I have this: https://aiyprojects.withgoogle.com/voice-v1/
[image]
[image]
[image]
[image]
[image]

aandroide, 2020-12-16:
instead everything works fine if you run main.py manually

aandroide, 2020-12-16:
[image]

I just discovered that the audio can be adjusted via internal analog stereo audio as shown in the picture when starting the assistant in automatic mode it does not stay synchronized with internal audio.

shivasiddharth, 2020-12-18:
Open alsamixer from the terminal and can you check if the volume changes if you change the volume there ?

aandroide, 2020-12-18:
Yes

shivasiddharth, 2020-12-18:
Do:
`sudo nano /usr/share/pulseaudio/alsa-mixer/paths/analog-output.conf.common`   

You should find a line that says `[Element PCM]`. Above that add

```
[Element Master]
switch = mute
volume = ignore
```

Do a restart and see what happens. As mentioned before, there is no guarantee that I can offer a fix for this issue.

aandroide, 2020-12-18:
with this change I can't even adjust the volume from the internal analog audio anymore

aandroide, 2020-12-18:
[image]
[image]
when i run main.py manually, the screens change i don't understand what pushes to use analog audio only when running sudo systemctl start gassistpi.service

aandroide, 2020-12-18:
it could hit the fact that as soon as I installed the new image and updated gassistpi, realizing that the microphone of my aiy-hat kit was not recognized, I had to run:  
sudo nano /etc/pulse/default.pa
and I had to add this line:
load-module module-alsa-source device=hw:0,0

shivasiddharth, 2020-12-19:
This was not at all required. Do you have something in /etc/asound.conf and ./.asoundrc or it is empty ?

aandroide, 2020-12-19:
/etc/asound.conf :
pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card 0
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card 0
}

and sudo nano ./.asoundrc this file is not present

shivasiddharth, 2020-12-19:
Do you have portaudio and alsa folder in /home/pi folder ?
Run
aplay --version

aandroide, 2020-12-19:
Aplay version 1.1.8 and in home/pi do not present the portaudio and alsa

shivasiddharth, 2020-12-19:
If you followed the AIY instructions for Dec build, you will not have anything inside asound.conf file.

> sudo nano /etc/pulse/default.pa
> and I had to add this line:
> load-module module-alsa-source device=hw:0,0

Is also not required.

aandroide, 2020-12-19:
I format everything and start over and update you as soon as I'm done

aandroide, 2020-12-19:
hi i formatted and installed everything from 0. the problem is that the microphone does not listen to me

aandroide, 2020-12-19:
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

if I run from the terminal:
arecord -f S32_LE -r 48000 --device="hw:0,0" -c 2 -d 10 1; sleep 1; aplay 1

I record feels a little low but still works.
I await your instructions before editing files that I don't have to.

aandroide, 2020-12-19:
[image]

maybe something is missing in the driver configuration for AIY-HAT

aandroide, 2020-12-19:
in the guide to install gassistpi for those who updated with the December image, I saw that you removed these lines:
cd /home/${USER}/
sudo chmod +x ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh
sudo ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh
 is this correct?

shivasiddharth, 2020-12-19:
The ALSA config installer are no longer required. In the volume control on the taskbar, you should be able to select your input and output device. Are those set to the AIY HAT ?

shivasiddharth, 2020-12-19:
Install pavucontrol and in the pavucontrol, check the audio levels.

aandroide, 2020-12-19:
>  Are those set to the AIY HAT ?
[image]
yes.

i installed pavucontrol and i don't see the microphone

aandroide, 2020-12-19:
> /etc/asound.conf
with installation from 0, all 2 files are non-existent

aandroide, 2020-12-19:
> This was not at all required. Do you have something in /etc/asound.conf and ./.asoundrc or it is empty ?

with installation from 0, all 2 files are non-existent

aandroide, 2020-12-19:
also i just found that the audiosetup file containing the word AIY-HAT, is not created in home / pi and this won't make the button light up when activating the google assistant. and the mediavolume.json is also missing.

shivasiddharth, 2020-12-20:
In the terminal, open alsamixer. There the card and device should be pulseaudio, Press F5, you will get the speaker and mic volume. I just tested my AIY HAT and it works.

aandroide, 2020-12-20:
I tried and it doesn't work for me. did you start from a clean installation or did you update the old gassistpi? also have you noticed that the audiosetup file is not created and therefore the hat does not light up? you can publish your default.pa for comparison. thanks

shivasiddharth, 2020-12-20:
I will get that file, before that can you confirm if speaker-test works for you ? Can you check if the Mic's meter on the pavucontrol moves if you make some noise on the mic ?

aandroide, 2020-12-20:
Can you check if the Mic's meter on the pavucontrol moves if you make some noise on the mic ? don't work
it only worked for me when I added hw0,0 in default.pa but you told me it is not necessary and so I have not added it anymore

aandroide, 2020-12-20:
my alsamixer have https://github.com/shivasiddharth/GassistPi/issues/1089#issuecomment-748495030

shivasiddharth, 2020-12-20:
In the alsamixer, can you choose the pulseaudio card. Press F6 and select Pulseaudio. AFter that can you post the screenshot ?

aandroide, 2020-12-20:
[image]

shivasiddharth, 2020-12-20:
In that window can you press F5 ? That will show the Mic also.

aandroide, 2020-12-20:
[image]

shivasiddharth, 2020-12-20:
[default.txt](https://github.com/shivasiddharth/GassistPi/files/5720733/default.txt)

Check the default file I have not modified anything. Can you set the mic volume to maximum here in the alsamixer ? If the arecord works, then the assistant should also work.

aandroide, 2020-12-20:
the default.pa file is identical to mine, I compared it with the bcompare program. I turned the mic volume up to maximum. please remind me how to record a test with arecord from terminal

aandroide, 2020-12-20:
[image]

shivasiddharth, 2020-12-20:
Can you check if this works

`arecord -d 10 -r 48000 -f S32_LE -c2 /home/pi/mic-test.wav`

aandroide, 2020-12-20:
don't work at
created the silent file

aandroide, 2020-12-20:
it only works if I add --device="hw: 0,0" to your arecord string

shivasiddharth, 2020-12-20:
Does that record the sound you produced. ?

aandroide, 2020-12-20:
yes

aandroide, 2020-12-20:
arecord -d 10 -r 48000 -f S32_LE -c2 --device="hw:0,0" /home/pi/mic-test.wav

aandroide, 2020-12-20:
the problem is that if you don't specify it, it doesn't recognize the device, according to my theory.

shivasiddharth, 2020-12-20:
Can you manually start the assistant now and check ?

aandroide, 2020-12-20:
tried, it starts but does not listen to me

aandroide, 2020-12-20:
i think you have an obligation to declare it in default.pa only the way it works

shivasiddharth, 2020-12-20:
This thread is getting long. Can we move to gitter ?

aandroide, 2020-12-20:
ok

closed 2021-01-07 by aandroide