shivasiddharth/GassistPi #1090: API key not valid. Please pass a valid API key

khush99, 2020-12-17:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```

```
Paste your Board and OS details here.   
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03114
Serial		: 10000000a3bd3534
Model		: Raspberry Pi 4 Model B Rev 1.4

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage4

```

### Describe the bug:
```
I have enabled the youtube api on google console and create an api in credential ,restriction is none and entered the api in conf file but ---------------
whenever i say" ok google play vande matarm from youtube " or any song name it  encounters error and says please chceck the logs
Here's the output:----------------------------------
stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
youtube on youtub
<HttpError 400 when requesting https://www.googleapis.com/discovery/v1/apis/youtube/v3/rest?key=%3D+AIzaSyBWECUdMuRNMtOLODVjq1SS0x1K-MttJow returned "API key not valid. Please pass a valid API key.". Details: "[{'@type': 'type.googleapis.com/google.rpc.ErrorInfo', 'reason': 'API_KEY_INVALID', 'domain': 'googleapis.com', 'metadata': {'service': 'youtube.googleapis.com'}}]">
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

```
---------------------------------------------------------------
### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes/No
Yes
### 2. Is the issue related to assistant installation ?
Yes/No
No
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No
no,manually start
### 4. Is the assistant service restarting automatically?
Yes/No
NO
### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2020-12-17:
It is what it says. I am not sure as to how I can be of help if the key is invalid. Please double check the key that you have entered.

khush99, 2020-12-17:
1. what step should i follow to remove this error.In how many files should i give the api key rather than config key.
please help me 
2. I am facing issue that whenever i try to import python3 supporting library in action.py file it says import module error

khush99, 2020-12-17:
sorry i was referring to python2 library ,not python3

shivasiddharth, 2020-12-17:
There is no way you are using python2 unless you have modified something. By default, GassistPi is in python3 environment. API key should be listed only in the config file. Paste the config file, i will take a look, but its not going to help much. Please double or triple check the API key creating process and the API key itself. Thats all I can say.

khush99, 2020-12-17:
how to import adafuit_dht DHT22 sensor library in action.py file,  when i import adafruit_dht in action.py file and manually start gassistpi using command /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id raspberry-pi-g-d6aa8 --device_model_id raspberry-pi-g-d6aa8-raspberrypi4-t8y7ht
 ,it shows------
import adafruit_dht
ModuleNotFoundError: No module named 'adafruit_dht'

while 'import adafruit_dht' works in python3

shivasiddharth, 2020-12-17:
You would have installed the python2 package and not the python3 version. This place is not for discussing your personal customisations. Please use gitter for that.

shivasiddharth, 2020-12-17:
First get some clarity. One place you are saying that importing python3 works and in another place python2 works.

khush99, 2020-12-17:
bro please help me regrading youtube api key is invalid, 
and i have sorted out other python 2 python 3 issue using activating env

khush99, 2020-12-17:
please help me regrading youtube key invalid issue

shivasiddharth, 2020-12-18:
I asked you to double check the API key from the cloud console and the API key that you have entered in the config file have you done that ?

Also I asked you to paste the config file here.

khush99, 2020-12-18:
[image]
[image]

I am attaching youtube api screenshot for your refrence

#Config file for declaring custom actions'parameters. The parameters have been listed here in alphabetical order (A-Z).

#Adafruit_IO. Sending commands from Genuine Google Home/Assistant Devices to control Domoticz Tasmota etc.
ADAFRUIT_IO:
  ADAFRUIT_IO_CONTROL: 'Disabled'
  ADAFRUIT_IO_USERNAME: 'ENTER YOUR ADAFRUIT_IO USERNAME HERE'
  ADAFRUIT_IO_KEY: 'ENTER YOUR ADAFRUIT_IO KEY HERE'
  FEEDNAME: 'ENTER YOUR ADAFRUIT_IO FEEDNAME THAT YOU WANT TO MONITOR HERE'

#Custom conversation. Add your queries and expected assistant response. Do not add symbols or special charecters.
Conversation:
  Conversation_Control: 'Enabled'
  question:
    1:
      - 'Who are you'
    2:
      - 'Do you have a family'
  answer:
    1:
      - 'I am not the one you expect to be'
      - 'Who do you think I am'
      - 'What bothers you'
    2:
      - 'Nope i am forever alone'
      - 'The whole world is my family'

#Send sms with clickatell
Clickatell:
  Clickatell_Control: 'Disabled'
  Clickatell_API: 'ENTER_YOUR_CLICKATELL_API'
  Name:
    - 'Bob'
    - 'Adam'
  Reciever:
    - '+48123456789'
    - '+46123456789'

# Deezer user credentials
Deezer:
  Deezer_Control: 'Enabled'
  User_id: '2247762504'

# DIY-HUE declarations
DIYHUE:
  DIYHUE_Control: 'Enabled'

# Domoticz declarations
Domoticz:
  # Set this to 'Enabled' to control devices registered to Domoticz server.
  Domoticz_Control: 'Disabled'
  Server_IP:
    # If you are running the server on the Pi itself, then change the ip to 'localhost'
    - '192.168.0.10'
  Server_port:
    - '443'
  Devices:
    #The number of names and number of ids should be equal
    #List the device names that you want to control
    Name:
      - 'Lamp1'
      - 'Lamp2'
      - 'Lamp3'
    #Assign the device ids in the same order as the devices listed above
    Id:
      - '1'
[... 208 lines cut ...]
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Enabled'
  #Set below to 'Picovoice' or 'Snowboy' as per your choice.
  Wakeword_Engine: 'Snowboy'
  Snowboy_wakeword_models:
    - '/home/pi/GassistPi/src/resources/snowboy_models/alexa.umdl'
    - '/home/pi/GassistPi/src/resources/snowboy_models/snowboy.umdl'
  Picovoice_wakeword_models:
    - '/home/pi/GassistPi/src/resources/picovoice_models/americano_raspberry-pi.ppn'

#Wemo/Emulated Wemo devices
Wemo:
  Wemo_Control: 'Disabled'
  Wemo_Devices:
    IP_Address:
      - 'Enter wemo device 1 ip here'
      - 'Enter wemo device 2 ip here'
    Device_Names:
      - 'Enter wemo device 1 name here'
      - 'Enter wemo device 1 name here'

#Youtube music streaming
YouTube:
  YouTube_Control: 'Enabled'

shivasiddharth, 2020-12-18:
> #API Key for YouTube and Kickstarter Search Engine
> 
> Google_cloud_api_key: = "AIzaSyBWECUdMuRNMtOLODVjq1SS0x1K-MttJow"  

I asked you to double check, triple check. Why do you have a "=" there in that line ? 

Check the original config file. https://github.com/shivasiddharth/GassistPi/blob/cffe4a7fbb67fc1782f4d791db43f3690e01afdd/src/config.yaml#L99

closed 2020-12-18 by shivasiddharth