shivasiddharth/GassistPi #1097: Stopping vlc...

JSa1987, 2020-12-27:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 0000000027ca57fb
Model		: Raspberry Pi 3 Model B Rev 1.2

cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage4
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```
After having issues with Gassist keep restarting on Raspberry Pi OS Lite as advised I started over again with a fresh install of Raspberry Pi OS Desktop.
Installation was completed without issues. When I try to start the assistant manually it plays the initial music. The process the gets stuck in a cycle of:
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes
### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'raspberry-pi-6b03d' --device-model-id 'raspberry-pi-6b03d-josto-amplifier-jwov8b'
```

### Paste the contents of the terminal below
```
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-12-27 16:06:26,324 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Initializing GPIOs 5 and 6 for assistant activity indication
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
stopping vlc
Stopped
```

shivasiddharth, 2020-12-27:
Check the way you have wired the pushbutton. Seems like the pushbutton is always high.

JSa1987, 2020-12-27:
Yes, you were right. GPIO23 on my HAT is connected to an LED, that I guess was interfering with the pushbutton logic.
I have now edit the config file to point only at unused GPIOs. However, when I manually start the assistant it still gets stuck. It plays the initial music and the gets stuck after initializing the GPIOs:

`pi@Apollo:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'raspberry-pi-6b03d' --device-model-id 'raspberry-pi-6b03d-josto-amplifier-jwov8b'
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2020-12-27 17:05:08,934 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
Initializing GPIOs 17 and 5 for assistant activity indication`

aandroide, 2020-12-27:
@JSa1987 hello do you use the AIY kit? hello do you use the AIY kit? otherwise what HAT do you use?

JSa1987, 2021-01-04:
Hello,

I'm using a IQaudio Codec Zero HAT (https://www.raspberrypi.org/blog/iqaudio-is-now-raspberry-pi/). 
If I start with the fresh installation of Raspberry Pi OS with desktop I get the issue above. If I start with a fresh install of Raspberry Pi OS Lite the assistant will start up and work, but it then robots every few minutes.

aandroide, 2021-01-04:
[image]
what do you learn from this pattern?

JSa1987, 2021-01-04:
I have defined the GPIOs in the config.yaml file as follow:

Gpios:
  AIY_indicator:
    - 26
  assistant_indicators:
    - 23
    - 24
  ir:
    - 17
  picontrol:
    - 05
    - 06
    - 13
  pushbutton_trigger:
    - 16
  servo:
    - 12
  stopbutton_music_AIY_pushbutton:
    - 27

Below is the list of GPIO used by this hat:

aandroide, 2021-01-04:
take my curiosity away if you physically press that button, does the green LED turn on?

JSa1987, 2021-01-04:
Hi,

I just tried. When is push the button on the HAT the green LED does not turn on. Based on this I re-assigned the GPIOs in the config.yaml file as follow.
Gpios:
  AIY_indicator:
    - 26
  assistant_indicators:
    - 23
    - 24
  ir:
    - 17
  picontrol:
    - 05
    - 06
    - 13
  pushbutton_trigger:
    - 16
  servo:
    - 12
  stopbutton_music_AIY_pushbutton:
    - 25

Now when I call the assistant the green LED turns on, and when the assistant is speaking the red LED turns on.

I'm able to start the assistant, but this then crashes after a few minutes. Below is he output on the terminal when the assistant crashes.

`python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Fatal Python error: Aborted

Thread 0x60efe460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 504 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x634fd460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 277 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x69cff460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/threading.py", line 552 in wait
  File "/usr/lib/python3.7/threading.py", line 1164 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x76f17ad0 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 300 in wait
  File "/usr/lib/python3.7/queue.py", line 179 in get
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 398 in next
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/event.py", line 393 in __next__
  File "/home/pi/GassistPi/src/main.py", line 1110 in main
  File "/home/pi/GassistPi/src/main.py", line 1121 in <module>
Aborted`

aandroide, 2021-01-04:
ok we have excluded that the leds are not connected to the button as a state but at the same time you have just discovered that your gpio 27 is always high so there is a small change to be made in action.py to reverse the states of the stop button high and low for the other problem he had told me just today @ I think we must necessarily update to pulseaudio but we ask for confirmation.

JSa1987, 2021-01-04:
Following some of the steps on #1014 I managed to get it to work starting from a fresh install of Raspberry Pi OS Lite. The steps I followed are:

1) Downgrade ALSA
'''
# wget "ftp://ftp.alsa-project.org/pub/lib/alsa-lib-1.1.3.tar.bz2"
# tar -xf alsa-lib-1.1.3.tar.bz2
# cd alsa-lib-1.1.3
# ./configure && make
# sudo make install
# sudo apt-get update
# sudo apt-get update --fix-missing
# sudo apt-get install libncursesw5-dev -y
# wget ftp://ftp.alsa-project.org/pub/utils/alsa-utils-1.1.3.tar.bz2
# tar xvjf alsa-utils-1.1.3.tar.bz2
# cd alsa-utils-1.1.3
# ./configure --disable-alsaconf --disable-bat --disable-xmlto --disable-rst2man --with-curses=ncursesw
# make clean
# sudo make install
'''

2) Install Pulseaudio
'''
# sudo apt-get install pulseaudio
# sudo apt-get purge portaudio19-dev
# sudo reboot
# wget http://www.portaudio.com/archives/pa_stable_v190600_20161030.tgz
# tar zxf pa_stable_v190600_20161030.tgz
# pushd portaudio
# ./configure --without-jack
# make
# sudo make install
# sudo reboot
'''

After this the assistant was not crashing anymore. However, the microphone of the HAT was not working. To get the microphone to work I had to make some changes to the /etc/pulse/default.pa file.

3) Add the following lines to /etc/pulse/default.pa
'''
load-module module-alsa-source device=hw:0,0
load-module module-remap-source master=alsa_input.hw_0_0 master_channel_map=front-left,front-right channels=2 channel_map=mono,mono
'''

At this point the assistant was working as expected. However, the audio was stuttering really bad when listening to radio stations through the assistant. Based on what I found on the internet I made some adjustments to the Pulaseaudio configuration files. This helped however, the audio is still stuttering once in a while.

4) Edit /etc/pulse/daemon.conf as follow
'''
high-priority = no
nice-level = -1
realtime-scheduling = yes
realtime-priority = 5
flat-volumes = no
resample-method = speex-float-1
default-sample-rate = 44100
default-sample-format = s16le
default-sample-channels = 2
default-fragments = 2
default-fragment-size-msec = 125
'''

5) And finally added the following line to /etc/pulse/default.pa
'''
- load-module module-udev-detect tsched=0
'''

In the past I had the the assistant running on Raspbian Stretch with just ALSA, no Pulseaudio and it worked well. Seems that with Raspberry Pi Buster and the later updated to the assistant Pulseaudio is now needed. However, Pulseaudio is causing some stuttering. Problem I never had when running only ALSA without Pulseaudio.

aandroide, 2021-01-04:
it had happened to me too and I solved it by changing this:
load-module module-udev-detect tsched=0
to:
load-module module-udev-detect
strangely it worked for me