shivasiddharth/GassistPi #1100: speaker-test doesn't output with aiy hat v2

bbrendon, 2021-01-03:
I have the pi zero with aiy hat v2. Nothing comes out when you run speaker-test

### Which branch of GassistPi are you using ?
master

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue

pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 0000000029602267
Model           : Raspberry Pi Zero W Rev 1.1
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2
pi@raspberrypi:~ $

```

### Describe the bug:
```
I have the pi zero with aiy hat v2. Nothing comes out when you run speaker-test

```

Install process was ... I didn't include all the terminal output. No errors in the driver setup.

```
ssh pi@10.5.0.153
sudo apt-get install git
git clone https://github.com/shivasiddharth/GassistPi

sudo apt-get update
cd /home/${USER}/
sudo chmod +x ./GassistPi/audio-drivers/AIY-HAT/scripts/configure-driver.sh
sudo ./GassistPi/audio-drivers/AIY-HAT/scripts/configure-driver.sh
sudo reboot
ssh pi@10.5.0.153

speaker-test

pi@raspberrypi:~ $ speaker-test

speaker-test 1.1.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 64 to 65536
Period size range from 32 to 32768
Using max buffer size 65536
Periods = 4
was set period_size = 16384
was set buffer_size = 65536
 0 - Front Left
Time per period = 1.380892
 0 - Front Left
Time per period = 2.729074
 0 - Front Left
Time per period = 16.271809
 0 - Front Left
Time per period = 2.729099
 0 - Front Left
Time per period = 2.729077
 0 - Front Left
Time per period = 2.729040
 0 - Front Left
Time per period = 2.729271
 0 - Front Left
^CWrite error: -4,Interrupted system call
xrun_recovery failed: -4,Interrupted system call
Transfer failed: Interrupted system call
pi@raspberrypi:~ $ ^C
```

closed 2021-01-03 by close-issue-app[bot]

aandroide, 2021-01-03:
hi check if there is this file in your system: sudo nano /lib/udev/rules.d/91-pulseaudio-rpi.rules

aandroide, 2021-01-03:
also if your kit is the one with the camera, you need to install specific drivers that are not present in this project but you can find them here: https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md # install-aiy-software-on-an-existing-raspbian-system

bbrendon, 2021-01-03:
@aandroide   I dont have a kit with a camera.

```
$ ls -l /lib/udev/rules.d/91-pulseaudio-rpi.rules
ls: cannot access '/lib/udev/rules.d/91-pulseaudio-rpi.rules': No such file or directory
```

Thanks.

aandroide, 2021-01-03:
do you have the december image installed?

aandroide, 2021-01-03:
[image]
do you see this if you right click on the volume icon?

aandroide, 2021-01-03:
if you run: aplay -l and arecord -l
what does it show you?

bbrendon, 2021-01-03:
I don't have a gui. just ssh.

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $
```

aandroide, 2021-01-03:
ok try this:
sudo nano /etc/pulse/default.pa 

add these 2 lines:
load-module module-alsa-source device = hw: 0,0 "
load-module module-alsa-sink
salva e riavvia

aandroide, 2021-01-03:
[image]

bbrendon, 2021-01-03:
I don't have a `/etc/pulse/default.pa`

```
$ ls -l /etc/pulse
ls: cannot access '/etc/pulse': No such file or directory
pi@raspberrypi:~ $
```

I have a feeling something is wrong with the installer. ? Or maybe it's because I installed the non bloat version (Raspberry Pi OS Lite)  and the installer doesn't understand?

aandroide, 2021-01-03:
so you didn't install the dec raspbian image from here https://www.raspberrypi.org/software/operating-systems/ ?

bbrendon, 2021-01-03:
Yes. I did install from that page. There a _three_ options on that page if you read it closely.

bbrendon, 2021-01-03:
Now that I read it, there are a lot more than three. But looks like three valid options.

aandroide, 2021-01-03:
yes now I understand. maybe you don't have pulseaudio installed you can run pulseaudio --version

bbrendon, 2021-01-03:
```
pi@raspberrypi:~ $ pulseaudio --version
-bash: pulseaudio: command not found
pi@raspberrypi:~ $ sudo pulseaudio --version
sudo: pulseaudio: command not found
pi@raspberrypi:~ $
```

bbrendon, 2021-01-03:
can you get on gitter? I'm running sudo apt install  pulseaudio

aandroide, 2021-01-03:
as I suspected you have installed the image without programs maybe it's good because you still use alsa you can try to install the drivers for alsa with

aandroide, 2021-01-03:
don't install pulseaudio yet let's do a little test first

aandroide, 2021-01-03:
cd /home/${USER}/
sudo chmod +x ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh
sudo ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh
speaker-test

bbrendon, 2021-01-03:
already ran this... :(.

I'll uninstall all of that.

```
pi@raspberrypi:~ $ sudo apt install pulseaudio
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dbus-user-session fontconfig fontconfig-config fonts-dejavu-core libaom0 libasound2-plugins libasyncns0 libavcodec58 libavresample4
  libavutil56 libcairo2 libcodec2-0.8.1 libcroco3 libdatrie1 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2
  libflac8 libfontconfig1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri libglapi-mesa
  libglvnd0 libglx-mesa0 libglx0 libgraphite2-3 libgsm1 libharfbuzz0b libice6 libjack-jackd2-0 libjbig0 libllvm9 libltdl7 libmp3lame0
  libogg0 libopenjp2-7 libopus0 liborc-0.4-0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpixman-1-0 libpulse0 libpulsedsp
  librsvg2-2 librsvg2-common libsensors-config libsensors5 libshine3 libsm6 libsnappy1v5 libsndfile1 libsoxr0 libspeex1 libspeexdsp1
  libswresample3 libtdb1 libthai-data libthai0 libtheora0 libtiff5 libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1
  libvdpau1 libvorbis0a libvorbisenc2 libvpx5 libwavpack1 libwebp6 libwebpmux3 libwebrtc-audio-processing1 libx11-xcb1 libx264-155
  libx265-165 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxdamage1 libxfixes3 libxi6 libxrender1 libxshmfence1 libxtst6 libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers
  mesa-vdpau-drivers pulseaudio-utils rtkit va-driver-all vdpau-driver-all x11-common
Suggested packages:
  jackd2 opus-tools librsvg2-bin lm-sensors speex pavumeter pavucontrol paman paprefs
The following NEW packages will be installed:
  dbus-user-session fontconfig fontconfig-config fonts-dejavu-core libaom0 libasound2-plugins libasyncns0 libavcodec58 libavresample4
  libavutil56 libcairo2 libcodec2-0.8.1 libcroco3 libdatrie1 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2
  libflac8 libfontconfig1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri libglapi-mesa
  libglvnd0 libglx-mesa0 libglx0 libgraphite2-3 libgsm1 libharfbuzz0b libice6 libjack-jackd2-0 libjbig0 libllvm9 libltdl7 libmp3lame0
  libogg0 libopenjp2-7 libopus0 liborc-0.4-0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpixman-1-0 libpulse0 libpulsedsp
  librsvg2-2 librsvg2-common libsensors-config libsensors5 libshine3 libsm6 libsnappy1v5 libsndfile1 libsoxr0 libspeex1 libspeexdsp1
  libswresample3 libtdb1 libthai-data libthai0 libtheora0 libtiff5 libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1
  libvdpau1 libvorbis0a libvorbisenc2 libvpx5 libwavpack1 libwebp6 libwebpmux3 libwebrtc-audio-processing1 libx11-xcb1 libx264-155
  libx265-165 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0
  libxdamage1 libxfixes3 libxi6 libxrender1 libxshmfence1 libxtst6 libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers
  mesa-vdpau-drivers pulseaudio pulseaudio-utils rtkit va-driver-all vdpau-driver-all x11-common
0 upgraded, 108 newly installed, 0 to remove and 28 not upgraded.
Need to get 49.2 MB of archives.
After this operation, 676 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```

aandroide, 2021-01-03:
yes

bbrendon, 2021-01-03:
ok. ran...

```
sudo dpkg -P  dbus-user-session fontconfig fontconfig-config fonts-dejavu-core libaom0 libasound2-plugins libasyncns0 libavcodec58 libavresample4 \
  libavutil56 libcairo2 libcodec2-0.8.1 libcroco3 libdatrie1 libdrm-amdgpu1 libdrm-common libdrm-nouveau2 libdrm-radeon1 libdrm2 \
  libflac8 libfontconfig1 libgdk-pixbuf2.0-0 libgdk-pixbuf2.0-bin libgdk-pixbuf2.0-common libgl1 libgl1-mesa-dri libglapi-mesa \
  libglvnd0 libglx-mesa0 libglx0 libgraphite2-3 libgsm1 libharfbuzz0b libice6 libjack-jackd2-0 libjbig0 libllvm9 libltdl7 libmp3lame0 \
  libogg0 libopenjp2-7 libopus0 liborc-0.4-0 libpango-1.0-0 libpangocairo-1.0-0 libpangoft2-1.0-0 libpixman-1-0 libpulse0 libpulsedsp \
  librsvg2-2 librsvg2-common libsensors-config libsensors5 libshine3 libsm6 libsnappy1v5 libsndfile1 libsoxr0 libspeex1 libspeexdsp1 \
  libswresample3 libtdb1 libthai-data libthai0 libtheora0 libtiff5 libtwolame0 libva-drm2 libva-x11-2 libva2 libvdpau-va-gl1 \
  libvdpau1 libvorbis0a libvorbisenc2 libvpx5 libwavpack1 libwebp6 libwebpmux3 libwebrtc-audio-processing1 libx11-xcb1 libx264-155 \
  libx265-165 libxcb-dri2-0 libxcb-dri3-0 libxcb-glx0 libxcb-present0 libxcb-render0 libxcb-shm0 libxcb-sync1 libxcb-xfixes0 \
  libxdamage1 libxfixes3 libxi6 libxrender1 libxshmfence1 libxtst6 libxvidcore4 libxxf86vm1 libzvbi-common libzvbi0 mesa-va-drivers \
  mesa-vdpau-drivers pulseaudio pulseaudio-utils rtkit va-driver-all vdpau-driver-all x11-common

...
Then...

cd /home/${USER}/
sudo chmod +x ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh
sudo ./GassistPi/audio-drivers/AIY-HAT/scripts/install-alsa-config.sh

reboot < added this :)

pi@raspberrypi:~ $ speaker-test

speaker-test 1.1.8

Playback device is default
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Rate set to 48000Hz (requested 48000Hz)
Buffer size range from 2048 to 16384
Period size range from 1024 to 1024
Using max buffer size 16384
Periods = 4
was set period_size = 1024
was set buffer_size = 16384
 0 - Front Left
Time per period = 2.661900
 0 - Front Left
Time per period = 15.995314
 0 - Front Left
Time per period = 2.985062
 0 - Front Left
Time per period = 2.985095
 0 - Front Left
^CTime per period = 1.130714
^Cpi@raspberrypi:~ $

  ```

No sounds :(

Does the previous AIY-HAT sound config I ran at the very beginning conflict with this?

aandroide, 2021-01-03:
it only conflicts if you had pulseaudio installed. you can run alsamixer and check that you have the volume high and then pressif5 to see if you also see the mic

bbrendon, 2021-01-03:
Check and check. any other ideaS?

[image]

[image]

aandroide, 2021-01-03:
you can post what you see in sudo nano /boot/config.txt

bbrendon, 2021-01-03:
```
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment this if your display has a black border of unused pixels visible
# and your display can output without overscan
#disable_overscan=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
#dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard
```

aandroide, 2021-01-03:
ok these configurations are alright. I don't know if this error could depend on your image without programs, we ask @shivasiddharth

aandroide, 2021-01-03:
one last question have you already installed gassistpi in the past or is it the first time on this kit?

bbrendon, 2021-01-03:
I've tried getting it to work on this hardware about a month ago or so and same results. That hardware works fine if I install the google aiy OS.

Ok. Thanks for trying.  unfortunately my understanding of linux sound cards is very bad.

aandroide, 2021-01-03:
I have the aiy v1 kit and with these configurations everything works fine. I would like to make sure your py zero does not require an additional audio driver for the v2 hat. maybe @shivasiddharth  he knows

aandroide, 2021-01-04:
I found this looks like your kit requires drivers other than v1: https://github.com/opencardev/aiy-voicebonnet-soundcard-dkms-driver

aandroide, 2021-01-04:
however we await the opinion of @shivasiddharth

shivasiddharth, 2021-01-04:
Install the driver for the AIY-HAT Bonnet version using
`sudo apt-get install -y aiy-voicebonnet-soundcard-dkms`

And then give the speaker test a try. If this gives out some errors, let me know. There is an ongoing driver issue that has not yet been closed https://github.com/google/aiyprojects-raspbian/issues/697 Not sure, if this is affecting.

Just a word of caution. The master branch is not optimized for Pi Zero and some features will be missing. If you just need the custom wakeword support, use the Wakeword Branch.

bbrendon, 2021-01-04:
I found this which explains how to get the repo installed.
https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

I installed `aiy-voicebonnet-soundcard-dkms` but `speaker-test` still does not output anything.

The link has something about installing pulse audio. Should I try that?

shivasiddharth, 2021-01-04:
Do you already have the puls audio package installed ? Skipe the pulseaudio part there. All you need to do is
 sudo apt-get install pulseaudio
It could be due to the ongoing issue that I already mentioned.

bbrendon, 2021-01-08:
@shivasiddharth  I installed pulse-audio and speaker-test still does not work.

```
$ cat /lib/udev/rules.d/91-pulseaudio-rpi.rules
SUBSYSTEM!="sound*", GOTO="end"
ACTION!="change", GOTO="end"
KERNEL!="card*", GOTO="end"
ENV{SOUND_FORM_FACTOR}!="internal", GOTO="end"

ATTRS{id}=="b1", ENV{PULSE_PROFILE_SET}="rpi-hdmi.conf", GOTO="end"
ATTRS{id}=="b2", ENV{PULSE_PROFILE_SET}="rpi-hdmi.conf", GOTO="end"
ENV{PULSE_PROFILE_SET}="rpi-analog.conf"

LABEL="end"

 $ cat  /etc/pulse/default.pa | grep load |grep -v ^#
load-module module-device-restore
load-module module-stream-restore restore_device=false
load-module module-card-restore
load-module module-augment-properties
load-module module-switch-on-port-available
load-module module-udev-detect tsched=0
load-module module-detect
load-module module-jackdbus-detect channels=2
load-module module-bluetooth-policy
load-module module-bluetooth-discover autodetect_mtu=yes
load-module module-esound-protocol-unix
load-module module-native-protocol-unix
load-module module-gsettings
load-module module-default-device-restore
load-module module-rescue-streams
load-module module-always-sink
load-module module-intended-roles
load-module module-suspend-on-idle
load-module module-console-kit
load-module module-systemd-login
load-module module-position-event-sounds
load-module module-role-cork
load-module module-filter-heuristics
load-module module-filter-apply

```

aandroide, 2021-01-08:
Sudo rm /lib/udev/rules.d/91-pulseaudio-rpi.rules.  
And reboot and test

bbrendon, 2021-01-08:
deleting `/lib/udev/rules.d/91-pulseaudio-rpi.rules` and rebooting didn't do it.

aandroide, 2021-01-08:
unfortunately it is a problem with those drivers and there is no solution currently. the only thing you can do is connect a usb microphone to your raspberry and as an audio output or use a tv or headphone jack. or if you have a radio with aux input you could connect them.

bbrendon, 2021-01-09:
But ...but ...it works with the google raspberry OS :)

shivasiddharth, 2021-01-09:
I don't have the version 2 HAT to test. Like I mentioned before, there is an ongoing issue of version 2 HAT not working with Raspbian. I don't know if it's a kernel or driver issue.

Does the card appear under the output and input devices in the taskbar volume icon ? Also check if the volume is set to maximum. From a terminal run ```alsamixer``` and set the volume to maximum.

bbrendon, 2021-01-09:
I checked alsamixer. Same as the screenshots I posted earlier. I don't have a gui.

```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@raspberrypi:~ $

```

shivasiddharth, 2021-01-09:
You should have been using the desktop version of the Raspbian Image. Anyways can you post the output of ```dmesg | grep rt5645``` ?

bbrendon, 2021-01-09:
```
pi@raspberrypi:~ $ dmesg | grep rt5645
[   25.632105] rt5645 1-001a: 1-001a supply avdd not found, using dummy regulator
[   25.632338] rt5645 1-001a: 1-001a supply cpvdd not found, using dummy regulator
pi@raspberrypi:~ $
```

bbrendon, 2021-01-11:
@shivasiddharth  the google image says below. Maybe some driver thing?
```
root@raspberrypi:~# dmesg | grep rt5645
[   17.366150] rt5645 1-001a: 1-001a supply avdd not found, using dummy regulator
[   17.366347] rt5645 1-001a: 1-001a supply cpvdd not found, using dummy regulator
[   18.350991] snd-soc-aiy-voicebonnet soc:sound: rt5645-aif1 <-> 20203000.i2s mapping ok
root@raspberrypi:~#
```

shivasiddharth, 2021-01-11:
Now that you are on AIY image, can you move into env folder and get back on the python version that is being used ?

bbrendon, 2021-01-11:
I copied the /boot/config.txt from the google to yours. And speaker-test worked.  Its below. Should I continue from a point in the instructions?

```
pi@raspberrypi:~ $ cat /boot/config.txt
[pi4]
dtoverlay=vc4-fkms-v3d
max_framebuffers=2

[all]
[pi0]
dtoverlay=dwc2
[all]
start_x=1
gpu_mem=128
dtoverlay=spi0-1cs,cs0_pin=7
pi@raspberrypi:~ $ dmesg | grep rt5645
[   15.698998] rt5645 1-001a: 1-001a supply avdd not found, using dummy regulator
[   15.699181] rt5645 1-001a: 1-001a supply cpvdd not found, using dummy regulator
[   17.114210] snd-soc-aiy-voicebonnet soc:sound: rt5645-aif1 <-> 20203000.i2s mapping ok
pi@raspberrypi:~ $

```

shivasiddharth, 2021-01-11:
Yes you can. Can you attach the full config.txt here. I will add a script for voice hat v2

bbrendon, 2021-01-11:
Thats the full config.txt without the comments.

bbrendon, 2021-01-11:
I found the full one with comments. https://pastebin.com/kEFCxsFP