shivasiddharth/GassistPi #1106: No feedback form the assistant in push button mode

EmadMokhtar, 2021-01-28:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using?
master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls 
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 000000003496dd1f
Model           : Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
Raspberry Pi reference 2020-12-02
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, cce27bd6f44a3b2e83855645986b3e21f771e852, stage2
```

### Describe the bug:

I installed the `` on my Raspberry pi zero W It has respeaker 2 mics HAT. There is no issue with the sound but when I try to run the Assistant with push-button I got 2 issues.

- I hear the Google welcome message, but when I try to speak to the mic it catches the keyword, but no feedback/sound comes from the speaker.
- I change the stop button to GPIO 17 but still not stopping.

```
pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'peerless-voyage-299414' --device-model-id 'peerless-voyage-299414-officeamp-m8ak3t'
/home/pi/GassistPi/src/youtube_search_engine.py:13: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  configuration = yaml.load(conf)
/home/pi/GassistPi/src/mediaplayer.py:17: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  configuration = yaml.load(conf)
/home/pi/env/lib/python3.7/site-packages/oauth2client/_helpers.py:255: UserWarning: Cannot access /home/pi/.local/share/gmusicapi/mobileclient.cred: No such file or directory
  warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
2021-01-28 17:37:32,500 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
/home/pi/GassistPi/src/actions.py:48: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  configuration = yaml.load(conf)
/home/pi/GassistPi/src/actions.py:51: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  langlist = yaml.load(lang)
/home/pi/GassistPi/src/actions.py:94: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  custom_action_keyword = yaml.load(conf)
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 1 detected at time: 2021-01-28 17:38:11
hello
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
NA. I'm starting the assistant manually.

shivasiddharth, 2021-01-29:
1. First of all, running any of the assistants either Alexa or Google on Pi Zero is not advised. 
2. How did you install the Respeaker driver ? From what you have pasted, looks like you have installed it from official repo and not mine.
3. Stopbutton is completely different from pushbutton trigger. Change the pushbutton trigger on config file to 17, disable the custom wakeword and see if trigger by pushbutton trigger works.

EmadMokhtar, 2021-01-29:
First of all, thanks for the great work and the fast reply. Really appreciated.

I installed the driver using the official repo. I wil give the suggested configuration a try and let you know.

Why raspberry pi zero is not adviced? And Waht are the boards you advise me to use?

shivasiddharth, 2021-01-29:
Raspberry Pi Zero W is ideal for server applications and not for computational task intensive applications like the assistants. I would suggest to use at least Pi 3B. If you are on a budget, then you can use Orange Pi Zero, its under 10 bucks if my memory serves right.

EmadMokhtar, 2021-01-29:
I followed your step and install the driver using your script. The google assistant service is running, I can hear the welcome (Hello) but after that nothing.

``` bash
pi@raspberrypi:~ $ sudo systemctl status google-assistant.service
● google-assistant.service - Google Assistant Voice Service
   Loaded: loaded (/lib/systemd/system/google-assistant.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2021-01-29 14:38:57 GMT; 2min 37s ago
 Main PID: 828 (python)
    Tasks: 21 (limit: 881)
   CGroup: /system.slice/google-assistant.service
           └─828 /home/pi/env/bin/python -u /home/pi/Assistants-Pi/Google-Assistant/src/main.py --project_id peerless-voyage-299414 -

Jan 29 14:39:07 raspberrypi python[828]: jack server is not running or cannot be started
Jan 29 14:39:07 raspberrypi python[828]: JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Jan 29 14:39:08 raspberrypi python[828]: /home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: Deprecati
Jan 29 14:39:08 raspberrypi python[828]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Jan 29 14:39:08 raspberrypi python[828]: device_model_id: peerless-voyage-299414-officeamp-m8ak3t
Jan 29 14:39:08 raspberrypi python[828]: device_id: 9E0C8CA68AEDCFDB17FC8862CE8F9A7C
Jan 29 14:39:09 raspberrypi python[828]: ON_MUTED_CHANGED:
Jan 29 14:39:09 raspberrypi python[828]:   {"is_muted": false}
Jan 29 14:39:12 raspberrypi python[828]: ON_MEDIA_STATE_IDLE
Jan 29 14:39:12 raspberrypi python[828]: ON_START_FINISHED
```

For Alexa, it is not running. 

``` bash

pi@raspberrypi:~ $ sudo systemctl status alexa.service 
● alexa.service - Amazon Alexa Voice Service
   Loaded: loaded (/lib/systemd/system/alexa.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Fri 2021-01-29 14:36:10 GMT; 3min 9s ago
  Process: 748 ExecStart=/home/pi/Assistants-Pi/Alexa/start.sh (code=exited, status=203/EXEC)

Jan 29 14:36:10 raspberrypi systemd[1]: alexa.service: Service RestartSec=100ms expired, scheduling restart.
Jan 29 14:36:10 raspberrypi systemd[1]: alexa.service: Scheduled restart job, restart counter is at 5.
Jan 29 14:36:10 raspberrypi systemd[1]: Stopped Amazon Alexa Voice Service.
Jan 29 14:36:10 raspberrypi systemd[1]: alexa.service: Start request repeated too quickly.
Jan 29 14:36:10 raspberrypi systemd[1]: alexa.service: Failed with result 'exit-code'.
Jan 29 14:36:10 raspberrypi systemd[1]: Failed to start Amazon Alexa Voice Service.
```

EmadMokhtar, 2021-01-31:
I will close this issue then

closed 2021-01-31 by EmadMokhtar