shivasiddharth/GassistPi #1109: Custom voice hat issues, nothing is working

xFrah, 2021-02-24:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000fa424a23
Model		: Raspberry Pi 3 Model B Rev 1.2

cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
[... 302 lines cut ...]
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: jarvis-29a68-jarv-------
device_id: 5792B2A33137B5----

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

```

shivasiddharth, 2021-02-24:
Custom Voice HAT is for Custom AIY Voice HAT and not some Voice HAT of yours.
In the first attempt that you mentioned, you assistant was actually working. No sound was coming out of the HAT because the assistant volume was set to low. If you check the documentation, I have clearly mentioned that you need to give the command to set the assistant volume to high. Please go through the documentation once again.

closed 2021-02-24 by shivasiddharth

xFrah, 2021-02-24:
No, told it to set to maximum volume but it did not work.

xFrah, 2021-02-24:
By the way so what would be my way to install it? To just install it the manifacturer way?

xFrah, 2021-02-24:
And nothing more i suppose

shivasiddharth, 2021-02-24:
What is the audio setup of yours ? What is the manufacturers way you mean ?  You need to start the assistant manually, give commands and then share ther terminal contents. If you don't see any abnormalities in the terminal, I would check the audio device in the taskbar/menubar. Move your mouse over the audio icon, and select the correct device out device. Sometimes at every restart, the device out tends to revert to HDMI out.

xFrah, 2021-02-24:
It's a speaker/microphone hat, the manifacturer way i'm talking about is to just install it using the manifacturer drivers. I will do that in a moment and then paste everything again, but really, I tried many things before posting here.

shivasiddharth, 2021-02-24:
Raspberry Pi foundation as of today does not make Speaker and Mic HATs. So all of the HATs are from external companies. Which HAT and which company is all I want to know.

xFrah, 2021-02-24:
It's a Raspiaudio Ultra++

xFrah, 2021-02-24:
[image]

Still not speaking

xFrah, 2021-02-24:
Also, I tried using my monitor speakers, and it works, any idea why it doesnt with the hat? The only sound that comes from the hat is the one that plays when you say the hotword.

shivasiddharth, 2021-02-24:
Unfortunately, it's not possible for me to support or debug hardware that I have not used or the ones less commonly used. This is going to be a long shot in the dark. After manually starting the assistant open pavucontrol and check for the volume level. If you don't have pavucontrol install it using,
sudo apt-get install pavucontrol

xFrah, 2021-02-24:
The volume is right, i checked, but the only thing that it's not working now is the voice. The starting sound and the hotword feedback works.

shivasiddharth, 2021-02-24:
Share th output of:
```
sudo nano ./.asoundrc
sudo nano /etc/asound.conf
```

xFrah, 2021-02-24:
[image]
[image]

shivasiddharth, 2021-02-24:
Can you see that asound.conf that you shared is truncated and it is not the full output. I would like to see the full output.

xFrah, 2021-02-24:
[image]
[image]

xFrah, 2021-02-24:
Take a look at this video, it may help you: https://youtu.be/d0FDop-fuJY

shivasiddharth, 2021-02-25:
I would suggest to try the following.
Do not uninstall the HAT. In the device out change the audio from HAT to HDMI, 3.5mm Jack and see what happens.

When the HAT is set as output, give it a long query like "what can you do for me" "how are you today".

xFrah, 2021-02-25:
already did that, when it's set as hdmi the voice of the assistant works, and comes through my monitor. When i output to jack nothing happens.

shivasiddharth, 2021-02-26:
This is Bizarre. If HDMI works, then Jack should also work. Use the raspi-config and try forcing the audio through Jack.

xFrah, 2021-02-26:
No ok, I've got it working, if you check the video one more time you'll see that the voice output only was "pre-muted", i don't know why but it comes like that on every fresh install. 
By the way now if i start it manually everything is fine, but if i make it autostart on boot it actually doesn't respond(after the starting sound). And I don't know how i can debug that.

shivasiddharth, 2021-02-26:
@xFrah That's why I asked you to give a long command. The command in the video was short for me to decipher anything. Open a terminal and run: 
`journalctl -u gassistpi.service -f`

xFrah, 2021-02-26:
[image]
this is the output of journalctl

shivasiddharth, 2021-02-26:
Can you please copy and paste the terminal contents. Screenshots dont cover the entire terminal output. If the terminal keeps scrolling, you can stop the service using:
`sudo systemctl stop gassistpi.service`

xFrah, 2021-02-26:
alsa-sink.c: ALSA woke us up to write new data to the device, but there was actually nothing to write!
alsa-sink.c: Most likely this is a bug in the ALSA driver *soundcardname*. Please report this issue to the ALSA developers.
alsa-sink.c: We were woken up with POLLOUT set -- however a subsequent snd_pcm_avail() returned 0 or another value < min_avail.
I wrote this by hand lol... Btw i believe there is nothing between this and the screenshots above.

xFrah, 2021-02-26:
If this doesn't help I can post all of the output, but it is kinda difficult since it is overritten in less than one second

xFrah, 2021-02-26:
I'm gonna post the whole thing tomorrow directly from my raspberry if this isn't enough

xFrah, 2021-02-27:
feb 27 20:50:52 raspberrypi python[515]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
----ERRORS FROM PREVIOUS MESSAGE BUT IN ITALIAN-----
feb 27 20:50:52 raspberrypi pulseaudio[976]: [alsa-source-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-source.c: Attivazione da parte di ALSA per leggere nuovi dati dal dispositivo, ma non c'era nulla da leggere.
feb 27 20:50:52 raspberrypi pulseaudio[976]: [alsa-source-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-source.c: Molto probabilmente si tratta di un bug nei driver ALSA "snd_soc_wm8960_soundcard". Segnalare questo problema agli sviluppatori di ALSA.
feb 27 20:50:52 raspberrypi pulseaudio[976]: [alsa-source-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-source.c: Attivazione avvenuta con POLLIN impostato -- tuttavia, una successiva snd_pcm_avail() ha ritornato 0 o un altro valore < min_avail.
-----------------------------------------------------------------------------
feb 27 20:51:11 raspberrypi python[515]: device_model_id: jarvis-29a68-jarvis-1.0-rc2c84
feb 27 20:51:11 raspberrypi python[515]: device_id: F807317F2A556C49A217957960A2BFD0
feb 27 20:51:12 raspberrypi python[515]: ON_MUTED_CHANGED:
feb 27 20:51:12 raspberrypi python[515]:   {"is_muted": false}
feb 27 20:51:16 raspberrypi python[515]: ON_MEDIA_STATE_IDLE
feb 27 20:51:16 raspberrypi python[515]: ON_START_FINISHED
feb 27 20:51:48 raspberrypi python[515]: Assistant is listening....
feb 27 20:51:48 raspberrypi python[515]: ON_CONVERSATION_TURN_STARTED
feb 27 20:51:51 raspberrypi python[515]: ON_END_OF_UTTERANCE
feb 27 20:51:51 raspberrypi python[515]: ON_END_OF_UTTERANCE
feb 27 20:51:51 raspberrypi python[515]: ON_RECOGNIZING_SPEECH_FINISHED:
feb 27 20:51:51 raspberrypi python[515]:   {"text": "Che tempo fa a Roma"}
feb 27 20:51:52 raspberrypi python[515]: ON_RENDER_RESPONSE:
feb 27 20:51:52 raspberrypi python[515]: {
feb 27 20:51:52 raspberrypi python[515]:   "text": "Adesso a Roma ci sono 11 gradi e cielo sereno. Stanotte si prevedono 10 gradi e cielo parzialmente nuvoloso.",
feb 27 20:51:52 raspberrypi python[515]:   "type": 0
feb 27 20:51:52 raspberrypi python[515]: }
feb 27 20:51:52 raspberrypi python[515]: ON_RESPONDING_STARTED:
feb 27 20:51:52 raspberrypi python[515]:   {"is_error_response": false}
feb 27 20:51:52 raspberrypi pulseaudio[976]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
feb 27 20:51:52 raspberrypi pulseaudio[976]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
feb 27 20:51:52 raspberrypi pulseaudio[976]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
feb 27 20:51:52 raspberrypi pulseaudio[976]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.

xFrah, 2021-02-27:
Also this only occurs with the gassistpi service, it works normally on manual start(even if after some time it crashes with a segmentation error)

shivasiddharth, 2021-02-28:
Try:
`sudo nano /etc/asound.conf`
In that, after the line `slave.pcm "hw:wm8960soundcard"` add the following.
```
slave.period_time 0
slave.period_size 1024
slave.buffer_size 8192
slave.rate 44100
```
Add # at the start of the line `defaults.pcm.rate_converter "samplerate"`

Then do
`sudo cp /etc/asound.conf ./.asoundrc`

Do a reboot and see.

This is the best shot i have got. I am waiting for the samples from this DAC manufacturer. Once I have them in hand, will try and see as to whats actually happening.

xFrah, 2021-02-28:
I tried and Unfortunately it didn't work, but I really appreciate all the work you put in, and i'm really sorry if i got you in this much trouble. I hope we can solve these issues.

shivasiddharth, 2021-02-28:
Can you try this:
```
sudo rm /etc/asound.conf
sudo rm ./.asoundrc
```
Reboot
Check if those files have not been overwritten. The following should be blank.
`sudo nano /etc/asound.conf`

xFrah, 2021-03-07:
I don't have an asoundrc, the command line says file not found, but I'm trying

xFrah, 2021-03-07:
Oh god, I just noticed that when i start the headless assistant without rebooting i have lots of errors, damn that's really impressive I don't even know how to send them all.

xFrah, 2021-03-07:
`mar 07 12:55:09 raspberrypi python[3648]:   serialized_end=171,
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:59: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:64: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   serialized_end=265,
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:77: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:82: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   serialized_end=440,
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:95: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:99: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:103: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:107: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:111: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:115: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:119: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:123: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   type=None),
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:128: DeprecationWarning: Call to deprecated create function EnumDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
mar 07 12:55:09 raspberrypi python[3648]:   serialized_end=606,
mar 07 12:55:09 raspberrypi python[3648]: /home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/uits_pb2.py:141: DeprecationWarning: Call to deprecated create function EnumValueDescriptor(). Note: Create unlinked descriptors is going to go away. Please use get/find descriptors from generated code or query the descriptor_pool.
[... cut ...]
mar 07 12:55:54 raspberrypi python[3716]:   warnings.warn(_MISSING_FILE_MESSAGE.format(filename))
mar 07 12:55:54 raspberrypi python[3716]: 2021-03-07 12:55:54,561 - gmusicapi.Mobileclient1 (shared:176) [WARNING]: could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
mar 07 12:55:54 raspberrypi python[3716]: WARNING:gmusicapi.Mobileclient1:could not retrieve oauth credentials from ''/home/pi/.local/share/gmusicapi/mobileclient.cred''
mar 07 12:55:54 raspberrypi python[3716]: INFO:gmusicapi.Mobileclient1:failed to authenticate
`

xFrah, 2021-03-07:
[GassistPi.log](https://github.com/shivasiddharth/GassistPi/files/6097047/GassistPi.log)

there is also this

and these are important i think, they come from the log sent above:
`mar 07 12:55:27 raspberrypi pulseaudio[3668]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
mar 07 12:55:27 raspberrypi pulseaudio[3668]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
mar 07 12:55:27 raspberrypi pulseaudio[3668]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
`

shivasiddharth, 2021-03-07:
I am lost here. Still manual start works but not the auto start right ? Those cards are one the way. Hold on untill i get to test them.

xFrah, 2021-03-07:
Yes, exactly, btw you are saving my whole project and you will make my gf happy in doing that ahahah. I even trained the hotword Jarvis in Italian, because the english one had to be pronounced perfectly and we don't do that here.

shivasiddharth, 2021-03-07:
Give me the output of:
```
aplay -l
arecord -l
```

xFrah, 2021-03-07:
`**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
`

`**** List of CAPTURE Hardware Devices ****
card 2: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
`

shivasiddharth, 2021-03-07:
Do:
`sudo nano /boot/config.txt`

There should be a line `dtparam=audio=on` add a `#` at the start. Save the file and reboot. Check what happens.

xFrah, 2021-03-19:
Sorry if it took so long, I'm gonna try soon. Also i read about the customs problem. Hope the cards get to you soon.