shivasiddharth/GassistPi #1112: Error when starting: gmusicapi: but your computer or network may be sending automated queries

JahFyahh, 2021-03-13:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Main Branch

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
pi@pi-google-assist:~/GassistPi $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 00000000f7cdd58b
Model           : Raspberry Pi Zero W Rev 1.1
pi@pi-google-assist:~/GassistPi $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@pi-google-assist:~/GassistPi $ cat /etc/rpi-issue
Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2
pi@pi-google-assist:~/GassistPi $

```

### Describe the bug:
When starting I receive the following error

```
Mar 13 14:38:58 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 13 14:39:33 pi-google-assist python[2241]: Traceback (most recent call last):
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/shared.py", line 216, in perform
Mar 13 14:39:33 pi-google-assist python[2241]:     response.raise_for_status()
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/env/lib/python3.7/site-packages/requests/models.py", line 943, in raise_for_status
Mar 13 14:39:33 pi-google-assist python[2241]:     raise HTTPError(http_error_msg, response=self)
Mar 13 14:39:33 pi-google-assist python[2241]: requests.exceptions.HTTPError: 404 Client Error: Not Found for url: https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo?alt=json&hl=en_US&dv=0&tier=fr
Mar 13 14:39:33 pi-google-assist python[2241]: During handling of the above exception, another exception occurred:
Mar 13 14:39:33 pi-google-assist python[2241]: Traceback (most recent call last):
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/GassistPi/src/pushbutton.py", line 41, in <module>
Mar 13 14:39:33 pi-google-assist python[2241]:     from actions import say
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
Mar 13 14:39:33 pi-google-assist python[2241]:     from mediaplayer import api
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/GassistPi/src/mediaplayer.py", line 23, in <module>
Mar 13 14:39:33 pi-google-assist python[2241]:     for device in api.get_registered_devices():
Mar 13 14:39:33 pi-google-assist python[2241]:   File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/mobileclient.py", line 880, in get_registered_devices
Mar 13 14:39:33 pi-google-assist python[2241]:     res = self._make_call(mobileclient.GetDeviceManagementInfo)
[... 95 lines cut ...]

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 41, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 14, in <module>
    from mediaplayer import api
  File "/home/pi/GassistPi/src/mediaplayer.py", line 23, in <module>
    for device in api.get_registered_devices():
  File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/mobileclient.py", line 880, in get_registered_devices
    res = self._make_call(mobileclient.GetDeviceManagementInfo)
  File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/mobileclient.py", line 43, in _make_call
    return super()._make_call(protocol, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/clients/shared.py", line 83, in _make_call
    return protocol.perform(self.session, self.validate, *args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/gmusicapi/protocol/shared.py", line 224, in perform
    raise CallFailure(err_msg, call_name)
gmusicapi.exceptions.CallFailure: GetDeviceManagementInfo: 404 Client Error: Not Found for url: https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo?alt=json&hl=en_US&dv=0&tier=fr
(requests kwargs: {'method': 'GET', 'url': 'https://mclients.googleapis.com/sj/v2.5/devicemanagementinfo', 'params': {'alt': 'json', 'hl': 'en_US', 'dv': 0, 'tier': 'fr'}, 'headers': {'Authorization': '<omitted>'}})
(response was: '<html><head><meta http-equiv="content-type" content="text/html; charset=utf-8"/><title>Sorry...</title><style> body { font-family: verdana, arial, sans-serif; background-color: #fff; color: #000; }</style></head><body><div><table><tr><td><b><font face=sans-serif size=10><font color=#4285f4>G</font><font color=#ea4335>o</font><font color=#fbbc05>o</font><font color=#4285f4>g</font><font color=#34a853>l</font><font color=#ea4335>e</font></font></b></td><td style="text-align: left; vertical-align: bottom; padding-bottom: 15px; width: 50%"><div style="border-bottom: 1px solid #dfdfdf;">Sorry...</div></td></tr></table></div><div style="margin-left: 4em;"><h1>We\'re sorry...</h1><p>... but your computer or network may be sending automated queries. To protect our users, we can\'t process your request right now.</p></div><div style="margin-left: 4em;">See <a href="https://support.google.com/websearch/answer/86640">Google Help</a> for more information.<br/><br/></div><div style="text-align: center; border-top: 1px solid #dfdfdf;"><a href="https://www.google.com">Google Home</a></div></body></html>')

```
  
I tried setting gmusicapi to Disabled in the config.yaml file, but this changes nothing.

shivasiddharth, 2021-03-13:
Hold on, gmusicapi needs to be removed as Google have brought down the Google Music Service. I will remove or comment the Google Music parts of the codes and post and update here.

JahFyahh, 2021-03-13:
@shivasiddharth That would be great, thank you. 
I'll await your update.

shivasiddharth, 2021-03-13:
I have commented out the gmusic parts. Delete or rename the existing GassistPi folder and re-clone the project and try it out.

JahFyahh, 2021-03-13:
@shivasiddharth Thanks for the quick work.

I renamed the folder and ran the manual command (without reboot), but i am now receiving this error:

```
[01b95b80] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01bcc680] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 57, in <module>
    from actions import refreshlists
ImportError: cannot import name 'refreshlists' from 'actions' (/home/pi/GassistPi/src/actions.py)

```

shivasiddharth, 2021-03-13:
I have fixed it in the master. You need not clone it again. Open the pushbutton.py file and at the top, you should have a line that says `from actions import refreshlists` just add a # at the start of the line and save the file.

JahFyahh, 2021-03-13:
Done. But here comes the next one.

```
[011b39c8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[011ea4c8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 67, in <module>
    import snowboydecoder
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 5, in <module>
    import snowboydetect
ModuleNotFoundError: No module named 'snowboydetect'
```
I tried `pip install snowboydetect` but no dice

pi@pi-google-assist:~ $ pip install snowboydetect
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting snowboydetect
Could not install packages due to an EnvironmentError: 404 Client Error: Not Found for url: https://pypi.org/simple/snowboydetect/

```
pi@pi-google-assist:~ $ source /env/bin/activate
-bash: /env/bin/activate: No such file or directory
pi@pi-google-assist:~ $ source env/bin/activate
(env) pi@pi-google-assist:~ $ pip install snowboydetect
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement snowboydetect
ERROR: No matching distribution found for snowboydetect
```

shivasiddharth, 2021-03-13:
Please do not try any random installations. God knows what breaks what and troubleshooting becomes even more difficult. Open the src folder in the GassistPi directory, within that you should find resources and within resources a folder called Buster-Wrapper. Copy those files and paste it in the src folder where you have all the other files/scripts. The installer script does this copying automatically. But as the original folder was deleted, this has to be done manually now.

JahFyahh, 2021-03-13:
oeps, sorry, though I was being smart and helpful. I copied the files as instructed and I got the following. It also keeps restarting

```
[01924a38] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[01934460] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Respeaker-2-Mic
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Aborted

```

shivasiddharth, 2021-03-13:
This is an old issue and was solved with the Dec release of the Raspberry Pi OS. What kernel version are you on ?

JahFyahh, 2021-03-13:
I had to run the respeaker installation with the compat switch due to it failing during the installation. I'm running the following:

```
pi@pi-google-assist:~ $ uname -a
Linux pi-google-assist 5.4.51+ #1333 Mon Aug 10 16:38:02 BST 2020 armv6l GNU/Linux
pi@pi-google-assist:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

shivasiddharth, 2021-03-13:
If you see these release notes https://downloads.raspberrypi.org/raspios_armhf/release_notes.txt you can see that so many audio related issues were fixed with the Dec and Jan release. Rolling down negates all those stuff. I need some time to look into things. In the meanwhile you could try this.
Uninstall Respeaker drivers, delete the old seeed-voicecard folder, restart and clone this branch https://github.com/HinTak/seeed-voicecard/tree/v5.5  using `git clone https://github.com/HinTak/seeed-voicecard -b v5.5` and then go through the installation process.

JahFyahh, 2021-03-14:
I started from scratch and reinstalled everything according, but now I am getting the following error, Gassist keeps restarting itself as well.

```
Mar 14 15:52:19 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 14 15:52:45 pi-google-assist python[715]: [0262fd28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 14 15:52:45 pi-google-assist python[715]: [0263b5e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 14 15:52:46 pi-google-assist python[715]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 14 15:52:56 pi-google-assist python[715]: INFO:snowboy:Keyword 1 detected at time: 2021-03-14 15:52:56
Mar 14 15:52:56 pi-google-assist python[715]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Mar 14 15:52:57 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 14 15:52:57 pi-google-assist systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 14 15:52:57 pi-google-assist systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 14 15:52:57 pi-google-assist systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 3.
Mar 14 15:52:57 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.

```

shivasiddharth, 2021-03-14:
Th Respeaker drivers are a mess right now.  Did you go through the updation process via the startup menu that you get on first boot ? Which branch of HinTak's drivers did you use now ?

shivasiddharth, 2021-03-14:
I would suggest you to try the following.
Uninstall the seeed-voicecard drivers and then do a reboot. Try if this works https://github.com/waveshare/WM8960-Audio-HAT the card is very much identical to the Respeaker 2 Mic HAT. The waveshare HAT misses out on the APA102 LEDs but the codec is all the same.

JahFyahh, 2021-03-15:
I did the routine installation of the seeed drivers, because the v5.5 branch also failed, i went with the drivers from HinTak's master branch, with that the speaker-test worked. I didn't get any startup menu on first boot. I did no manual update as your instructions described.

I removed the seeed drivers, installed the waveshare drivers and rebooted. The drivers work if i do a speaker-test, and Gassist does not seam to be restarting, but its also not responding to Ok, Google or Hey, Google. The journal says the following:

```
-- Logs begin at Mon 2021-03-15 08:50:58 CET. --
Mar 15 08:51:28 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 15 08:52:23 pi-google-assist python[481]: [00e04d28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 08:52:24 pi-google-assist python[481]: [00e105e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 08:52:25 pi-google-assist python[481]: Initializing GPIOs 5 and 6 for assistant activity indication
```

shivasiddharth, 2021-03-15:
The journal is truncated can you post the full journal output using `journalctl -u gassistpi.service -f`

JahFyahh, 2021-03-15:
That is the full journal, I used the same command as you posted. After the previous output nothing else happens, no matter how hard I scream at google. When I restarted the service, I still get the same output, only that time it did restart itself a couple of times. 

At one point I also got the following msg in the error:
`Mar 15 09:04:21 pi-google-assist python[485]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed. `

shivasiddharth, 2021-03-15:
Check the config.yaml. By default, I have set the wakewords to Alexa and Snowboy. Hey Google or Ok Google is not the default wakeword. If you want, you can change the wakeword in the config.yaml file. Just make sure to check that you provide the paths to the models that are there in the resources directory.

JahFyahh, 2021-03-15:
I tried it with snowboy as wake word and it does react, but it restarts again, this is from the journal after 2 tries

```
Mar 15 10:02:15 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 15 19:53:25 pi-google-assist python[479]: [00d79d28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 19:53:26 pi-google-assist python[479]: [00d855e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 19:53:27 pi-google-assist python[479]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 15 19:53:52 pi-google-assist python[479]: INFO:snowboy:Keyword 2 detected at time: 2021-03-15 19:53:52
Mar 15 19:53:53 pi-google-assist python[479]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Mar 15 19:53:53 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 15 19:53:53 pi-google-assist systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 15 19:53:54 pi-google-assist systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 15 19:53:54 pi-google-assist systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 1.
Mar 15 19:53:54 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.
Mar 15 19:53:54 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 15 19:54:19 pi-google-assist python[587]: [01f42d28] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 19:54:19 pi-google-assist python[587]: [01f4e5e8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 15 19:54:20 pi-google-assist python[587]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 15 19:55:53 pi-google-assist python[587]: INFO:snowboy:Keyword 2 detected at time: 2021-03-15 19:55:53
Mar 15 19:55:54 pi-google-assist python[587]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Mar 15 19:55:54 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 15 19:55:54 pi-google-assist systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 15 19:55:54 pi-google-assist systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 15 19:55:54 pi-google-assist systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 2.
Mar 15 19:55:54 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.
Mar 15 19:55:54 pi-google-assist systemd[1]: Started Google Assistant-Voice.

```

shivasiddharth, 2021-03-16:
Can you check if pulseaudio is installed and up to date ?

JahFyahh, 2021-03-16:
PulseAdio was not installed. at the moment I am following these instructions as it seams to be the same type of problem from this point: https://github.com/shivasiddharth/GassistPi/issues/1014#issuecomment-634395863

shivasiddharth, 2021-03-17:
Does speaker-test work ?

JahFyahh, 2021-03-17:
Yes that works, after I manually increase the volume of the ReSpeaker headphones (this resets on every boot)

JahFyahh, 2021-03-17:
My steps from scratch with the latest Raspbian destop image

1. raspi-config: Change password, hostname, timezone, WLAN zone, expand card
2. apt install: git tmux
3. git clone and install https://github.com/waveshare/WM8960-Audio-HAT

that failed
```
DKMS: add completed.
Error! echo
Your kernel headers for kernel 5.4.83+ cannot be found at
/lib/modules/5.4.83+/build or /lib/modules/5.4.83+/source.
```

4. reboot now
5. uninstall https://github.com/waveshare/WM8960-Audio-HAT
6. reboot now
7. install https://github.com/waveshare/WM8960-Audio-HAT (worked this time)
8. sudo rm /etc/xdg/autostart/piwiz.desktop (running headless, this stops the screenreader msg)
9. reboot now
10. speaker-test works (white noise)
11. git clone and install https://github.com/shivasiddharth/GassistPi
12. Forgot the move the client secret, ran the installer again with success
13. reboot now
14. Ran the headless scripts

journalctl -u gassistpi.service -f
```
-- Logs begin at Wed 2021-03-17 10:53:08 CET. --
Mar 17 11:17:37 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 17 11:18:13 pi-google-assist pulseaudio[892]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 17 11:18:13 pi-google-assist pulseaudio[892]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 17 11:18:13 pi-google-assist pulseaudio[892]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Mar 17 11:18:14 pi-google-assist python[879]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 17 11:18:16 pi-google-assist python[879]: WARNING:root:Device config not found: [Errno 2] No such file or directory: '/home/pi/.config/googlesamples-assistant/device_config.json'
Mar 17 11:20:45 pi-google-assist python[879]: INFO:snowboy:Keyword 2 detected at time: 2021-03-17 11:20:45
Mar 17 11:20:46 pi-google-assist pulseaudio[892]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
```
The last line keeps repeating itself until I restart the service. After restart I see the following journal, untill i say snowboy then it loops the line again: `pi-google-assist pulseaudio[892]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.`

```
Mar 17 11:22:58 pi-google-assist systemd[1]: Stopping Google Assistant-Voice...
Mar 17 11:22:58 pi-google-assist pulseaudio[963]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 17 11:22:58 pi-google-assist pulseaudio[963]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 17 11:22:58 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=15/TERM
Mar 17 11:22:58 pi-google-assist systemd[1]: gassistpi.service: Succeeded.
Mar 17 11:22:58 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.
Mar 17 11:22:58 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 17 11:23:29 pi-google-assist pulseaudio[1022]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 17 11:23:29 pi-google-assist pulseaudio[1022]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 17 11:23:29 pi-google-assist pulseaudio[1022]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Mar 17 11:23:30 pi-google-assist python[1010]: Initializing GPIOs 5 and 6 for assistant activity indication
```

shivasiddharth, 2021-03-17:
Does the manual start work ?

JahFyahh, 2021-03-17:
I'm afraid not, running manually gives me this:

```
pi@pi-google-assist:~ $ /home/pi/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'pi--assist-67406' --device_model_id 'pi--assist-67406-pi-google-assist-pandze'
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 51, in <module>
    from google.assistant.library import Assistant
ModuleNotFoundError: No module named 'google.assistant.library'
```

shivasiddharth, 2021-03-17:
You need to start the pushbutton manually not the main.py. https://gassistpi-documentation.readthedocs.io/en/latest/headless.html#steps-to-manually-start-the-assistant

JahFyahh, 2021-03-17:
ohh right, I ran, i get no errors. When i call out Snowboy it triggers and gets my command, but then silence....
This is the output:

```
pi@pi-google-assist:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi--assist-67406' --device-model-id 'pi--assist-67406-pi-google-assist-pandze'
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-17 16:17:21
what time is it
```

JahFyahh, 2021-03-18:
I did `sudo raspi-config > System options > Audio`  and this is set to the WM8960 drivers (this is the only option there by the way)

is there another way to check and change this?

shivasiddharth, 2021-03-18:
Move you mouse over the audio icon on the taskbar. It should give Audio Out and Audio In devices list. Select the HAT as the Audio Out Device. 
Also at the first instance of Google Assistant running, the audio volume should be low. Give "Set volume to Maximum" command. 
This situation is similar to the one posted by one of the users before. Sometimes, the audio will be muted on start (only in case of WM8960 based devices). You need to install pulseaudio control using `sudo apt-get install pavucontrol`.  Open pavucontrol from the menubar. It should be under Audio Configuration.. etc. Check if the audio is muted.

After triggering the assistant you should hear a "bleep" audio. Let me know if you hear that. These are some of my thoughts. 

Once you get the manual start working, we can work on the service (autostart).

JahFyahh, 2021-03-18:
I run it in headless mode, but I connected a monitor just to check this. When I hover the mouse over the audio icon is see 'volume control 100' when I right click on it u get 'No audio devices found', but when it started I did hear the assistant chime and 'hi' message. 

According to pavucontrol, all sounds are at  100%. 
In the playback and Recording tab I see python3.7 ALSA Capture/Playback both on 100%
The output devices tab only shows Built-in Audio
Input Devices and Configuration tabs show no available devices

After starting Gassist manually I see that python is using the audio as well (2 capture & 1 playback). But it does not respond to Snowboy anymore.

shivasiddharth, 2021-03-18:
Check this video that was shared by another user. https://www.youtube.com/watch?v=d0FDop-fuJY In that, when the python audio starts playing, you can clearly see a red cross. You are in the same boat as #1109 .

JahFyahh, 2021-03-18:
Did you ever get those cards to test with?

shivasiddharth, 2021-03-18:
Its been held up at customs for well over 2 weeks. Not cleared yet.

JahFyahh, 2021-03-18:
I just put the autologin to desktop via raspi-config and now it seams to be working when I run manually. It's still a bit buggy but this could be due to the low specs of the pi ZeroW, On boot I have to set the headphones volume manually. 

```
pi@pi-google-assist:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi--assist-67406' --device-model-id 'pi--assist-67406-pi-google-assist-pandze'
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 14:49:42
what time
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 14:50:02
what are you doing
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
WARNING:root:SoundDeviceStream write underflow (size: 1600)
Continuing conversation
what time is it
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 14:58:59
show me
Continuing conversation
i want to know the time
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 14:59:26
play some music
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 15:00:57
play some music on spotify
stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
some music on spotif
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/pushbutton.py", line 959, in <module>
    main()
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/env/lib/python3.7/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 938, in main
    start_detector()
  File "/home/pi/GassistPi/src/pushbutton.py", line 927, in start_detector
    sleep_time=0.03)
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 172, in start
    callback()
  File "/home/pi/GassistPi/src/pushbutton.py", line 915, in detected
    continue_conversation=assistant.assist()
  File "/home/pi/env/lib/python3.7/site-packages/tenacity/__init__.py", line 241, in wrapped_f
    return self.call(f, *args, **kw)
  File "/home/pi/env/lib/python3.7/site-packages/tenacity/__init__.py", line 330, in call
    start_time=start_time)
  File "/home/pi/env/lib/python3.7/site-packages/tenacity/__init__.py", line 279, in iter
    return fut.result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 425, in result
    return self.__get_result()
  File "/usr/lib/python3.7/concurrent/futures/_base.py", line 384, in __get_result
    raise self._exception
  File "/home/pi/env/lib/python3.7/site-packages/tenacity/__init__.py", line 333, in call
    result = fn(*args, **kwargs)
  File "/home/pi/GassistPi/src/pushbutton.py", line 559, in assist
    spotify_playlist_select(str(usrcmd).lower())
  File "/home/pi/GassistPi/src/actions.py", line 1532, in spotify_playlist_select
    playlists,num=scan_spotify_playlists()
  File "/home/pi/GassistPi/src/actions.py", line 1481, in scan_spotify_playlists
    if spotify_token:
NameError: name 'spotify_token' is not defined
```

shivasiddharth, 2021-03-18:
Glad we are making progress. Custom commands have specific command syntax. Only with that it will work. Please check the documentation for using the custom commands.

JahFyahh, 2021-03-18:
Ohh yeah or course, forgot about that one.

unfortunately, when using the automated start it throws errors at me when I call snowboy

```
pi@pi-google-assist:~ $ journalctl -u gassistpi.service -f                                                           -- Logs begin at Thu 2021-03-18 15:59:20 CET. --
Mar 18 15:59:47 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 18 16:01:40 pi-google-assist pulseaudio[839]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 18 16:01:40 pi-google-assist pulseaudio[839]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 18 16:01:40 pi-google-assist pulseaudio[839]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Mar 18 16:01:43 pi-google-assist python[374]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 18 16:02:39 pi-google-assist python[374]: INFO:snowboy:Keyword 2 detected at time: 2021-03-18 16:02:39
Mar 18 16:02:39 pi-google-assist pulseaudio[839]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 16:02:39 pi-google-assist pulseaudio[839]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 16:02:40 pi-google-assist pulseaudio[839]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 16:02:42 pi-google-assist python[374]: what is the time
Mar 18 16:02:43 pi-google-assist pulseaudio[839]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
```

JahFyahh, 2021-03-18:
I just added the api key to the config.yaml and ran Gassist manually and tried the bellow but no sound/music and snowboy is not listening to me anymore (got his headphones on listening to my music choice):

```
pi@pi-google-assist:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi--assist-67406' --device-model-id 'pi--assist-67406-pi-google-assist-pandze'
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 16:20:28
play happy from youtube
stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

```

shivasiddharth, 2021-03-18:
You should hear some audio messages like getting music links etc. Do you get them ? There are lot more comments from youtube feature that should be printed which are missing.

JahFyahh, 2021-03-18:
No I am not hearing anything, After a reboot it stopped working, I also disabled the service from starting.

```
pi@pi-google-assist:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi--assist-67406' --device-model-id 'pi--assist-67406-pi-google-assist-pandze'
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 17:42:21
tell me something
```

JahFyahh, 2021-03-18:
When I enable the service restart and stop the service and run Gassist manually it works again.
When trying to play youtube it gets the link (about a minute), returns the link but playback really dodgy

```
pi@pi-google-assist:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'pi--assist-67406' --device-model-id 'pi--assist-67406-pi-google-assist-pandze'
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 17:53:05
what time is it
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 17:53:19
tell me a joke
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
WARNING:root:SoundDeviceStream write underflow (size: 2732)
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
WARNING:root:SoundDeviceStream write underflow (size: 1600)
ALSA lib pcm.c:8424:(snd_pcm_recover) underrun occurred
WARNING:root:SoundDeviceStream write underflow (size: 2732)
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 17:53:39
play happy from youtube
stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
happy
https://r5---sn-5hnekn7z.googlevideo.com/videoplayback?expire=1616108068&ei=xIVTYMnuBcjB1gLyp7GAAg&ip=94.157.204.121&id=o-AJjfry-d8LGfIuF3ook7-N2EU-9of5nN2Y6tzkep79tA&itag=140&source=youtube&requiressl=yes&mh=4s&mm=31%2C29&mn=sn-5hnekn7z%2Csn-5hne6nzs&ms=au%2Crdu&mv=m&mvi=5&pl=16&gcr=nl&initcwndbps=1993750&vprv=1&mime=audio%2Fmp4&ns=ybOQRRm47WaYm_d5hmyC_msF&gir=yes&clen=3895821&dur=240.674&lmt=1584647994407070&mt=1616086373&fvip=5&keepalive=yes&fexp=24001373%2C24007246&c=WEB&txp=5431432&n=EvFJPzUhQKvwLMk2wa&sparams=expire%2Cei%2Cip%2Cid%2Citag%2Csource%2Crequiressl%2Cgcr%2Cvprv%2Cmime%2Cns%2Cgir%2Cclen%2Cdur%2Clmt&lsparams=mh%2Cmm%2Cmn%2Cms%2Cmv%2Cmvi%2Cpl%2Cinitcwndbps&lsig=AG3C_xAwRAIgRfeGvp5rnC-bHDnOfx_Yu2CFO-M3eizP90heLZ-tfRoCIFD4BIQkHmY019vtMdUOFATZWJusj9hd6_udkvZyFwUU&sig=AOq0QJ8wRAIgHmtModyoNrbbUPpsryO_SXr3bMPlAfDgmRH_6q4y97UCIFLJ33XXRsz-kKBJ1sMM9Qb5GQicngb7IjJOtwzJpWcP
[85533bc8] mp4 demux: Fragment sequence discontinuity detected 1 != 0
```

Do I have to wait for the song to end before being able to call on snowboy again?

JahFyahh, 2021-03-18:
at the moment its working but it keeps on "listening", after every command the listening sound goes off even in silence. How to stop it?

shivasiddharth, 2021-03-18:
From my understanding, you mean that Assistant automatically expects a second query after your first? If that's the case, then probably you have continued conversation enabled in the Google Home app.

Open Google Home app, press on the account photo or pic at the top right corner, select Assistant Settings, then scroll down untill you reach Continued Conversation. Select that and disable that.

shivasiddharth, 2021-03-18:
I have got notes throughout the documentation that Pi Zero is not ideal for music or advanced features. So please do not have high expectations.

JahFyahh, 2021-03-18:
Understood, what is the minimal recommended pi? Pi2B?

Now need to see how to get it working headless from boot. 
The service still fails
```
pi@pi-google-assist:~ $ journalctl -u gassistpi.service -f                                                           -- Logs begin at Thu 2021-03-18 17:46:00 CET. --
Mar 18 17:50:26 pi-google-assist pulseaudio[882]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 17:50:26 pi-google-assist pulseaudio[882]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 17:50:30 pi-google-assist python[399]: tell me a joke
Mar 18 17:50:31 pi-google-assist pulseaudio[882]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Resume failed, couldn't restore original sample settings.
Mar 18 17:52:01 pi-google-assist systemd[1]: Stopping Google Assistant-Voice...
Mar 18 17:52:01 pi-google-assist pulseaudio[882]: [alsa-sink-bcm2835-i2s-wm8960-hifi wm8960-hifi-0] alsa-sink.c: Error opening PCM device hw:0,0: Device or resource busy
Mar 18 17:52:01 pi-google-assist systemd[1]: gassistpi.service: Main process exited, code=killed, status=15/TERM
Mar 18 17:52:01 pi-google-assist systemd[1]: gassistpi.service: Succeeded.
Mar 18 17:52:01 pi-google-assist systemd[1]: Stopped Google Assistant-Voice.
Mar 18 18:47:27 pi-google-assist systemd[1]: Started Google Assistant-Voice.
Mar 18 18:48:02 pi-google-assist pulseaudio[1834]: [pulseaudio] module.c: Failed to load module "module-alsa-sink" (argument: "device=hw:0,0"): initialization failed.
Mar 18 18:48:02 pi-google-assist pulseaudio[1834]: [pulseaudio] module.c: Failed to load module "module-alsa-source" (argument: "device=hw:0,0"): initialization failed.
Mar 18 18:48:02 pi-google-assist pulseaudio[1834]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 18 18:48:02 pi-google-assist pulseaudio[1834]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Mar 18 18:48:02 pi-google-assist pulseaudio[1834]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown: The name org.ofono was not provided by any .service files
Mar 18 18:48:03 pi-google-assist python[1815]: Initializing GPIOs 5 and 6 for assistant activity indication
```

And the manual command does not know when to stop, it this a setting somewhere I could tweak?

JahFyahh, 2021-03-18:
After a series of reboots it stopped working again, after calling snowboy and the first request is does nothing any more. Will try again tomorrow to see why its not responding

```
pi@pi-google-assist:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id pi--assist-67406 --device-model-id pi--assist-67406-pi-google-assist-pandze
Initializing GPIOs 5 and 6 for assistant activity indication
INFO:snowboy:Keyword 2 detected at time: 2021-03-18 21:06:03
snowboy
```

shivasiddharth, 2021-03-19:
I would recommend nothing below Pi 3B, which itself is almost a 4 year old board. To fix autostart try this:

Do:
sudo nano /boot/config.txt

There should be a line dtparam=audio=on add a # at the start. Save the file and reboot. Check what happens.

JahFyahh, 2021-03-19:
I already took those steps based on your comment ```You are in the same boat as #1109 .````

shivasiddharth, 2021-03-19:
Can you post the outpus of:
```
aplay -l
arecord -l
sudo nano /etc/asound.conf
sudo nano ./.asoundrc
```

JahFyahh, 2021-03-19:
I do not have a `sudo nano ./.asoundrc` file. But here are the outputs of the other ones:

```
pi@pi-google-assist:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
[1]+  Killed                  /home/pi/env/bin/python -u /home/pi/GassistPi/src/pushbutton.py --project-id pi--assist-67406 --device-model-id pi--assist-67406-pi-google-assist-pandze
pi@pi-google-assist:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: wm8960soundcard [wm8960-soundcard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 [bcm2835-i2s-wm8960-hifi wm8960-hifi-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```
sudo nano /etc/asound.conf
```
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one

# use samplerate to resample as speexdsp resample is bad
#defaults.pcm.rate_converter "samplerate"

pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:wm8960soundcard"
    slave.period_time 0
    slave.period_size 1024
    slave.buffer_size 8192
    slave.rate 44100
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:wm8960soundcard"
        channels 2
    }
    ipc_key 666666
}
```

shivasiddharth, 2021-03-19:
ok try this:
sudo nano /etc/pulse/default.pa

add these 2 lines:
```
load-module module-alsa-source device = hw: 0,0 
load-module module-alsa-sink
```
save and reboot

JahFyahh, 2021-03-19:
I currently have these lines in that file:

```
load-module module-alsa-sink device=hw:0,0
load-module module-alsa-source device=hw:0,0
```

JahFyahh, 2021-03-19:
I removed the 'device=hw:0,0' from behind de line `load-module module-alsa-sink` and rebooted with service enabled and it hears and replies as well.

shivasiddharth, 2021-03-19:
Finally, issue solved.

JahFyahh, 2021-03-19:
Will reboot and test a couple more times and let you know. Then I will install from scratch on a new card and see which changes fixed it and report back. 

Thank you for your time and patience with me.

JahFyahh, 2021-03-19:
I added `sleep 20 && sudo amixer cset numid=11 100%` to the rc.local file and that puts the volume to the desired state at boot.

shivasiddharth, 2021-03-20:
Can we consider this issue closed or do you need assistance with something else ?

JahFyahh, 2021-03-20:
Almost, let me reinstall from scratch to see if I can replicate a working installation. Then I will paste my steps for others and close this issue.

JahFyahh, 2021-03-20:
Unfortunaly, I still cannot get the volume at the proper level at boot, don't know what fixed it last time. but here are the steps that that work on the Pi Zero W with Respeaker 2 mic card

```
# Installed Raspberry Pi OS with desktop (Rel. Jan 11th 2021) on Raspberry Pi Zero W
# All steps are taken headless via SSH

sudo rm /etc/xdg/autostart/piwiz.desktop # To stop that screenreader message)

sudo apt install git -y

raspi-config: Change password, hostname, timezone, WLAN zone, desktop auto login, expand card

git clone https://github.com/waveshare/WM8960-Audio-HAT

cd WM8960-Audio-HAT && sudo ./install.sh  # this might fail the first time, Reboot > Uninstall.sh > Reboot > Install usually works

sudo reboot now

speaker-test works # (white noise)  Stop with Ctrl + C and wait

comment line dtparam=audio=on in sudo nano /boot/config.txt

added the following lines to sudo nano /etc/pulse/default.pa
- load-module module-alsa-source device = hw: 0,0 
- load-module module-alsa-sink

updated asound.conf to https://github.com/shivasiddharth/GassistPi/issues/1112#issuecomment-802714803

upload the client secret

git clone https://github.com/shivasiddharth/GassistPi 

sudo chmod +x ./GassistPi/scripts/gassist-installer.sh

sudo  ./GassistPi/scripts/gassist-installer.sh

sudo reboot now

sudo chmod +x ./GassistPi/scripts/service-installer.sh

sudo ./GassistPi/scripts/service-installer.sh

sudo systemctl enable gassistpi.service

sudo systemctl start gassistpi.service

```

@shivasiddharth  thank you so much for your assistance in this. It works, will do a couple more tests, especially with the sound volume at boot. If there is anything worth reporting I will.

closed 2021-03-20 by JahFyahh

JahFyahh, 2021-03-20:
It looks like after boot, the volume gets lowered by some process. I added 
`sleep 60 && /usr/bin/sudo /usr/bin/amixer cset numid=11 93%`  to `sudo nano /etc/rc.local` and for some strange reason this sets the volume level of the respeaker headphone jack to 74%. So I'm happy now, on to the advanced features.