shivasiddharth/GassistPi #1113: authorization error

alias13e, 2021-03-21:
First of all thank you so much this is amazing and must have been a tremendous amount of work.  Everything installs fine and I get the long url to authorize this application, but when I go there and sign in it says Error 403: Access Denied and tells me to email the developer, which is the same email I'm trying to log in with.  I've run this tutorial from scratch twice and hit the same error.  Please advise.

closed 2021-03-21 by close-issue-app[bot]

alias13e, 2021-03-21:
I have scoured the closed issues. I’ve found similar problems, but not anything applicable to my issue. Sorry, I’ll keep looking, I’m sure it’s a simple fix.

shivasiddharth, 2021-03-22:
Refer to the instructions in the video https://youtu.be/TLP1Ysu8d3g Google Assistant project creation starts at 3:13

alias13e, 2021-03-22:
That worked! Thank you so much! This is some amazing work you’re doing here. 

Sent from a pneumatic tube.