shivasiddharth/GassistPi #1115: Start GassistPi continuously after each hot word "Alexa" is called.

hoangtruong2412, 2021-03-23:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  

processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 997.08
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 000000001f5a69b1
Model           : Raspberry Pi Zero W Rev 1.1

cat /etc/os-release  

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

cat /etc/rpi-issue

Raspberry Pi reference 2021-01-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 21090519d85bdaa1615d5d5057d37b09368ea5d2, stage2

Paste your Board and OS details here.   

### Describe the bug:
Hello, first of all thanks to Shivasiddharth for creating a great application, I'm currently having an error when installing GassistPi, every time I call "Alexa" it instantly starts the service, not starts Reboot the Rasbian operating system. I have a short and descriptive video, because it was shot during the day so it's a bit noisy due to outside noise, hope everyone understand. At first I thought my 5V 3A power was weak so every time I got a sound there was a drop of current, but I tried adding another power supply plug in on the ReSpeaker 2-Mics Pi HAT Module and the results were still the same. I am currently using the hardware Pi zero W and the ReSpeaker 2-Mics Pi HAT module. I am using the 2021-01-11-raspios-buster-armhf-lite version I downloaded last night. Before I used many other rasbian versions, before buying the ReSpeaker 2-Mics Pi HAT module, I used a cheap Chinese USB Sound Card that did not suffer from that phenomenon, only the sensitivity of the MIC was not good. so I bought the ReSpeaker 2-Mics Pi HAT with the desire to improve the MIC sensitivity. Everyone help me. I have now followed the instructions from here:
https://gassistpi-documentation.readthedocs.io/en/latest/headless.html
Here is my error video link:
https://www.youtube.com/watch?v=itdWB1Toc9Q

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below

pi@raspberrypi:~ $ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'smartspeak-hat-2-mic' --device-model-id 'smartspeak-hat-2-mic-smartspeak-hat-2-mic-w78wrm'
[024d63c0] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
[0250cf40] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Initializing GPIOs 5 and 6 for assistant activity indication
python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Aborted
pi@raspberrypi:~ $

shivasiddharth, 2021-03-23:
The whole idea of using a template is to help users troubleshoot faster. In the template you say that you have gone through old issues but clearly have not checked the old issues. Check #1112  So much of effort was put into resolving that and @JahFyah has created a small writeup as well.

closed 2021-03-23 by shivasiddharth

hoangtruong2412, 2021-03-23:
sorry i have seen many ways, many problems and worked on recovering from many problems on your github site, but all fail since i assembled the ReSpeaker 2-Mics Pi HAT for USB Sound card. I installing and reinstalling many times. Did you watch my short clip about my problem? I try the reference # 1112. Thanks very much

shivasiddharth, 2021-03-23:
Your issue is the same as #1112 . Please go through that.

hoangtruong2412, 2021-03-23:
yes i have been there since last night, with Seeed V5.5 driver, but no, now i am trying this https://github.com/waveshare/WM8960-Audio-HAT

hoangtruong2412, 2021-03-27:
Mar 27 04:38:45 raspberrypi systemd[1]: Started Google Assistant-Voice.
Mar 27 04:39:11 raspberrypi python[10057]: [01d61ac8] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 27 04:39:11 raspberrypi python[10057]: [01d98750] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 27 04:39:12 raspberrypi python[10057]: Respeaker-2-Mic
Mar 27 04:41:06 raspberrypi python[10057]: INFO:snowboy:Keyword 1 detected at time: 2021-03-27 04:41:06
Mar 27 04:41:06 raspberrypi python[10057]: python: src/hostapi/alsa/pa_linux_alsa.c:3641: PaAlsaStreamComponent_BeginPolling: Assertion `ret == self->nfds' failed.
Mar 27 04:41:06 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 27 04:41:07 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 27 04:41:07 raspberrypi systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Mar 27 04:41:07 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 3.
Mar 27 04:41:07 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 27 04:41:07 raspberrypi systemd[1]: Started Google Assistant-Voice.

hoangtruong2412, 2021-03-27:
I have reinstalled, is this still the error? as I posted above.