shivasiddharth/GassistPi #1117: Spotify; Assistant says getting link then it restarts

javegab21734, 2021-04-03:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
https://github.com/shivasiddharth/GassistPi
cloned the file directly following the instructions from https://gassistpi-documentation.readthedocs.io/en/latest/starting.html

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03114
Serial		: 10000000407ffd4f
Model		: Raspberry Pi 4 Model B Rev 1.4

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

### Describe the bug:
```
After telling the assistant to play a playlist from spotify, it says getting link and then does nothing. The assistant restarts afterward saying the greeting when booting up.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play try from spotify"}

stopping vlc
Playing WAVE '/tmp/male-say.wav' : Signed 16 bit Little Endian, Rate 24000 Hz, Mono
try
2021-04-03 15:09:06,579 : ERROR : HTTP Error for GET to https://api.spotify.com/v1/users/12146898328/playlists returned 400 due to Only valid bearer authentication supported
Fatal Python error: Segmentation fault

Thread 0x5b1fe460 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 159 in start
  File "/home/pi/GassistPi/src/main.py", line 504 in start_detector
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0x5e1fd460 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 277 in pushbutton
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Thread 0xb6f18ad0 (most recent call first):
  File "/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1121 in main
  File "/home/pi/GassistPi/src/main.py", line 1129 in <module>
Segmentation fault
```

shivasiddharth, 2021-04-03:
This seems to be an issue with your Spotify token authentication. I am sure as to what's happening there. I would suggest you to Google and see first.

shivasiddharth, 2021-04-03:
Please double check the credentials you have entered in the config file.

shivasiddharth, 2021-04-27:
Closing this for inactivity.

closed 2021-04-27 by shivasiddharth