shivasiddharth/GassistPi #1126: module 'yaml' has no attribute 'FullLoader

nathanshankar, 2021-04-25:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03112
Serial		: 1000000049b737c5
Model		: Raspberry Pi 4 Model B Rev 1.2
pi@raspberrypi:~ $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

```

### Describe the bug:
```

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py --project_id 'mayo-f5c83' --device_model_id 'mayo-f5c83-mayo-assistant-cuw9ss'

```

### Paste the contents of the terminal below
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 29, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 15, in <module>
    from youtube_search_engine import google_cloud_api_key
  File "/home/pi/GassistPi/src/youtube_search_engine.py", line 13, in <module>
    configuration = yaml.load(conf, Loader=yaml.FullLoader)
AttributeError: module 'yaml' has no attribute 'FullLoader'

```

shivasiddharth, 2021-04-25:
Open a terminal and run:
```
source /env/bin/activate
pip3 install --upgrade pyyaml
```
You have a number of issues opened without the template. Also, you opened up an issue and never reverted back with the requested details. Please make sure to stick to the git etiquette.

closed 2021-04-25 by shivasiddharth

nathanshankar, 2021-04-26:
Thank you for your reply I was able to rectify it by changing the Loader=yaml.SafeLoader