shivasiddharth/GassistPi #1128: Error with pyusb causing assistant not to run

Phil-MC, 2021-04-26:
pi@SlideShow:~ $ **cat /proc/cpuinfo**
processor : 0
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 44.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 1
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 44.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 2
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 44.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 3
model name : ARMv7 Processor rev 4 (v7l)
BogoMIPS : 44.80
Features : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

Hardware : BCM2835
Revision : a02082
Serial : 00000000eccdc925
Model : Raspberry Pi 3 Model B Rev 1.2
pi@SlideShow:~ $

pi@SlideShow:~ $ **cat /etc/os-release**
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@SlideShow:~ $

**Describe the bug:**
Error ModuleNotFoundError: No module named 'usb'

**Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues).**
Yes

**2. Is the issue related to assistant installation ?**
No

**If answer to question 2 is No, then proceed further.**
**3. Does the assistant service start normally?**
No

**4. Is the assistant service restarting automatically?**
No

**If answer to question 4 is Yes, then paste the contents of the terminal below.**

**If answer to question 4 is No, then proceed further.
Manually start the assistant. For guidelines to start the assistant manually refer this
Paste the command below, that crashed the assistant**

/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --device_model_id 'slideshow-166614-slideshow2018-zjhcuk'

**Paste the contents of the terminal below**

Error ModuleNotFoundError: No module named 'usb'

**Attach the log file named Gassistpi.log**

pi@SlideShow:/tmp $ cat GassistPi.log
2021-04-26 16:22:18,769 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-04-26 16:22:18,778 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-04-26 16:22:19,030 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-04-26 16:22:19,067 : DEBUG : Converted retries value: 3 -> Retry(total=3, connect=None, read=None, redirect=None, status=None)
2021-04-26 16:22:19,076 : DEBUG : Starting new HTTPS connection (1): embeddedassistant.googleapis.com:443
2021-04-26 16:22:19,348 : DEBUG : https://embeddedassistant.googleapis.com:443 "GET /v1alpha2/projects/slideshow-166614/devices/49939C25B6D5B1B9AB132A9670CB8673 HTTP/1.1" 404 None
2021-04-26 16:22:22,216 : DEBUG : https://embeddedassistant.googleapis.com:443 "POST /v1alpha2/projects/slideshow-166614/devices HTTP/1.1" 200 None
2021-04-26 17:22:17,037 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-04-26 17:22:17,047 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-04-26 17:22:17,298 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-04-26 18:22:15,303 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-04-26 18:22:15,312 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-04-26 18:22:15,568 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
pi@SlideShow:/tmp $

**Explanation**
I installed the Latest Raspios Buster 04-03-2021
and followed these instructions

Do not run the upgrade command.
Skip the update process when setting up a fresh copy of Raspbian OS.
and those at :
[https://gassistpi-documentation.readthedocs.io/en/latest/starting.html#](https://gassistpi-documentation.readthedocs.io/en/latest/starting.html#)
as far as including 'Installing Google Assistant'
At this point I used the manual start command and received:

`Error ModuleNotFoundError: No module named 'usb'
`
Investigation of the cause led me to think there was a problem with pyusb.
I ran:
`sudo apt-get remove pyusb`
and
`pip uninstall pyusb`
both at the /home/pi prompt and in /env
Then I reinstalled the latest pyusb in /env with:
`source env/bin/activate`
and
`pip install pyusb`
followed by
`sudo reboot`

I then repeated the manual start command and the system started and ran as expected.
Could you please look into this in order to insure that the correct pyusb is the only one available. I guess there must be a clash with the operating system files.

shivasiddharth, 2021-04-26:
When did you clone the repo ? Coz I fixed it two days back. Can you open the pip requirements file to see if pyusb is listed there. If it's missing, then you must have cloned the repo prior to me fixing it.

Phil-MC, 2021-04-27:
Hi shivasiddharth
I cloned it on April the 23rd.
The pip requirements file does not list pysub or the system requirements file for that matter.
Hope that helps...

shivasiddharth, 2021-04-27:
OK, I fixed the issue on the 24th. This issue has already been fixed.

closed 2021-04-27 by shivasiddharth

Phil-MC, 2021-04-27:
OK Thanks