shivasiddharth/GassistPi #1129: After successfull installation gassistpi.service throws exception and stops with error

Zeprakadebra, 2021-04-26:
Dear developer team,

many thanks for your great work. I have seen the result on some youtube videos and know I want to use it also on my smart mirror. However I face a problem after installation process. The service seems to not run well. Here are the details according to your form.

Looking forward to getting any advise how to solve the issue.

Best regards,
zebrakadebra

### Which branch of GassistPi are you using ?
master branch pulled on 26th, April 2021

### Board and OS details:    
cat /proc/cpuinfo  

Paste your Board and OS details here.   
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 270.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: d03114
Serial		: 100000001ff1c3c9
Model		: Raspberry Pi 4 Model B Rev 1.4

cat /etc/os-release  

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

cat /etc/rpi-issue

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage5

```

### Describe the bug:
```
When starting the service automatically or manually the systemctl status show that the service is failed.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No, the installation went successfully.

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
sudo systemctl status gassistpi.service
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-04-26 21:19:57 CEST; 22min ago
  Process: 2409 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id magicmirror-1d94e --device_model_id magicmirror-1d94e-magicmirror-7zt4h9 (code=exited, status=1/
 Main PID: 2409 (code=exited, status=1/FAILURE)

Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Apr 26 21:19:57 magicmirror systemd[1]: Stopped Google Assistant-Voice.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Start request repeated too quickly.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Apr 26 21:19:57 magicmirror systemd[1]: Failed to start Google Assistant-Voice.
...skipping...
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2021-04-26 21:19:57 CEST; 22min ago
  Process: 2409 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id magicmirror-1d94e --device_model_id magicmirror-1d94e-magicmirror-7zt4h9 (code=exited, status=1/
 Main PID: 2409 (code=exited, status=1/FAILURE)

Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Service RestartSec=100ms expired, scheduling restart.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Apr 26 21:19:57 magicmirror systemd[1]: Stopped Google Assistant-Voice.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Start request repeated too quickly.
Apr 26 21:19:57 magicmirror systemd[1]: gassistpi.service: Failed with result 'exit-code'.
Apr 26 21:19:57 magicmirror systemd[1]: Failed to start Google Assistant-Voice.
```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

Starting the service manually results in same error.

### Paste the command below, that crashed the assistant
```
see below
```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2021-04-26:
Are you using Raspberry Pi OS modified for magic mirror ?

Zeprakadebra, 2021-04-26:
I use the standard raspberry OS Full Desktop Version.

shivasiddharth, 2021-04-26:
Pi comes with the faulthandler module. Open a terminal and run:
source env/bin/activate
pip3 install faulthandler

After that try running the assistant.

Zeprakadebra, 2021-04-27:
Thanks for your your strong support and quick reply.

**pip3 install faulthandler** comes up with

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting faulthandler
  Downloading https://files.pythonhosted.org/packages/b9/35/16c89cd8f6e5c422b2b58143f52e12cb04904c14bb0bb7b1c8e5a4db6ac2/faulthandler-3.2.tar.gz (59kB)
    100% |████████████████████████████████| 61kB 1.5MB/s 
    Complete output from command python setup.py egg_info:
    ERROR: faulthandler is a builtin module since Python 3.3    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-nlw8_bo_/faulthandler/

However **python --version*** says I run on

Python 2.7.16

That is strange. Do you have any other advise for me to get it running?

shivasiddharth, 2021-04-27:
Did you run it outside the environment ? You should first run ```source env/bin/activate``` The environment is python3. This is really strange. I have never had someone report that faulthandler is missing.

Zeprakadebra, 2021-04-27:
I set the (env) first, of cause. But maybe you gave me the helping hint: I made python3 to my default python (env) - for some reason it started with python2 - now it seems to be working. I am going to test it in detail and let you know if everything works now as expected.

Zeprakadebra, 2021-04-27:
Update: It all works fine now for me. This issue can be closed. Many thanks for your strong support!

shivasiddharth, 2021-04-27:
Ok, Cheers.

closed 2021-04-27 by shivasiddharth