shivasiddharth/GassistPi #1130: Google Assistant Service breaks with errors after rolling back to custom_wakeword='disabled'

Zeprakadebra, 2021-04-28:
Hi shivasiddarth,

Many thanks for your strong product and support.

I have another issue. I hope you get time to take a look into it.

My Google Assistant service worked fine until I tried using the custom wakeword detection.

I tested snowboy with jarvis.umdl and picovoice with jarvis_raspberry-pi.ppn of cause with the necessary changes in config.yaml (custom_wakeword=enabled, wakeword_engine=xxx) accordingly.

However after restarting the service google assistant stopped both times with errors. Snowboy run into a script error and Picovoice claimed an outdated file format and broke too.

So I decided to roll back to custom wakeword = disabled. But although the config.yaml is now same like it was before google assistant does not work anymore.

Please find my issue form filled out below.

I like to use the google assistant again even if it works with 'ok google' only.
Can you please give me an advise how to get it running again?

Looking forward to your soonest reply.

Kind regards,
Zebrakadebra

**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

### Board and OS details:    
Open a terminal and run the commands
Paste your Board and OS details here.   
```
cat /proc/cpuinfo  
Hardware	: BCM2711
Revision	: d03114
Serial		: 100000001ff1c3c9
Model		: Raspberry Pi 4 Model B Rev 1.4

cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

cat /etc/rpi-issue
Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage5

### Describe the bug:
```
Google Assistant breaks with errors.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes, I did not find an issue reported like mine

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```
● gassistpi.service - Google Assistant-Voice
   Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2021-04-28 14:33:01 CEST; 8s ago
 Main PID: 1267 (python3)
    Tasks: 11 (limit: 4915)
   CGroup: /system.slice/gassistpi.service
           ├─1267 /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id xxxx --device_model_id xxxx
           └─1450 /usr/bin/pulseaudio --start --log-target=syslog

Apr 28 14:33:01 magicmirror systemd[1]: Started Google Assistant-Voice.
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemo
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [pulseaudio] backend-ofono.c: Failed to register as a handsfree audio agent with ofono: org.freedesktop.DBus.Error.ServiceUnknown
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [alsa-sink-bcm2835 Headphones] alsa-sink.c: ALSA weckte uns auf, um neue Daten auf das Gerät zu schreiben, doch es gab nichts zum
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [alsa-sink-bcm2835 Headphones] alsa-sink.c: Dies ist höchstwahrscheinlich ein Fehler im ALSA-Treiber »snd_bcm2835«. Bitte melden 
Apr 28 14:33:09 magicmirror pulseaudio[1450]: [alsa-sink-bcm2835 Headphones] alsa-sink.c: Wir wurden durch das POLLOUT-Set geweckt, allerdings lieferte ein anschließender snd_

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id xxxx --device_model_id xxxx
```

### Paste the contents of the terminal below
```
Respeaker-4-Mic
ERROR (ConvertStringToIntegerOrFloat():/snowboy-utils.h:
terminate called after throwing an instance of 'std::runtime_error'
  what():  ERROR (ConvertStringToIntegerOrFloat():/snowboy-utils.h:

[stack trace: ]
Backtrace has not been implemented yet.

Fatal Python error: Aborted

Thread 0x8becd460 (most recent call first):
  File "/usr/lib/python3.7/threading.py", line 296 in wait
  File "/usr/lib/python3.7/queue.py", line 170 in get
  File "/home/pi/GassistPi/src/indicator.py", line 203 in _run
  File "/usr/lib/python3.7/threading.py", line 865 in run
  File "/usr/lib/python3.7/threading.py", line 917 in _bootstrap_inner
  File "/usr/lib/python3.7/threading.py", line 885 in _bootstrap

Current thread 0xb6ff6ad0 (most recent call first):
  File "/home/pi/GassistPi/src/snowboydetect.py", line 69 in __init__
  File "/home/pi/GassistPi/src/snowboydecoder.py", line 91 in __init__
  File "/home/pi/GassistPi/src/main.py", line 220 in __init__
  File "/home/pi/GassistPi/src/main.py", line 1131 in <module>
Abgebrochen

```

shivasiddharth, 2021-04-28:
Snowboy is broken. To get picovoice working do:
```
source env/bin/activate
pip3 install pvporcupine==1.9.5
```

Zeprakadebra, 2021-04-29:
Thanks for your advise. I am happy to hear that. I am going to test it and let you know the result.

Zeprakadebra, 2021-05-02:
This fixed my issue. Many thanks!

shivasiddharth, 2021-05-02:
closing this then.

closed 2021-05-02 by shivasiddharth