shivasiddharth/GassistPi #1131: Illegal instruction when running examples after install, Buster on Pi Zero W

CaptainDimak, 2021-05-02:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

The main one -- I just git cloned the base repository.

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 697.95
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7

Hardware        : BCM2835
Revision        : 9000c1
Serial          : 0000000040f91eea
Model           : Raspberry Pi Zero W Rev 1.1

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2021-03-04
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 461109415073d2eb67083709662ba983cc191f14, stage4

### Describe the bug:
```
After following install instructions, the pushbutton.py script gives an "Illegal instruction" error. Other scripts seem not to work as well, though the service starts just fine. This shows up both on desktop and lite versions of Raspberry Pi OS. 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes -- someone reported seeing the issue on Stretch as well, but I'm on Buster. The official Google repo also has a closed issue from 2018: https://github.com/googlesamples/assistant-sdk-python/issues/235

### 2. Is the issue related to assistant installation ?
Yes, something in the install seems to be pulling binaries compiled for armv7.

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```
I have a copy of my most recent session, but it's the second time I ran the installer as I slightly mistyped my .json's file path the first time. I can get a clean one if needed, earlier attempts where I didn't mistype also failed in the same way. 

[gassistpi-installdump.txt](https://github.com/shivasiddharth/GassistPi/files/6412179/gassistpi-installdump.txt)
```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Can't tell

### 4. Is the assistant service restarting automatically?
Can't tell

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
this, but I'm using the unedited version to not share my IDs: 

/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'replace this with your project id' --device-model-id 'replace this with the model id'

I actually get the error whether the IDs are correct or not, upon testing. So this is a valid command to reproduce on my end. 
```

### Paste the contents of the terminal below
```
$ /home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/pushbutton.py --project-id 'replace this with your project id' --device-model-id 'replace this with the model id'
Illegal instruction
```

shivasiddharth, 2021-05-03:
Can you do:
```
source env/bin/activate
googlesamples-assistant-pushtotalk  --project-id 'replace this with your project id' --device-model-id 'replace this with the model id'
```
See if this runs on the terminal and revert back.

CaptainDimak, 2021-05-03:
Tested just now -- no change.

shivasiddharth, 2021-05-03:
Can you post the output of:
`uname -m`

CaptainDimak, 2021-05-03:
```
armv6l
```

shivasiddharth, 2021-05-03:
Try this:
```
source env/bin/activate
pip uninstall google-assistant-grpc==0.3.0
pip uninstall google-assistant-sdk==0.6.0
pip uninstall google-assistant-sdk[samples]==0.6.0
pip uninstall grpcio

pip install --upgrade --no-binary :all: grpcio
pip install google-assistant-grpc==0.3.0
pip install google-assistant-sdk==0.6.0
pip install google-assistant-sdk[samples]==0.6.0
```

Sk3yNet, 2021-05-03:
I have the same issue as OP (RPi Zero W/Illegal Instruction) & get stuck for hours at "Running setup.py install".

`(env) pi@raspberrypi:~ $ pip install --upgrade --no-binary :all: grpcio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting grpcio
  Downloading grpcio-1.37.1.tar.gz (21.7 MB)
     |████████████████████████████████| 21.7 MB 3.4 kB/s
Requirement already satisfied: six>=1.5.2 in ./env/lib/python3.7/site-packages (from grpcio) (1.15.0)
Skipping wheel build for grpcio, due to binaries being disabled for it.
Installing collected packages: grpcio
    Running setup.py install for grpcio ... \`

shivasiddharth, 2021-05-03:
@Sk3yNet  Pi Zero is not meant for Voice Assistant projects. I have been telling the same thing to those who keep asking me to support Alexa installation.  Pi Zero has very limited RAM as a result, compiling and installing modules take a lot of time. That's the trade off for a cheap setup.  This GRPCIO should take at least 40-50 minutes. To speed up the process you can increase the swap file size as shown here https://www.youtube.com/watch?v=u_0Y40yO2VU Again, the swap file size is limited by RAM. There are some articles that ask you to set up a swap partition, even I have posted a couple tutorials on that. But that should be the last resort. Setting up swap partition on the SD Card will shorten the card's life span.

Performing the installation on a Pi3 or Pi4 to speedup things and then using the card back on Pi Zero will not work.

CaptainDimak, 2021-05-03:
Just ran everything, and can confirm grpcio takes its time to install. Not an issue, just a Pi Zero thing. In my case, I'd switch to a Pi 3/4 in a heartbeat, but I dimensionally can't fit one in my project. 

Unfortunately, nothing seems to have changed on my end. Here's my install log, the ssh connection dropped when my pc fell asleep, but other than that it seems to have worked.

[gassistpi-reinstalldump.txt](https://github.com/shivasiddharth/GassistPi/files/6417545/gassistpi-reinstalldump.txt)

shivasiddharth, 2021-05-04:
Was about to ask you for those logs. Something interesting happening there. The SDK module is installing an old version of the GRPC client which is bonkers to me.
Try this:
```
source env/bin/activate
pip install --upgrade google-assistant-grpc

```
Its just not about the wakeword detection, the audio implementation for the GRPC has not been well implemented. It will raise audio rate errors for some audio hats. 

Just for enthusiasts sake I am still supporting Pi Zero for Google Assistant. Alexa compilation on Pi Zero will take well over 8 hours, newbies feel restless and they keep keying in Cntrl+C thinking that its an anomaly and end up opening issues after issues for something which is normal. Thats the main result I scrapped support for Alexa.

shivasiddharth, 2021-05-06:
Do we have an update on the issue status ?

CaptainDimak, 2021-05-06:
Sorry, got busy with work for a few days there. Ran the command, and there doesn't seem to be any changes. I might have already run it a few days ago, judging by the output:

```
(env) pi@red-phone:~ $ pip install --upgrade google-assistant-grpc
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: google-assistant-grpc in ./env/lib/python3.7/site-packages (0.3.0)
Requirement already satisfied: googleapis-common-protos>=1.5.2 in ./env/lib/python3.7/site-packages (from google-assistant-grpc) (1.53.0)
Requirement already satisfied: grpcio>=1.3.5 in ./env/lib/python3.7/site-packages (from google-assistant-grpc) (1.37.1)
Requirement already satisfied: protobuf>=3.12.0 in ./env/lib/python3.7/site-packages (from googleapis-common-protos>=1.5.2->google-assistant-grpc) (3.15.8)
Requirement already satisfied: six>=1.5.2 in ./env/lib/python3.7/site-packages (from grpcio>=1.3.5->google-assistant-grpc) (1.15.0)
```

shivasiddharth, 2021-05-07:
Can't do much. Its an upstream issue.

CaptainDimak, 2021-05-07:
I will attempt to follow up with the upstream sample repository.

shivasiddharth, 2021-12-01:
Closing this issue for inactivity.

closed 2021-12-01 by shivasiddharth