shivasiddharth/GassistPi #1137: Re-authentication needed every week

michaelbeljaars, 2021-09-08:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
GassistPi Master

### Board and OS details:    
Open a terminal and run   
```
pi@matrixcreator:~/GassistPi/scripts $ cat /proc/cpuinfo  
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 44.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 000000002603431e
Model		: Raspberry Pi 3 Model B Rev 1.2
pi@matrixcreator:~/GassistPi/scripts $ 
```
```
pi@matrixcreator:~/GassistPi/scripts $ cat /etc/os-release  
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@matrixcreator:~/GassistPi/scripts $ 
```
```
pi@matrixcreator:~/GassistPi/scripts $ cat /etc/rpi-issue
Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage4
pi@matrixcreator:~/GassistPi/scripts $ 
```

### Describe the bug:
```
GassistPi seems to function normally, but every week or so it has a problem with the authentication and I need to re-authenticate the device in order to make it functional again. I have looked for settings within the Google console that may have to do with expiration times of the tokens, but could not find any. Also searching for similar problems here and on the internet in general have yielded nothing.

I suspect it is not an issue with GassistPi itself, but may be a configuration problem. Any help is greatly appreciated.
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes (provided that it is successfully authenticated)

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2021-09-09:
Frankly, I have no clue as to why it is happening. Pointers are taking me nowhere. Thats why to reduce the inconvenience, I have given a script for re-authentication.

michaelbeljaars, 2021-09-16:
It may not be of any help, but I have updated my issue with the output of the terminal, when running the command manually. The assistant has worked for about a week.

michaelbeljaars, 2021-11-03:
> This is a Google API issue. I have no control over it. Also, I cannot find any information regarding this anywhere in the assistant documentation. The public interest in the Google Assistant has tapered off and I have halted the developmental activities for the same reason. If this trend continues, I will pull the plug on this project sometime.

That is too bad, but understandable. I will be looking for a different implementation of the google assistant for the Raspberry Pi. Thanks for your great work and help!

ASAD-BE18, 2022-11-18:
Hi, did you solve your problem?

pesengineeringclub, 2022-11-29:
Hello i have the same problem can any one help please ? Or did anyone tried to solve this problem

michaelbeljaars, 2022-11-29:
It's a limitation of an application in the Google API still being in the testing phase. You can change the status of your app to be "In production" (see screenshot) and it will solve the issue with re-authentication.

[image]

ASAD-BE18, 2022-11-29:
If we change the project mode to production then an Out-Of-Band authentication error occurs.

pesengineeringclub, 2022-11-29:
We did change the mode but after 7 days i got this again
[image]

meybe i can send you my teamviewer and you can connect and check my code ?

michaelbeljaars, 2022-11-29:
> If we change the project mode to production then an Out-Of-Band authentication error occurs. https://developers.google.com/identity/protocols/oauth2/resources/oob-migration#:~:text=2.0%20authorization%20endpoints.-,What%20is%20OOB%3F,approves%20an%20OAuth%20consent%20request.

I haven't used Gassistpi in a while, I guess since way before February 2022. I scrapped the assistant I had because this repo is no longer maintained. I suppose that was before the more secure implementation (and scrapping of OOB) and hence I did not have this problem. What would be needed (as far as I can tell) is an updated implementation of the authentication in accordance with Google's latest security requirements. I don't think that is going to happen.

michaelbeljaars, 2022-11-29:
> We did change the mode but after 7 days i got this again [image]
> 
> meybe i can send you my teamviewer and you can connect and check my code ?

**AFTER** switching to production mode, I had to activate one last time.