shivasiddharth/GassistPi #1141: Google assistant not loading up at boot

SamShah30112004, 2021-11-23:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
Just-Google-Assistant

### Board and OS details:    
Raspberry Pi OS Bullseye
Raspberry Pi 4B 2GB

Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 118.02
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 118.02
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 118.02
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 118.02
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: b03112
Serial		: 10000000bb582f90
Model		: Raspberry Pi 4 Model B Rev 1.2
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
Raspberry Pi reference 2021-10-30
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 288b21fc27e128ea6b330777aca68e0061ebf4fe, stage5

### Describe the bug:
```
The google assistant isnt starting at boot but starts and works perfectly by using the manual command I have tried the steps given in other similar issues but didnt work out for me..
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
After rebooting the system I checked the status of the service got the following thing:
gassistpi.service - Google Assistant-Voice
     Loaded: loaded (/lib/systemd/system/gassistpi.service; enabled; vendor preset: enabled)
     Active: failed (Result: signal) since Tue 2021-11-23 14:25:22 IST; 20min ago
    Process: 1562 ExecStart=/home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id pi-assistant-c3bdf --device_model_id pi-assistant-c3bdf-assistant-pi-x7iiqd (code=killed, signal=ABRT)
   Main PID: 1562 (code=killed, signal=ABRT)
        CPU: 436ms

Nov 23 14:25:22 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 5.
Nov 23 14:25:22 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Nov 23 14:25:22 raspberrypi systemd[1]: gassistpi.service: Start request repeated too quickly.
Nov 23 14:25:22 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Nov 23 14:25:22 raspberrypi systemd[1]: Failed to start Google Assistant-Voice.

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2021-11-23:
Post the output of the service journal.

SamShah30112004, 2021-11-24:
Hey I am tied up die to sudden health issues will not be available till next monday.. Sorry and thanks for such quick response

SamShah30112004, 2021-12-23:
Hey it got solved how i dont know. Thanks

closed 2021-12-23 by SamShah30112004