shivasiddharth/GassistPi #1142: don't want to play music from YouTube, Encountered an exception

MIKHANYA, 2021-11-30:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 135.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2835
Revision        : b03112
Serial          : confidential?
Model           : Raspberry Pi 4 Model B Rev 1.2
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspberrypi:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2020-02-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 5f884374b6ac6e155330c58caa1fb7249b8badf1, stage5

### Describe the bug:
```
don't want to play music from YouTube, Encountered an exception
```
### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
NO
### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes, but I use manual start

### 4. Is the assistant service restarting automatically?
Yes

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play some song from YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
some song
'dislike_count'
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
        version 1.25.10; written and copyright by Michael Hipp and others
        free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```
  
###LOG
2021-11-30 11:34:49,515 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-11-30 11:34:49,521 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-11-30 11:34:49,683 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-11-30 11:35:49,758 : DEBUG : text: Getting youtube link
2021-11-30 11:35:49,761 : DEBUG : tld: com
2021-11-30 11:35:49,762 : DEBUG : lang: en
2021-11-30 11:35:49,762 : DEBUG : slow: False
2021-11-30 11:35:49,762 : DEBUG : lang_check: True
2021-11-30 11:35:49,763 : DEBUG : pre_processor_funcs: [<function tone_marks at 0xb3212cd8>, <function end_of_line at 0xb3212e40>, <function abbreviations at 0xb3212f1$
2021-11-30 11:35:49,763 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!$
2021-11-30 11:35:49,764 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Da$
2021-11-30 11:35:49,766 : DEBUG : pre-processing: <function tone_marks at 0xb3212cd8>
2021-11-30 11:35:49,770 : DEBUG : pre-processing: <function end_of_line at 0xb3212e40>
2021-11-30 11:35:49,771 : DEBUG : pre-processing: <function abbreviations at 0xb3212f18>
2021-11-30 11:35:49,776 : DEBUG : pre-processing: <function word_sub at 0xb3226348>
2021-11-30 11:35:49,777 : DEBUG : text_parts: ['Getting youtube link']
2021-11-30 11:35:49,778 : DEBUG : text_parts: 1
2021-11-30 11:35:49,780 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Getting%20youtube%20link%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cn$
2021-11-30 11:35:49,791 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2021-11-30 11:35:50,128 : DEBUG : https://translate.google.com:443 "POST /_/TranslateWebserverUi/data/batchexecute HTTP/1.1" 200 None
2021-11-30 11:35:50,134 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML,$
2021-11-30 11:35:50,135 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-11-30 11:35:50,135 : DEBUG : status-0: 200
2021-11-30 11:35:50,137 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-11-30 11:35:50,138 : DEBUG : Saved to /tmp/female-say.mp3
2021-11-30 11:35:52,221 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-11-30 11:35:52,267 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=some+song&part=id%2Csnippet&key=AIzaSyATpQ_3tVEmXc1UiEI28$
2021-11-30 11:35:55,282 : DEBUG : text: Encountered an exception please check the logs.
2021-11-30 11:35:55,282 : DEBUG : tld: com
2021-11-30 11:35:55,282 : DEBUG : lang: en
2021-11-30 11:35:55,283 : DEBUG : slow: False
2021-11-30 11:35:55,283 : DEBUG : lang_check: True
2021-11-30 11:35:55,283 : DEBUG : pre_processor_funcs: [<function tone_marks at 0xb3212cd8>, <function end_of_line at 0xb3212e40>, <function abbreviations at 0xb3212f1$
2021-11-30 11:35:55,283 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!$
2021-11-30 11:35:55,283 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Da$
2021-11-30 11:35:55,284 : DEBUG : pre-processing: <function tone_marks at 0xb3212cd8>
2021-11-30 11:35:55,284 : DEBUG : pre-processing: <function end_of_line at 0xb3212e40>
2021-11-30 11:35:55,285 : DEBUG : pre-processing: <function abbreviations at 0xb3212f18>
2021-11-30 11:35:55,287 : DEBUG : pre-processing: <function word_sub at 0xb3226348>

shivasiddharth, 2021-11-30:
Your API key is invalid. Check your API key.

MIKHANYA, 2021-11-30:
> 
i changed api key 2 times but the problem remains. like in this https://cdn.shopify.com/s/files/1/0260/1864/4031/files/apiKey_1024x1024.png?v=1605774200
`
2021-11-30 15:29:52,898 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-11-30 15:29:52,904 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-11-30 15:29:53,088 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-11-30 15:30:08,291 : DEBUG : text: Getting youtube link
2021-11-30 15:30:08,292 : DEBUG : tld: com
2021-11-30 15:30:08,293 : DEBUG : lang: en
2021-11-30 15:30:08,293 : DEBUG : slow: False
2021-11-30 15:30:08,293 : DEBUG : lang_check: True
2021-11-30 15:30:08,294 : DEBUG : pre_processor_funcs: [<function tone_marks at 0xb31f4cd8>, <function end_of_line at 0xb31f4e40>, <function abbreviations at 0xb31f4f1$
2021-11-30 15:30:08,294 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!$
2021-11-30 15:30:08,295 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bg': 'Bulgarian', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy':$
2021-11-30 15:30:08,296 : DEBUG : pre-processing: <function tone_marks at 0xb31f4cd8>
2021-11-30 15:30:08,301 : DEBUG : pre-processing: <function end_of_line at 0xb31f4e40>
2021-11-30 15:30:08,302 : DEBUG : pre-processing: <function abbreviations at 0xb31f4f18>
2021-11-30 15:30:08,308 : DEBUG : pre-processing: <function word_sub at 0xb3207348>
2021-11-30 15:30:08,309 : DEBUG : text_parts: ['Getting youtube link']
2021-11-30 15:30:08,310 : DEBUG : text_parts: 1
2021-11-30 15:30:08,311 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Getting%20youtube%20link%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cn$
2021-11-30 15:30:08,320 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2021-11-30 15:30:08,511 : DEBUG : https://translate.google.com:443 "POST /_/TranslateWebserverUi/data/batchexecute HTTP/1.1" 200 None
2021-11-30 15:30:08,518 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML,$
2021-11-30 15:30:08,519 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-11-30 15:30:08,519 : DEBUG : status-0: 200
2021-11-30 15:30:08,521 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-11-30 15:30:08,523 : DEBUG : Saved to /tmp/female-say.mp3
2021-11-30 15:30:10,600 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-11-30 15:30:10,648 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=youtub&part=id%2Csnippet&key=AIzaSyAJENMDGeDEGRCxwQb2DFxe$
2021-11-30 15:30:13,492 : DEBUG : text: Encountered an exception please check the logs.
2021-11-30 15:30:13,494 : DEBUG : tld: com
2021-11-30 15:30:13,494 : DEBUG : lang: en
2021-11-30 15:30:13,494 : DEBUG : slow: False
2021-11-30 15:30:13,494 : DEBUG : lang_check: True
2021-11-30 15:30:13,495 : DEBUG : pre_processor_funcs: [<function tone_marks at 0xb31f4cd8>, <function end_of_line at 0xb31f4e40>, <function abbreviations at 0xb31f4f1$
2021-11-30 15:30:13,495 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!$
2021-11-30 15:30:13,495 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bg': 'Bulgarian', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy':$
2021-11-30 15:30:13,496 : DEBUG : pre-processing: <function tone_marks at 0xb31f4cd8>
2021-11-30 15:30:13,497 : DEBUG : pre-processing: <function end_of_line at 0xb31f4e40>
2021-11-30 15:30:13,497 : DEBUG : pre-processing: <function abbreviations at 0xb31f4f18>
2021-11-30 15:30:13,499 : DEBUG : pre-processing: <function word_sub at 0xb3207348>
                                                                           [ Read 95 lines ]
`

shivasiddharth, 2021-11-30:
What is this Shopify, etc. It does not belong to this project.

MIKHANYA, 2021-11-30:
> i changed api key 2 times but the problem remains. like in this https://cdn.shopify.com/s/files/1/0260/1864/4031/files/apiKey_1024x1024.png?v=1605774200
>
> 2021-11-30 15:29:52,898 : DEBUG : Making [...]

> What is this Shopify, etc. It does not belong to this project.

someone made a guide on your guide
https://store.createlabz.com/blogs/createlabz-tutorials/google-assistant2-4-integration-of-youtube-and-spotify-on-raspberry-pi?_pos=2&_sid=c34cd5a66&_ss=r

shivasiddharth, 2021-12-01:
That post you shared has some additional commands for some hanging issue. I do not support third party guides.  Certain library versions are frozen for a reason. Running additional commands break these. I would suggest you to drop a comment in that blog and seek help from the blog post creator. The part for spotify has also changed.

MIKHANYA, 2021-12-01:
> 

> That post you shared has some additional commands for some hanging issue. I do not support third party guides. Certain library versions are frozen for a reason. Running additional commands break [...]

It looks like I was late and missed the peak of popularity of the project :( . I used only Your guide https://gassistpi-documentation.readthedocs.io/en/latest/ and no other people's code, I just took a picture that supported my question. It looks like a lot here is outdated and just doesn't work like in the guide.

shivasiddharth, 2021-12-01:
Other than Snowboy, pretty much everything else is functional. Run the following command and share the output
```   
source env/bin/activate
pip list
```

shivasiddharth, 2021-12-01:
The log that you have posted here is truncated. If you see almost all the lines in the log are ending with $ That is why the API is showing up as invalid. I am re-opening this issue.

shivasiddharth, 2021-12-01:
What is the language that you have set in config.yaml ? Post the language here as how you have in the file.

jaedeezy, 2021-12-03:
I'm having practically an identical issue when asking to play a song from youtube. It was working fine a couple weeks ago with no changes. Language in config.yaml is the default 'en-US'

processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 51.20
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 2a020d3
Serial          : 000000001079adb7
Model           : Raspberry Pi 3 Model B Plus Rev 1.3

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2021-05-07
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, dcfd74d7d1fa293065ac6d565711e9ff891fe2b8, stage2

pip list:

Package                   Version
------------------------- ------------
adafruit-io               2.1
aftership                 1.3.0
cachetools                4.2.4
casttube                  0.2.1
certifi                   2021.10.8
[... 118 lines cut ...]
2021-12-03 14:15:48,282 : DEBUG : Saved to /tmp/female-say.mp3
2021-12-03 14:15:50,012 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-12-03 14:15:50,079 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=i+got+you&part=id%2Csnippet&key=AIzaSyDiJY6q2L2ay399pu0_58u9wOzjRgbjBYI&alt=json
2021-12-03 14:16:13,057 : DEBUG : text: Encountered an exception please check the logs.
2021-12-03 14:16:13,058 : DEBUG : tld: com
2021-12-03 14:16:13,058 : DEBUG : lang: en
2021-12-03 14:16:13,058 : DEBUG : slow: False
2021-12-03 14:16:13,059 : DEBUG : lang_check: True
2021-12-03 14:16:13,059 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x73224660>, <function end_of_line at 0x732247c8>, <function abbreviations at 0x732248a0>, <function word_sub at 0x73224c90>]
2021-12-03 14:16:13,059 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|—|¿|،|，|\\\n|…|：|‥|\\]|、|。|;|\\(|\\)|¡|\\[', re.IGNORECASE) from: [<function tone_marks at 0x73224d20>, <function period_comma at 0x73224d68>, <function colon at 0x73224db0>, <function other_punctuation at 0x73224df8>]>
2021-12-03 14:16:13,060 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bg': 'Bulgarian', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Danish', 'de': 'German', 'el': 'Greek', 'en': 'English', 'eo': 'Esperanto', 'es': 'Spanish', 'et': 'Estonian', 'fi': 'Finnish', 'fr': 'French', 'gu': 'Gujarati', 'hi': 'Hindi', 'hr': 'Croatian', 'hu': 'Hungarian', 'hy': 'Armenian', 'id': 'Indonesian', 'is': 'Icelandic', 'it': 'Italian', 'ja': 'Japanese', 'jw': 'Javanese', 'km': 'Khmer', 'kn': 'Kannada', 'ko': 'Korean', 'la': 'Latin', 'lv': 'Latvian', 'mk': 'Macedonian', 'ml': 'Malayalam', 'mr': 'Marathi', 'my': 'Myanmar (Burmese)', 'ne': 'Nepali', 'nl': 'Dutch', 'no': 'Norwegian', 'pl': 'Polish', 'pt': 'Portuguese', 'ro': 'Romanian', 'ru': 'Russian', 'si': 'Sinhala', 'sk': 'Slovak', 'sq': 'Albanian', 'sr': 'Serbian', 'su': 'Sundanese', 'sv': 'Swedish', 'sw': 'Swahili', 'ta': 'Tamil', 'te': 'Telugu', 'th': 'Thai', 'tl': 'Filipino', 'tr': 'Turkish', 'uk': 'Ukrainian', 'ur': 'Urdu', 'vi': 'Vietnamese', 'zh-CN': 'Chinese', 'zh-TW': 'Chinese (Mandarin/Taiwan)', 'zh': 'Chinese (Mandarin)'}
2021-12-03 14:16:13,063 : DEBUG : pre-processing: <function tone_marks at 0x73224660>
2021-12-03 14:16:13,065 : DEBUG : pre-processing: <function end_of_line at 0x732247c8>
2021-12-03 14:16:13,066 : DEBUG : pre-processing: <function abbreviations at 0x732248a0>
2021-12-03 14:16:13,070 : DEBUG : pre-processing: <function word_sub at 0x73224c90>
2021-12-03 14:16:13,071 : DEBUG : text_parts: ['Encountered an exception please check the logs.']
2021-12-03 14:16:13,072 : DEBUG : text_parts: 1
2021-12-03 14:16:13,072 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Encountered%20an%20exception%20please%20check%20the%20logs.%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cnull%2C%22generic%22%5D%5D%5D&
2021-12-03 14:16:13,077 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
[... cut ...]
2021-12-03 14:16:13,588 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '189'}
2021-12-03 14:16:13,589 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-03 14:16:13,589 : DEBUG : status-0: 200
2021-12-03 14:16:13,592 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-03 14:16:13,592 : DEBUG : Saved to /tmp/female-say.mp3

amitron01, 2021-12-03:
i also phase same issue. please solved the youtube playing issue. Almost 4 days i phased this problem. youtube song not playing.

log
`2021-12-04 04:52:24,777 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-12-04 04:52:24,816 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-12-04 04:52:25,239 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-12-04 04:52:45,383 : DEBUG : text: Getting youtube link
2021-12-04 04:52:45,390 : DEBUG : tld: com
2021-12-04 04:52:45,392 : DEBUG : lang: en
2021-12-04 04:52:45,394 : DEBUG : slow: False
2021-12-04 04:52:45,397 : DEBUG : lang_check: True
2021-12-04 04:52:45,401 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x7320fe88>, <function end_of_line at 0x73223030>, <function abbreviations at 0x73223108>, <function word_sub at 0x732234f8>]
2021-12-04 04:52:45,406 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|¡|，|\\]|¿|…|،|\\(|\\)|：|;|、|\\[|\\\n|—|。|‥', re.IGNORECASE) from: [<function tone_marks at 0x73223588>, <function period_comma at 0x732235d0>, <function colon at 0x73223618>, <function other_punctuation at 0x73223660>]>
2021-12-04 04:52:45,412 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Danish', 'de': 'German', 'el': 'Greek', 'en': 'English', 'eo': 'Esperanto', 'es': 'Spanish', 'et': 'Estonian', 'fi': 'Finnish', 'fr': 'French', 'gu': 'Gujarati', 'hi': 'Hindi', 'hr': 'Croatian', 'hu': 'Hungarian', 'hy': 'Armenian', 'id': 'Indonesian', 'is': 'Icelandic', 'it': 'Italian', 'ja': 'Japanese', 'jw': 'Javanese', 'km': 'Khmer', 'kn': 'Kannada', 'ko': 'Korean', 'la': 'Latin', 'lv': 'Latvian', 'mk': 'Macedonian', 'ml': 'Malayalam', 'mr': 'Marathi', 'my': 'Myanmar (Burmese)', 'ne': 'Nepali', 'nl': 'Dutch', 'no': 'Norwegian', 'pl': 'Polish', 'pt': 'Portuguese', 'ro': 'Romanian', 'ru': 'Russian', 'si': 'Sinhala', 'sk': 'Slovak', 'sq': 'Albanian', 'sr': 'Serbian', 'su': 'Sundanese', 'sv': 'Swedish', 'sw': 'Swahili', 'ta': 'Tamil', 'te': 'Telugu', 'th': 'Thai', 'tl': 'Filipino', 'tr': 'Turkish', 'uk': 'Ukrainian', 'ur': 'Urdu', 'vi': 'Vietnamese', 'zh-CN': 'Chinese', 'zh-cn': 'Chinese (Mandarin/China)', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'en-us': 'English (US)', 'en-ca': 'English (Canada)', 'en-uk': 'English (UK)', 'en-gb': 'English (UK)', 'en-au': 'English (Australia)', 'en-gh': 'English (Ghana)', 'en-in': 'English (India)', 'en-ie': 'English (Ireland)', 'en-nz': 'English (New Zealand)', 'en-ng': 'English (Nigeria)', 'en-ph': 'English (Philippines)', 'en-za': 'English (South Africa)', 'en-tz': 'English (Tanzania)', 'fr-ca': 'French (Canada)', 'fr-fr': 'French (France)', 'pt-br': 'Portuguese (Brazil)', 'pt-pt': 'Portuguese (Portugal)', 'es-es': 'Spanish (Spain)', 'es-us': 'Spanish (United States)'}
2021-12-04 04:52:45,421 : DEBUG : pre-processing: <function tone_marks at 0x7320fe88>
2021-12-04 04:52:45,440 : DEBUG : pre-processing: <function end_of_line at 0x73223030>
2021-12-04 04:52:45,446 : DEBUG : pre-processing: <function abbreviations at 0x73223108>
2021-12-04 04:52:45,475 : DEBUG : pre-processing: <function word_sub at 0x732234f8>
2021-12-04 04:52:45,483 : DEBUG : text_parts: ['Getting youtube link']
2021-12-04 04:52:45,488 : DEBUG : text_parts: 1
2021-12-04 04:52:45,493 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Getting%20youtube%20link%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cnull%2C%22generic%22%5D%5D%5D&
2021-12-04 04:52:45,530 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2021-12-04 04:52:46,240 : DEBUG : https://translate.google.com:443 "POST /_/TranslateWebserverUi/data/batchexecute HTTP/1.1" 200 None
2021-12-04 04:52:46,259 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '154'}
2021-12-04 04:52:46,262 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-04 04:52:46,265 : DEBUG : status-0: 200
2021-12-04 04:52:46,275 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-04 04:52:46,277 : DEBUG : Saved to /tmp/female-say.mp3
2021-12-04 04:52:48,981 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-12-04 04:52:49,166 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=dilbar+dilbar+song&part=id%2Csnippet&key=AIzaSyBHThWLxQ-em5v5MecJVttghmUvW37XYbA&alt=json
2021-12-04 04:53:01,431 : DEBUG : text: Encountered an exception please check the logs.
2021-12-04 04:53:01,434 : DEBUG : tld: com
2021-12-04 04:53:01,436 : DEBUG : lang: en
2021-12-04 04:53:01,439 : DEBUG : slow: False
2021-12-04 04:53:01,443 : DEBUG : lang_check: True
2021-12-04 04:53:01,447 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x7320fe88>, <function end_of_line at 0x73223030>, <function abbreviations at 0x73223108>, <function word_sub at 0x732234f8>]
2021-12-04 04:53:01,449 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|¡|，|\\]|¿|…|،|\\(|\\)|：|;|、|\\[|\\\n|—|。|‥', re.IGNORECASE) from: [<function  [... cut ...]
2021-12-04 04:53:02,289 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '189'}
2021-12-04 04:53:02,291 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-04 04:53:02,293 : DEBUG : status-0: 200
2021-12-04 04:53:02,302 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-04 04:53:02,306 : DEBUG : Saved to /tmp/female-say.mp3
`

amitron01, 2021-12-03:
```
CPU INFO

Raspberry Pi 2 Modecat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 1
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 2
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

processor	: 3
model name	: ARMv7 Processor rev 5 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xc07
CPU revision	: 5

Hardware	: BCM2835
Revision	: a21041
Serial		: 000000004c6499ff
Model		: Raspberry Pi 2 Model B Rev 1.1

OS

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```

shivasiddharth, 2021-12-04:
Use the custom conversation to see if it works.

shivasiddharth, 2021-12-04:
@jaedeezy , did you update the gtts ? The version you have does not match the version in the requirements file.

jaedeezy, 2021-12-04:
> @jaedeezy , did you update the gtts ? The version you have does not match the version in the requirements file.

Yea, I was hoping to resolve the issue but that did not work. Would it cause any problems leaving it at that version? Also, I just re-enabled the custom conversation and that works as it should while youtube commands still returns an exception error.

shivasiddharth, 2021-12-04:
The assistant is not crashing. Could be a slow youtube-dl issue (there are so many issues opened with their official git over the week). Give the YouTube command and just leave it like that to see if it starts at all

amitron01, 2021-12-04:
Please tell me how i solved youtube playing problem.

jaedeezy, 2021-12-05:
Can it be an authorization issue?

In the terminal I see this:

> pi@raspberrypi:~ $ GET https://youtube.googleapis.com/youtube/v3/search?q=i+got+you&part=id%2Csnippet&key=AIzaSyAExlV-kz3GG5zQtsKMQtZmPsl8piL8LEw&alt=json
[1] 11241
[2] 11242
[3] 11243
pi@raspberrypi:~ $ {
  "error": {
    "code": 403,
    "message": "The request is missing a valid API key.",
    "errors": [
      {
        "message": "The request is missing a valid API key.",
        "domain": "global",
        "reason": "forbidden"
      }
    ],
    "status": "PERMISSION_DENIED"
  }
}

 I've changed and checked the api key numerous times

shivasiddharth, 2021-12-06:
@jaedeezy  The key that you first posted was invalid, but this one works. Every time you change something in the config.yaml, you need to restart the assistant. It does not pick up the change dynamically. YouTube API is not the issue. youtube-dl ha sbeen reported to be slow over the last week.

De-Vincci, 2021-12-06:
@shivasiddharth Did you find any solution about this problem? Please update and tell me how to fix this.

shivasiddharth, 2021-12-06:
It's like searching for a black cat in a dark room. 
1. Gtts is working.
2. YouTube Data API is working.
3. youtube-dl is working but slow.
So I don't see an issue at all.

De-Vincci, 2021-12-06:
I couldn't find any solution. I also changed, checked the API key, and restart. But no fixes happened. Please help me.😭

shivasiddharth, 2021-12-06:
You guys need to be patient and give some time for the YouTube Data API to get activated. From the looks of it, you guys are not giving it the time it  needs.

shivasiddharth, 2021-12-06:
To check if the API is activated, https://youtube.googleapis.com/youtube/v3/search?q=i+got+you&part=id%2Csnippet&key=INSERT-YOUR-KEY-HERE&alt=json 

Insert your key in the above given space and open it in the browser. If you get some details, then you can confirm that the key has been activated.

De-Vincci, 2021-12-06:
API is activated. But still, I can't play music on youtube.☹

amitron01, 2021-12-06:
Api is activated. But Youtube not work.

shivasiddharth, 2021-12-06:
Attach the log file here

shivasiddharth, 2021-12-06:
Add 
```
print(videoids)  
``` 
After this line: https://github.com/shivasiddharth/GassistPi/blob/0cad5a4e3564968c556c13c8d7c92cf848f6621c/src/youtube_search_engine.py#L53
Manually restart the assistant. Give the command and share the terminal contents.

amitron01, 2021-12-06:
Log File

` 2021-12-06 13:37:43,582 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-12-06 13:37:43,667 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-12-06 13:37:44,147 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-12-06 13:37:57,061 : DEBUG : text: Getting youtube link
2021-12-06 13:37:57,066 : DEBUG : tld: com
2021-12-06 13:37:57,071 : DEBUG : lang: en
2021-12-06 13:37:57,074 : DEBUG : slow: False
2021-12-06 13:37:57,080 : DEBUG : lang_check: True
2021-12-06 13:37:57,086 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x731bfed0>, <function end_of_line at 0x731d4078>, <function abbreviations at 0x731d4150>, <function word_sub at 0x731d4540>]
2021-12-06 13:37:57,089 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|—|，|…|\\]|،|。|¡|\\)|¿|;|、|：|\\(|\\[|‥|\\\n', re.IGNORECASE) from: [<function tone_marks at 0x731d45d0>, <function period_comma at 0x731d4618>, <function colon at 0x731d4660>, <function other_punctuation at 0x731d46a8>]>
2021-12-06 13:37:57,095 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Danish', 'de': 'German', 'el': 'Greek', 'en': 'English', 'eo': 'Esperanto', 'es': 'Spanish', 'et': 'Estonian', 'fi': 'Finnish', 'fr': 'French', 'gu': 'Gujarati', 'hi': 'Hindi', 'hr': 'Croatian', 'hu': 'Hungarian', 'hy': 'Armenian', 'id': 'Indonesian', 'is': 'Icelandic', 'it': 'Italian', 'ja': 'Japanese', 'jw': 'Javanese', 'km': 'Khmer', 'kn': 'Kannada', 'ko': 'Korean', 'la': 'Latin', 'lv': 'Latvian', 'mk': 'Macedonian', 'ml': 'Malayalam', 'mr': 'Marathi', 'my': 'Myanmar (Burmese)', 'ne': 'Nepali', 'nl': 'Dutch', 'no': 'Norwegian', 'pl': 'Polish', 'pt': 'Portuguese', 'ro': 'Romanian', 'ru': 'Russian', 'si': 'Sinhala', 'sk': 'Slovak', 'sq': 'Albanian', 'sr': 'Serbian', 'su': 'Sundanese', 'sv': 'Swedish', 'sw': 'Swahili', 'ta': 'Tamil', 'te': 'Telugu', 'th': 'Thai', 'tl': 'Filipino', 'tr': 'Turkish', 'uk': 'Ukrainian', 'ur': 'Urdu', 'vi': 'Vietnamese', 'zh-CN': 'Chinese', 'zh-cn': 'Chinese (Mandarin/China)', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'en-us': 'English (US)', 'en-ca': 'English (Canada)', 'en-uk': 'English (UK)', 'en-gb': 'English (UK)', 'en-au': 'English (Australia)', 'en-gh': 'English (Ghana)', 'en-in': 'English (India)', 'en-ie': 'English (Ireland)', 'en-nz': 'English (New Zealand)', 'en-ng': 'English (Nigeria)', 'en-ph': 'English (Philippines)', 'en-za': 'English (South Africa)', 'en-tz': 'English (Tanzania)', 'fr-ca': 'French (Canada)', 'fr-fr': 'French (France)', 'pt-br': 'Portuguese (Brazil)', 'pt-pt': 'Portuguese (Portugal)', 'es-es': 'Spanish (Spain)', 'es-us': 'Spanish (United States)'}
2021-12-06 13:37:57,100 : DEBUG : pre-processing: <function tone_marks at 0x731bfed0>
2021-12-06 13:37:57,124 : DEBUG : pre-processing: <function end_of_line at 0x731d4078>
2021-12-06 13:37:57,126 : DEBUG : pre-processing: <function abbreviations at 0x731d4150>
2021-12-06 13:37:57,154 : DEBUG : pre-processing: <function word_sub at 0x731d4540>
2021-12-06 13:37:57,157 : DEBUG : text_parts: ['Getting youtube link']
2021-12-06 13:37:57,158 : DEBUG : text_parts: 1
2021-12-06 13:37:57,162 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Getting%20youtube%20link%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cnull%2C%22generic%22%5D%5D%5D&
2021-12-06 13:37:57,194 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2021-12-06 13:37:57,594 : DEBUG : https://translate.google.com:443 "POST /_/TranslateWebserverUi/data/batchexecute HTTP/1.1" 200 None
2021-12-06 13:37:57,623 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '154'}
2021-12-06 13:37:57,625 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-06 13:37:57,625 : DEBUG : status-0: 200
2021-12-06 13:37:57,636 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-06 13:37:57,639 : DEBUG : Saved to /tmp/female-say.mp3
2021-12-06 13:38:00,343 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-12-06 13:38:00,488 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=waka+waka+song&part=id%2Csnippet&key=AIzaSyBHThWLxQ-em5v5MecJVttghmUvW37XYbA&alt=json
2021-12-06 13:38:27,185 : DEBUG : text: Encountered an exception please check the logs.
2021-12-06 13:38:27,187 : DEBUG : tld: com
2021-12-06 13:38:27,187 : DEBUG : lang: en
2021-12-06 13:38:27,188 : DEBUG : slow: False
2021-12-06 13:38:27,188 : DEBUG : lang_check: True
2021-12-06 13:38:27,188 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x731bfed0>, <function end_of_line at 0x731d4078>, <function abbreviations at 0x731d4150>, <function word_sub at 0x731d4540>]
2021-12-06 13:38:27,189 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|—|，|…|\\]|،|。|¡|\\)|¿|;|、|：|\\(|\\[|‥|\\\n', re.IGNORECASE) from: [<function tone_marks at 0x731d45d0>, <function period_comma at 0x731d4618>, <function colon at 0x731d4660>, <function other_punctuation a [... cut ...]
2021-12-06 13:38:28,160 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '189'}
2021-12-06 13:38:28,162 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-06 13:38:28,163 : DEBUG : status-0: 200
2021-12-06 13:38:28,179 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-06 13:38:28,182 : DEBUG : Saved to /tmp/female-say.mp3
`

shivasiddharth, 2021-12-06:
Attach it as a file. I cannot make any sense of what is pasted.

De-Vincci, 2021-12-06:
[log.txt](https://github.com/shivasiddharth/GassistPi/files/7658901/log.txt)

shivasiddharth, 2021-12-06:
> [log.txt](https://github.com/shivasiddharth/GassistPi/files/7658901/log.txt)

Now this please https://github.com/shivasiddharth/GassistPi/issues/1142#issuecomment-986513668

De-Vincci, 2021-12-06:
I already do that before uploading the log file. I will just remove it and upload the log file again.

shivasiddharth, 2021-12-06:
Log file has nothing to do with that. I need to see as to whats getting printed in the terminal. Perform that change, give the instruction and share the terminal contents.

amitron01, 2021-12-06:
After add your command . Still not working youtube. 
```
2021-12-06 13:49:52,824 : DEBUG : Making request: POST https://accounts.google.com/o/oauth2/token
2021-12-06 13:49:52,872 : DEBUG : Starting new HTTPS connection (1): accounts.google.com:443
2021-12-06 13:49:53,346 : DEBUG : https://accounts.google.com:443 "POST /o/oauth2/token HTTP/1.1" 200 None
2021-12-06 13:50:06,877 : DEBUG : text: Getting youtube link
2021-12-06 13:50:06,886 : DEBUG : tld: com
2021-12-06 13:50:06,889 : DEBUG : lang: en
2021-12-06 13:50:06,892 : DEBUG : slow: False
2021-12-06 13:50:06,895 : DEBUG : lang_check: True
2021-12-06 13:50:06,897 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x731dce40>, <function end_of_line at 0x731dcfa8>, <function abbreviations at 0x731ef0c0>, <function word_sub at 0x731ef4b0>]
2021-12-06 13:50:06,901 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|…|‥|、|¿|\\[|;|，|：|。|\\(|،|\\)|—|¡|\\]|\\\n', re.IGNORECASE) from: [<function tone_marks at 0x731ef540>, <function period_comma at 0x731ef588>, <function colon at 0x731ef5d0>, <function other_punctuation at 0x731ef618>]>
2021-12-06 13:50:06,907 : DEBUG : langs: {'af': 'Afrikaans', 'ar': 'Arabic', 'bn': 'Bengali', 'bs': 'Bosnian', 'ca': 'Catalan', 'cs': 'Czech', 'cy': 'Welsh', 'da': 'Danish', 'de': 'German', 'el': 'Greek', 'en': 'English', 'eo': 'Esperanto', 'es': 'Spanish', 'et': 'Estonian', 'fi': 'Finnish', 'fr': 'French', 'gu': 'Gujarati', 'hi': 'Hindi', 'hr': 'Croatian', 'hu': 'Hungarian', 'hy': 'Armenian', 'id': 'Indonesian', 'is': 'Icelandic', 'it': 'Italian', 'ja': 'Japanese', 'jw': 'Javanese', 'km': 'Khmer', 'kn': 'Kannada', 'ko': 'Korean', 'la': 'Latin', 'lv': 'Latvian', 'mk': 'Macedonian', 'ml': 'Malayalam', 'mr': 'Marathi', 'my': 'Myanmar (Burmese)', 'ne': 'Nepali', 'nl': 'Dutch', 'no': 'Norwegian', 'pl': 'Polish', 'pt': 'Portuguese', 'ro': 'Romanian', 'ru': 'Russian', 'si': 'Sinhala', 'sk': 'Slovak', 'sq': 'Albanian', 'sr': 'Serbian', 'su': 'Sundanese', 'sv': 'Swedish', 'sw': 'Swahili', 'ta': 'Tamil', 'te': 'Telugu', 'th': 'Thai', 'tl': 'Filipino', 'tr': 'Turkish', 'uk': 'Ukrainian', 'ur': 'Urdu', 'vi': 'Vietnamese', 'zh-CN': 'Chinese', 'zh-cn': 'Chinese (Mandarin/China)', 'zh-tw': 'Chinese (Mandarin/Taiwan)', 'en-us': 'English (US)', 'en-ca': 'English (Canada)', 'en-uk': 'English (UK)', 'en-gb': 'English (UK)', 'en-au': 'English (Australia)', 'en-gh': 'English (Ghana)', 'en-in': 'English (India)', 'en-ie': 'English (Ireland)', 'en-nz': 'English (New Zealand)', 'en-ng': 'English (Nigeria)', 'en-ph': 'English (Philippines)', 'en-za': 'English (South Africa)', 'en-tz': 'English (Tanzania)', 'fr-ca': 'French (Canada)', 'fr-fr': 'French (France)', 'pt-br': 'Portuguese (Brazil)', 'pt-pt': 'Portuguese (Portugal)', 'es-es': 'Spanish (Spain)', 'es-us': 'Spanish (United States)'}
2021-12-06 13:50:06,915 : DEBUG : pre-processing: <function tone_marks at 0x731dce40>
2021-12-06 13:50:06,926 : DEBUG : pre-processing: <function end_of_line at 0x731dcfa8>
2021-12-06 13:50:06,931 : DEBUG : pre-processing: <function abbreviations at 0x731ef0c0>
2021-12-06 13:50:06,985 : DEBUG : pre-processing: <function word_sub at 0x731ef4b0>
2021-12-06 13:50:06,990 : DEBUG : text_parts: ['Getting youtube link']
2021-12-06 13:50:06,992 : DEBUG : text_parts: 1
2021-12-06 13:50:06,996 : DEBUG : data-0: f.req=%5B%5B%5B%22jQ1olc%22%2C%22%5B%5C%22Getting%20youtube%20link%5C%22%2C%5C%22en%5C%22%2Cnull%2C%5C%22null%5C%22%5D%22%2Cnull%2C%22generic%22%5D%5D%5D&
2021-12-06 13:50:07,027 : DEBUG : Starting new HTTPS connection (1): translate.google.com:443
2021-12-06 13:50:07,735 : DEBUG : https://translate.google.com:443 "POST /_/TranslateWebserverUi/data/batchexecute HTTP/1.1" 200 None
2021-12-06 13:50:07,762 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '154'}
2021-12-06 13:50:07,766 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-06 13:50:07,768 : DEBUG : status-0: 200
2021-12-06 13:50:07,779 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-06 13:50:07,782 : DEBUG : Saved to /tmp/female-say.mp3
2021-12-06 13:50:10,331 : INFO : file_cache is only supported with oauth2client<4.0.0
2021-12-06 13:50:10,469 : DEBUG : URL being requested: GET https://youtube.googleapis.com/youtube/v3/search?q=waka+waka+song&part=id%2Csnippet&key=AIzaSyBHThWLxQ-em5v5MecJVttghmUvW37XYbA&alt=json
2021-12-06 13:50:21,535 : DEBUG : text: Encountered an exception please check the logs.
2021-12-06 13:50:21,537 : DEBUG : tld: com
2021-12-06 13:50:21,538 : DEBUG : lang: en
2021-12-06 13:50:21,538 : DEBUG : slow: False
2021-12-06 13:50:21,539 : DEBUG : lang_check: True
2021-12-06 13:50:21,540 : DEBUG : pre_processor_funcs: [<function tone_marks at 0x731dce40>, <function end_of_line at 0x731dcfa8>, <function abbreviations at 0x731ef0c0>, <function word_sub at 0x731ef4b0>]
2021-12-06 13:50:21,540 : DEBUG : tokenizer_func: <bound method Tokenizer.run of re.compile('(?<=\\?).|(?<=!).|(?<=？).|(?<=！).|(?<!\\.[a-z])\\. |(?<!\\.[a-z]), |(?<!\\d):|…|‥|、|¿|\\[|;|，|：|。|\\(|،|\\)|—|¡|\\]|\\\n', re.IGNORECASE) from: [<function tone_marks at 0x731ef540>, <function period_comma at 0x731ef588>, <function colon  [... cut ...]
2021-12-06 13:50:43,945 : DEBUG : headers-0: {'Referer': 'http://translate.google.com/', 'User-Agent': 'Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/47.0.2526.106 Safari/537.36', 'Content-Type': 'application/x-www-form-urlencoded;charset=utf-8', 'Content-Length': '189'}
2021-12-06 13:50:43,949 : DEBUG : url-0: https://translate.google.com/_/TranslateWebserverUi/data/batchexecute
2021-12-06 13:50:43,951 : DEBUG : status-0: 200
2021-12-06 13:50:43,956 : DEBUG : part-0 written to <_io.BufferedWriter name='/tmp/female-say.mp3'>
2021-12-06 13:50:43,957 : DEBUG : Saved to /tmp/female-say.mp3

```

De-Vincci, 2021-12-06:
ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play stereo from youtube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
stereo
['_U0PrwLpb_Q']
['_U0PrwLpb_Q', 'T3E9Wjbq44E']
['_U0PrwLpb_Q', 'T3E9Wjbq44E', 'bY1ydcTV-hA']
['_U0PrwLpb_Q', 'T3E9Wjbq44E', 'bY1ydcTV-hA', '02lW6TpjVJk']
['_U0PrwLpb_Q', 'T3E9Wjbq44E', 'bY1ydcTV-hA', '02lW6TpjVJk', 'yp9uCy_oudU']
'dislike_count'
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

amitron01, 2021-12-06:
Terminal log

```
pi@raspberrypi:~ $ /home/pi/env/bin/python -u /home/pi/GassistPi/src/main.py --project_id 'mirrorv3-87bb8' --device_model_id 'mirrorv3-87bb8-mirrorv3-tbtjeo'
USB-MIC-JACK
Initializing GPIOs 5 and 6 for assistant activity indication
/home/pi/env/lib/python3.7/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
  warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
device_model_id: mirrorv3-87bb8-mirrorv3-tbtjeo
device_id: 1FC2027C98B5899D1329A30DDEA02778

ON_MUTED_CHANGED:
  {"is_muted": false}

ON_START_FINISHED

ON_MEDIA_STATE_IDLE

ON_CONVERSATION_TURN_STARTED

ON_END_OF_UTTERANCE

ON_END_OF_UTTERANCE

ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play Waka Waka song from YouTube"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
waka waka song
['pRpeEdMmmQ0']
['pRpeEdMmmQ0', '1zacYmrdexA']
['pRpeEdMmmQ0', '1zacYmrdexA', 'Ntn1-SocNiY']
['pRpeEdMmmQ0', '1zacYmrdexA', 'Ntn1-SocNiY', 'gVfgTw_W_JY']
['pRpeEdMmmQ0', '1zacYmrdexA', 'Ntn1-SocNiY', 'gVfgTw_W_JY', '6U58KYsh09w']
'dislike_count'
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.25.10; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:03] Decoding of female-say.mp3 finished.
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}

```

De-Vincci, 2021-12-06:
@shivasiddharth I'm so sorry. I already uploaded the command line terminal reply. I hope it fixes. Thank you in advance!

amitron01, 2021-12-06:
Right now not work youtube. Please fix it

closed 2021-12-06 by shivasiddharth

De-Vincci, 2021-12-06:
Thank you very much!

jaedeezy, 2021-12-06:
That did it. Really do appreciate your continued work on this project.

shivasiddharth, 2021-12-06:
@jaedeezy Thanks for your donation.

amitron01, 2021-12-06:
Than you so much solved this problem