shivasiddharth/GassistPi #1145: Executing installer returns error.

nadunchandrabahu, 2021-12-16:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

I'm using the master branch.

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   

processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03114
Serial		: 100000008e8b8abc
Model		: Raspberry Pi 4 Model B Rev 1.4

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
[... 128 lines cut ...]
### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes/No

### 4. Is the assistant service restarting automatically?
Yes/No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```

```

### Paste the contents of the terminal below
```

```

shivasiddharth, 2021-12-16:
Couple of things:
1. https://deb.nodesource.com/node_14.x is not in the offical sources list.
2. Is that 64 bit version of OS or 32 bit ?

nadunchandrabahu, 2021-12-16:
Hi,

1. OK I see, what can I do to fix that? And what is the "official sources list"?, I've only been referring to your GitHub page and GassistPi documentation, is there another source? (Sorry for the newbie questions)
2. It's the 32 bit version, do I need to use the 64 bit OS?

Thank you for all your work.

shivasiddharth, 2021-12-16:
Do: 
```
sudo nano /etc/apt/sources.list
```
Share the output of what you are seeing.

nadunchandrabahu, 2021-12-17:
[image]
^ this is what I'm seeing

shivasiddharth, 2021-12-17:
In the pip requirements file, change the line ```git+https://github.com/shivasiddharth/pafy.git``` to ```pafy``` And then try again.

nadunchandrabahu, 2021-12-17:
@shivasiddharth Installation works now! Thank you so much for your help.

shivasiddharth, 2021-12-17:
Will close this issue. But consider using a newer version of the OS. What you are using is a year old.

shivasiddharth, 2021-12-17:
Apologies. My bad ! I saw that as i did not see the 2021 properly in the realse date.

closed 2021-12-17 by shivasiddharth

nadunchandrabahu, 2021-12-17:
@shivasiddharth it is infact the year old OS I'm using now, that is Raspbian Buster. I was using Bullseye, the latest OS and still running into issues, and since your guide says to use either Raspbian Stretch or Buster, I switched to buster. I plan on switching to the newer OS and testing it out with the above fix, though I recall running into a different set of issues on there as well.

shivasiddharth, 2021-12-17:
Right now Bullseye has a lot of problems. Thats why Raspberry Pi foundation has rolled back to a legacy version. If you want to try the Bullseye version, git clone the Snowboy-Elim branch. That works with Raspbian Bullseye 32 bit.

Instead of ```git clone https://github.com/shivasiddharth/GassistPi``` 
use ```git clone https://github.com/shivasiddharth/GassistPi -b Snowboy-Elim```  

Other than this minor change, rest all the steps are same from the guide.

nadunchandrabahu, 2021-12-17:
I see, thank you.