shivasiddharth/GassistPi #1149: Made YouTube work again (dislike_count)

ContraNL, 2022-01-05:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ? latest

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```

### Describe the bug:
```
YouTube not working right and wrong Dutch translation in keywords
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
'dislike_count'
```

### Paste the contents of the terminal below
```
Not necessary, did the work for you
```

Hi @shivasiddharth ,

I like your project and would like to tribute a little seeing I had a problem getting YouTube to work right.
First of when I changed the language to Dutch I only got some weird link and after troubleshooting I found out that Play has not been translated to the best word

So if possible please change keywords_nl.yaml so that Play: 'spelen' is changed to Play: 'speel'
This because spelen is plural to speel and you want to play one thing.

After that I ran into the 'dislike_count' error I have seen people mention before.
I found out that this problem resides within pafy.

If you go into backend_youtube_dl.py and comment "self._dislikes = self._ydl_info['dislike_count']" verything works great.
This may be because of the changes YouTube made to counteract the dislike wars.

Hope this helps, keep up the great work!!

shivasiddharth, 2022-01-05:
I fixed the YouTube issue sometime back. SO its should work now. 
It would be great if you can take a look into the https://github.com/shivasiddharth/GassistPi/blob/master/src/keywords_nl.yaml keywords file suggest changes to make it close the actual language !

ContraNL, 2022-01-06:
For me downloading 5.0 it did not. So maybe you can update the releases?
Below I have also included the altered keywords_nl document, I don't know if you support multiple translations for all values so check the changed information to see if you support it in your code.

I've also noticed that GassistPi crashes if you say 'from Spotify' wrong. So if you would say "Play ... on Spotify" (in Dutch) the application crashes and starts up again. I will try and find the problem, correct it, and come back to this issue to inform you of the changes I made.

`# Keywords are certain specific word/words that must be present in the user query/command to trigger the custom action.
# Feel free to modify, but make sure that each custom action has unique keyword.
# CHANGE THE WORD PRESENT WITHIN THE QUOTES ONLY. DO NOT ALTER OTHERS

Keywords:
  # Magic Mirror
  Magic_mirror:
    - 'magic mirror'
  # Sending recipe to mobile devices
  Recipe_pushbullet:
    - 'ingredients'
  # Kickstarter campaign tracking
  Kickstarter_tracking:
    - 'kickstarter'
  # Pi power off, servo control and Pi GPIO control
  Pi_GPIO_control:
    - 'trigger'
  # Stream songs from YouTube
  YouTube_music_stream:
    - 'youtube'
  # Stop music and radio playback
  Stop_music:
    - 'stop'
  # ESP control
  ESP_control:
    - 'wireless'
    - 'draadloos'
  # Parcel tracking
  Parcel_tracking:
    - 'parcel'
    - 'pakket'
  # RSS feed playback
  RSS:
    - 'feed'
    - 'quote'
  # Kodi actions
  Kodi_actions:
    - 'on kodi'
    - 'op kodi'
  # Music pause/resume
  Pause_resume:
    - 'pauzeer muziek'
    - 'doorgaan met muziek'
  # Change tracks
  Track_change:
    Next:
      - 'speel volgende'
      - 'speel volgende nummer'
      - 'speel het volgende nummer'
    Previous:
      - 'speel vorige'
      - 'speel vorige nummer'
      - 'speel het vorige nummer'
  #Control music/vlc volume
  VLC_music_volume:
    - 'muziek volume'
  # Refresh music lists. Both the words given below should be present in the command issued.
  Music_index_refresh:
    - 'refresh'
    - 'muziek'
  # # Streaming music from google music
  # Google_music_streaming:
  #   - 'google muziek'
  # Streaming music from spotify
  Spotify_music_streaming:
    - 'spotify'
  # Streaming music from gaana
  Gaana_music_streaming:
    - 'gaana.com'
  Deezer_music_streaming:
    - 'deezer'
  Send_sms_clickatell:
    - 'clickatell'
  notify_TTS:
    - 'notify'
    - 'informeer'
  # Send message vocal to pushbullet
  Send_Message:
    - 'stuur bericht'
  Wemo_Discovery:
    - 'discover emulated'
  # System volume adjustment
  Volume_up:
    - 'het volume hoger zetten'
    - 'zet het volume omhoog'
    - 'volume omhoog'
  Volume_down:
    - 'het volume zachter zetten'
    - 'zet het volume zachter'
    - 'volume omlaag'

#Do not change the Dict below
Dict:
  Album: 'album'
  All_songs: 'alle nummers'
  Artist: 'artiest'
  Brightness: 'helderheid'
  By: 'van'
  Change: 'andere'
  Colour: 'kleur'
  Decrease: 'verlagen'
  Down:  'omlaag'
  From: 'van'
  From_deezer: 'van deezer'
  From_gaana: 'van gaana.com'
  From_google_music: 'van google muziek'
  From_spotify: 'van spotify'
  From_youtube: 'van youtube'
  Increase: 'verhogen'
  Maximum: 'maximaal'
  Minimum: 'minimaal'
  'Off': 'uit'
  'On': 'aan'
  Play: 'speel'
  Reduce: 'verminderen'
  Set: 'instellen'
  To: 'naar'
  Up: 'omhoog'
`

shivasiddharth, 2022-01-06:
I have changed the setup script for the pafy (dislike) issue. It will install a fork of the pafy git.

shivasiddharth, 2022-01-06:
Do not try the Bullseye yet ! Its a mess.

ContraNL, 2022-01-06:
I know, I first started out this project with Bullseye but after trying to change your code as well as that of Magic Mirror I soon realised I would have to rewrite too much. Therefor I reflashed to Buster.

I hope to have some time for the Spotify crash tomorrow. Will let you know.

shivasiddharth, 2022-01-06:
Spotify has changed  a bit. You need to mention the localhost:8888 address in the app that you create online. After you enable Spotify in the config file, and  when you start the assistant for first time, you will be prompted for authentication. If 8888 port is not working change it to 8080.

ContraNL, 2022-01-06:
I know, I followed your guide and found the option when clicking on edit at the created app.
When I login and say remember login that part works.

I even see the Playlist names in the log but after that it crashes.

ContraNL, 2022-01-07:
I have re-authenticated after the shutdown last time and everything regarding Spotify is working now.
Is there a way so you don't have to re-authenticate everytime the raspberry has a shutdown for a day or more?

I want to use this as a Magic Mirror project and as such it won't be started every day. I want to power it on using a button connected to a smart plug that will start the pi as well as the display.

Hope to hear from you.

shivasiddharth, 2022-01-07:
Spotify authentication needs to be done once only as per my experience I last time used it. But, Google assistant needs to be re-authenticated frequently. Thats because, Google decided to modify something.