shivasiddharth/GassistPi #1154: Import error:/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found

Prasanna-16, 2022-02-20:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master
### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
```
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a02082
Serial		: 00000000ab9dd143
Model		: Raspberry Pi 3 Model B Rev 1.2

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 60f6a26ed5701eec6be5c864dd0db48fe6244fad, stage4

### Describe the bug:
```I have installed the gassit pi successfully but the assistant is neither starting automatically during boot nor manually able to start.
Its showing some Importerror:/lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found

```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```/home/${USER}/env/bin/python3 -u /home/${USER}/GassistPi/src/main.py  --project_id  'beatz-f197e'  --device_model_id  'beatz-f197e-high-beatz-2kusd8'

```

### Paste the contents of the terminal below
```Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 25, in <module>
    from google.cloud import speech
  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech.py", line 19, in <module>
    from google.cloud.speech_v1 import SpeechClient
  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/__init__.py", line 17, in <module>
    from google.cloud.speech_v1.gapic import speech_client
  File "/home/pi/env/lib/python3.7/site-packages/google/cloud/speech_v1/gapic/speech_client.py", line 22, in <module>
    import google.api_core.gapic_v1.client_info
  File "/home/pi/env/lib/python3.7/site-packages/google/api_core/gapic_v1/__init__.py", line 18, in <module>
    from google.api_core.gapic_v1 import config
  File "/home/pi/env/lib/python3.7/site-packages/google/api_core/gapic_v1/config.py", line 23, in <module>
    import grpc
  File "/home/pi/env/lib/python3.7/site-packages/grpc/__init__.py", line 22, in <module>
     from grpc import _compression
  File "/home/pi/env/lib/python3.7/site-packages/grpc/_compression.py", line 15, in <module>
    from grpc._cython import cygrpc
ImportError: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.33' not found (required by /home/pi/env/lib/python3.7/site-packages/grpc/_cython/cygrpc.cpython-37m-arm-linux-gnueabihf.so)

```
  
I have already installed gassit pi on other pi board with same specifications it never ran into this error.
I tried reinstalling the os also still getting the same error.
Hope you help sir.

shivasiddharth, 2022-02-20:
Sorry I don't support Legacy version. Legacy is a special modified version of Buster. Please use Bullsye.

closed 2022-02-20 by shivasiddharth

Prasanna-16, 2022-02-20:
Ok sir thankyou