shivasiddharth/GassistPi #1155: ImportError: /home/pi/env/lib/python3.9/site-packages/grpc/_cython/cygrpc.cpython-39-arm-linux-gnueabihf.so: undefined symbol: __atomic_exchange_8

ganeshArwat, 2022-02-22:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?
master

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
Paste your Board and OS details here.   
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2711
Revision	: c03111
Serial		: 10000000a7f44295
Model		: Raspberry Pi 4 Model B Rev 1.1

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage5

### Describe the bug:
```
when i try to install  gassistpi on my raspberry pi 4 board it shows me installtion complete please reboot. then start it not get starting at all 
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
no

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
No

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'blindassistant-d987c' --device_model_id 'blindassistant-d987c-blindassistant-e803on'
```

### Paste the contents of the terminal below
```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 25, in <module>
    from google.cloud import speech
  File "/home/pi/env/lib/python3.9/site-packages/google/cloud/speech.py", line 19, in <module>
    from google.cloud.speech_v1 import SpeechClient
  File "/home/pi/env/lib/python3.9/site-packages/google/cloud/speech_v1/__init__.py", line 17, in <module>
    from google.cloud.speech_v1.gapic import speech_client
  File "/home/pi/env/lib/python3.9/site-packages/google/cloud/speech_v1/gapic/speech_client.py", line 22, in <module>
    import google.api_core.gapic_v1.client_info
  File "/home/pi/env/lib/python3.9/site-packages/google/api_core/gapic_v1/__init__.py", line 18, in <module>
    from google.api_core.gapic_v1 import config
  File "/home/pi/env/lib/python3.9/site-packages/google/api_core/gapic_v1/config.py", line 23, in <module>
    import grpc
  File "/home/pi/env/lib/python3.9/site-packages/grpc/__init__.py", line 22, in <module>
    from grpc import _compression
  File "/home/pi/env/lib/python3.9/site-packages/grpc/_compression.py", line 15, in <module>
    from grpc._cython import cygrpc
ImportError: /home/pi/env/lib/python3.9/site-packages/grpc/_cython/cygrpc.cpython-39-arm-linux-gnueabihf.so: undefined symbol: __atomic_exchange_8

```

ganeshArwat, 2022-02-22:
when i installed it in before 3 months at that time its working normally, but when i installed it now because then it is not get started

shivasiddharth, 2022-02-22:
Share the output of 
```
source env/bin/activate
pip3 list
```

ganeshArwat, 2022-02-22:
Package                   Version
------------------------- ------------
adafruit-io               2.1
aftership                 1.3.0
cachetools                4.2.4
casttube                  0.2.1
certifi                   2021.10.8
cffi                      1.15.0
chardet                   3.0.4
charset-normalizer        2.0.12
click                     6.7
colorzero                 2.0
cryptography              36.0.1
Deprecated                1.2.13
enum34                    1.1.10
feedparser                6.0.8
futures                   3.1.1
google-api-core           1.31.5
google-api-python-client  2.37.0
google-assistant-grpc     0.2.1
google-assistant-library  1.1.0
google-assistant-sdk      0.6.0
google-auth               1.35.0
google-auth-httplib2      0.1.0
google-auth-oauthlib      0.2.0
google-cloud-speech       0.36.0
google-cloud-texttospeech 0.3.0
googleapis-common-protos  1.54.0
googletrans               3.1.0a0
gpiozero                  1.6.2
grpcio                    1.43.0
gTTS                      2.2.1
gTTS-token                1.1.4
h11                       0.9.0
h2                        3.2.0
hpack                     3.0.0
hstspreload               2021.12.1
httpcore                  0.9.1
httplib2                  0.20.4
httpx                     0.13.3
hyperframe                5.2.0
idna                      2.10
ifaddr                    0.1.7
kodi-json                 1.0.0
mock                      3.0.5
netifaces                 0.11.0
numpy                     1.21.1
oauth2client              4.1.3
oauthlib                  3.2.0
packaging                 21.3
pafy                      0.5.5
paho-mqtt                 1.4.0
pathlib2                  2.3.7.post1
pip                       22.0.3
pkg_resources             0.0.0
protobuf                  3.19.4
psutil                    5.9.0
pushbullet.py             0.12.0
pvporcupine               1.9.5
pyasn1                    0.4.8
pyasn1-modules            0.2.8
PyAudio                   0.2.11
PyChromecast              10.2.3
pycparser                 2.21
pycryptodomex             3.7.2
pyOpenSSL                 22.0.0
pyparsing                 3.0.7
python-magic              0.4.25
python-vlc                3.0.12118
pytz                      2021.3
pyusb                     1.2.1
pywemo                    0.4.39
PyYAML                    6.0
redis                     4.1.4
requests                  2.27.1
requests-oauthlib         1.3.1
rfc3986                   1.5.0
RPi.GPIO                  0.7.1a4
rsa                       4.8
setuptools                60.9.3
sgmllib3k                 1.0.0
six                       1.16.0
sniffio                   1.2.0
sounddevice               0.3.15
SoundFile                 0.10.3.post1
spidev                    3.5
spotipy                   2.19.0
tenacity                  4.12.0
uritemplate               4.1.1
urllib3                   1.26.8
websocket-client          1.2.3
wheel                     0.37.1
wrapt                     1.13.3
youtube-dl                2021.12.17
zeroconf                  0.38.3

shivasiddharth, 2022-02-22:
Do:
```
source env/bin/activate
pip3 install --upgrade grpcio
```

ganeshArwat, 2022-02-22:
now its working ,thank you so much

closed 2022-02-22 by shivasiddharth

ganeshArwat, 2022-02-22:
Thanks a lot.