shivasiddharth/GassistPi #1159: Putting in spotify details stops the Assistant from working completely.

Harry989, 2022-03-21:
**IMPORTANT NOTICE  
If you do not complete the template below it is likely that your issue will not be addressed. When providing information about your issue please be as extensive as possible so that it can be solved by as little as possible responses.**  

**FAILURE TO COMPLETE THE REQUESTED INFORMATION WILL RESULT IN YOUR ISSUE BEING CLOSED**    

### Which branch of GassistPi are you using ?

Main

### Board and OS details:    
Open a terminal and run   
```
cat /proc/cpuinfo  
cat /etc/os-release  
cat /etc/rpi-issue
```
```
processor       : 0
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : d03114
Serial          : 10000000f4047c32
Model           : Raspberry Pi 4 Model B Rev 1.4

PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Raspberry Pi reference 2022-01-28
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, f01430c9d8f67a4b9719cc00e74a2079d3834d5d, stage2
```

### Describe the bug:
```
Upon adding the correct details to use spotify and adding the Youtube API V3, When launching the assistant from either boot or manually It stays stuck and produces no output
```

### Did you go through or search the issues section to check if your issue was already discussed (either currently open issues or closed issues)?
Yes

### 2. Is the issue related to assistant installation ?
No

### If answer to question 2 is Yes, then attach the contents of terminal as a file or paste it below.
```

```

### If answer to question 2 is No, then proceed further.
### 3. Does the assistant service start normally?
Yes if I have reset the spotify settings in config.yaml to default

### 4. Is the assistant service restarting automatically?
No

### If answer to question 4 is Yes, then paste the contents of the terminal below.

```

```

### If answer to question 4 is No, then proceed further.
### Manually start the assistant. For guidelines to start the assistant manually refer [this](https://github.com/shivasiddharth/GassistPi#manually-start-the-assistant)

### Paste the command below, that crashed the assistant
```
/home/${USER}/env/bin/python -u /home/${USER}/GassistPi/src/main.py  --project_id 'gassistpi-cc2e4' --device_model_id 'gassistpi-cc2e4-gassistpi-6g86gf'
```

### Paste the contents of the terminal below
I get no terminal output until I press Ctrl+C

```
Traceback (most recent call last):
  File "/home/pi/GassistPi/src/main.py", line 29, in <module>
    from actions import say
  File "/home/pi/GassistPi/src/actions.py", line 125, in <module>
    spotify_token = util.prompt_for_user_token(username,scope=thescope,client_id=clientid,client_secret=clientsecret,redirect_uri=redirecturi)
  File "/home/pi/env/lib/python3.9/site-packages/spotipy/util.py", line 99, in prompt_for_user_token
    code = sp_oauth.get_auth_response()
  File "/home/pi/env/lib/python3.9/site-packages/spotipy/oauth2.py", line 490, in get_auth_response
    return self._get_auth_response_local_server(redirect_port)
  File "/home/pi/env/lib/python3.9/site-packages/spotipy/oauth2.py", line 459, in _get_auth_response_local_server
    server.handle_request()
  File "/usr/lib/python3.9/socketserver.py", line 294, in handle_request
    ready = selector.select(timeout)
  File "/usr/lib/python3.9/selectors.py", line 416, in select
    fd_event_list = self._selector.poll(timeout)
KeyboardInterrupt
```

Harry989, 2022-03-21:
I know the YouTube API is functioning correctly as I am able to play music through youtube with no issues

shivasiddharth, 2022-03-22:
Could be an issue with the way you have entered the spotify credentials. share the config.yaml here with the spotify credentials. Do not burr or change anything.

shivasiddharth, 2022-03-22:
Did you authenticate the spotify ?

Harry989, 2022-03-22:
> Did you authenticate the spotify ?

I am unsure what you mean by authenticating the spotify but this is my config.yaml file

``

#Config file for declaring custom actions'parameters. The parameters have been listed here in alphabetical order (A-Z).

#Adafruit_IO. Sending commands from Genuine Google Home/Assistant Devices to control Domoticz Tasmota etc.
ADAFRUIT_IO:
  ADAFRUIT_IO_CONTROL: 'Disabled'
  ADAFRUIT_IO_USERNAME: 'ENTER YOUR ADAFRUIT_IO USERNAME HERE'
  ADAFRUIT_IO_KEY: 'ENTER YOUR ADAFRUIT_IO KEY HERE'
  FEEDNAME: 'ENTER YOUR ADAFRUIT_IO FEEDNAME THAT YOU WANT TO MONITOR HERE'

#Aftership API to Track Parcels. Create a developer account with Aftership and generate a key.
#The number of couriers should match the number of tracking codes.
AFTERSHIP:
  Key: 'ENTER YOUR AFTERSHIP KEY HERE'
  Parcels:
    Courier_Name:
      - 'usps'
      - 'fedex'
    Tracking_Code:
      - '9400111699000119607527'
      - '953255692957'

#Custom conversation. Add your queries and expected assistant response. Do not add symbols or special charecters.
Conversation:
  Conversation_Control: 'Enabled'
  question:
    1:
      - 'Who are you'
    2:
      - 'Do you have a family'
  answer:
    1:
      - 'I am not the one you expect to be'
      - 'Who do you think I am'
      - 'What bothers you'
    2:
      - 'Nope i am forever alone'
      - 'The whole world is my family'

#Send sms with clickatell
Clickatell:
  Clickatell_Control: 'Disabled'
  Clickatell_API: 'ENTER_YOUR_CLICKATELL_API'
  Name:
    - 'Bob'
    - 'Adam'
  Reciever:
    - '+48123456789'
    - '+46123456789'

# Deezer user credentials
Deezer:
  Deezer_Control: 'Enabled'
  User_id: '2247762504'

# DIY-HUE declarations
DIYHUE:
  DIYHUE_Control: 'Enabled'

# Domoticz declarations
Domoticz:
  # Set this to 'Enabled' to control devices registered to Domoticz server.
  Domoticz_Control: 'Disabled'
  Server_IP:
    # If you are running the server on the Pi itself, then change the ip to 'localhost'
[... 219 lines cut ...]
Wakewords:
  Ok_Google: 'Enabled'
  Custom_Wakeword: 'Disabled'
  Wakeword_Engine: 'Picovoice'
  Picovoice_wakeword_models:
    - '/home/pi/GassistPi/src/resources/picovoice_models/americano_raspberry-pi.ppn'

#Wemo/Emulated Wemo devices
Wemo:
  Wemo_Control: 'Disabled'
  Wemo_Devices:
    IP_Address:
      - 'Enter wemo device 1 ip here'
      - 'Enter wemo device 2 ip here'
    Device_Names:
      - 'Enter wemo device 1 name here'
      - 'Enter wemo device 1 name here'

#Youtube music streaming
YouTube:
  YouTube_Control: 'Enabled'

``

I also believe I entered the credentials correctly.

shivasiddharth, 2022-03-22:
Your spotify username looks wrong ! After entering the Spotify credentials, you need to manually start the assistant. It will open up a browser tab in which you need to authenticate the Spotify. Without authenticating, the spotify will leave the assistant hanging.

Harry989, 2022-03-22:
> Your spotify username looks wrong ! After entering the Spotify credentials, you need to manually start the assistant. It will open up a browser tab in which you need to authenticate the Spotify. Without authenticating, the spotify will leave the assistant hanging.

I am running my pi headless, and what do you mean the spotify username looks wrong?

Harry989, 2022-03-22:
And that apparently, is my username.
[image]

shivasiddharth, 2022-03-22:
Till your post, no one ever reported to me an autogenerated alphanumeric username. Thats why I thought that the username was wrong. Sorry ! In most cases, what people have is a custom set username.

For spotify authentication, you need to have a display connected or at least connect to the pi via VNC. Stop the headless service and manually start the assistant. A browser window will pop up for your to confirm the spotify permission, etc.

Harry989, 2022-03-22:
> Till your post, no one ever reported to me an autogenerated alphanumeric username. Thats why I thought that the username was wrong. Sorry ! In most cases, what people have is a custom set username. [...]

Well I do have a normal username, It's Harry989. Yet this is different. Weird.

Also I have no desktop environment installed on the pi, It's just the 32bit lite img. So I'm not too sure what to do from here.

shivasiddharth, 2022-03-22:
You really need to have the desktop environment.

Harry989, 2022-03-22:
I see, Is there a way to install it without having to wipe my pi?

shivasiddharth, 2022-03-22:
It is there, but it tends to break so many other things. My sincere advise would be to start from scratch with a desktop image.

Harry989, 2022-03-22:
Damn. I'll go do that then really quick.

Harry989, 2022-03-22:
Do you know if the 64bit environment causes any issues with GassistPi?

shivasiddharth, 2022-03-22:
Google Assistant does not have a 64 bit version for arm Linux. OS has to be 32 bit.

Harry989, 2022-03-22:
> Google Assistant does not have a 64 bit version for arm Linux. OS has to be 32 bit.

Yeah I had figured, I'm just installing GassistPi as we speak.

Harry989, 2022-03-22:
I also notice I am running into a concerningly less amount of errors than I had before with the lite img.

Harry989, 2022-03-22:
Okay I have gotten to the point where it asks for auth in the browser but I keep getting this
[image]

Harry989, 2022-03-22:
Nervermind I just copied the link and opened it in a new tab which worked weirdly

Harry989, 2022-03-22:
Which brings me to another error, upon attempting to play a playlist I get:
'''
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play favourite favourite songs playlist on Spotify"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.26.4; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
favourite favourite songs playlist on spotif
2022-03-22 10:14:10,065 : ERROR : HTTP Error for GET to https://api.spotify.com/v1/users/e425a9949284439192a4184e81b1f828/playlists with Params: {'limit': 50, 'offset': 0} returned 404 due to Not found.
Fatal Python error: Bus error

Thread 0x6fcfd440 (most recent call first):
  File "/home/pi/GassistPi/src/main.py", line 263 in pushbutton
  File "/usr/lib/python3.9/threading.py", line 892 in run
  File "/usr/lib/python3.9/threading.py", line 954 in _bootstrap_inner
  File "/usr/lib/python3.9/threading.py", line 912 in _bootstrap

Thread 0xb6f1f980 (most recent call first):
  File "/home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py", line 119 in __exit__
  File "/home/pi/GassistPi/src/main.py", line 1100 in main
  File "/home/pi/GassistPi/src/main.py", line 1105 in <module>
Bus error
'''

Harry989, 2022-03-22:
I just realised that is definitely A username issue, I'll change it to what it probably should be and see if that fixes it.

Harry989, 2022-03-22:
And we have another new issue, I love fixing issues just to make more issues. Great fun
[image]
I have a feeling that from the number 0 underneath the playlist name means it isn't seeing something but there's just nothing coming out of the speakers so in my eyes that's a mild issue.

shivasiddharth, 2022-03-22:
How many songs do you have in the playlist ?

Harry989, 2022-03-22:
> How many songs do you have in the playlist ?
I have 202 songs in it, but it still fails when I try and play one with 22 songs in it.

shivasiddharth, 2022-03-22:
Are all those playlists created by you or someone else's playlist added to your account ?

shivasiddharth, 2022-03-22:
Remove the "#" in these lines. 
https://github.com/shivasiddharth/GassistPi/blob/cd416148e46a661d767c371814d398f17bccb460/src/actions.py#L1494
https://github.com/shivasiddharth/GassistPi/blob/cd416148e46a661d767c371814d398f17bccb460/src/actions.py#L1516
https://github.com/shivasiddharth/GassistPi/blob/cd416148e46a661d767c371814d398f17bccb460/src/actions.py#L1502

Then give a command to play a playlist. And then share the terminal screenshot like above.

Harry989, 2022-03-22:
This is what I get
[image]

shivasiddharth, 2022-03-22:
Its not picking any of your playlists.

Harry989, 2022-03-22:
Weird,  It does get the name right. but would capitalisation matter?

shivasiddharth, 2022-03-22:
It says there are 0 playlists against the username. The "favourite songs" is extracted from the command you gave. Its not actually from your account. The system is checking for "favourite songs" and as it sees no playlist, it prints that 0 below. remove the "#" ion this line and see what it says.
I suspect it to be some kind of username issue.

Harry989, 2022-03-22:
> It says there are 0 playlists against the username. The "favourite songs" is extracted from the command you gave. Its not actually from your account. The system is checking for "favourite songs" and [...]

What do you want me to uncomment? And yeah, I agree it must be some sort of user issue.

shivasiddharth, 2022-03-22:
oopsie forgot to paste the link https://github.com/shivasiddharth/GassistPi/blob/cd416148e46a661d767c371814d398f17bccb460/src/actions.py#L1487

There seems to be some kind of an issue between that alphanumeric username and proper custom username.

Harry989, 2022-03-22:
Yeah well now I don't think this is an error with the whole username thing
```
ON_RECOGNIZING_SPEECH_FINISHED:
  {"text": "play favourite songs from Spotify"}

stopping vlc
High Performance MPEG 1.0/2.0/2.5 Audio Player for Layers 1, 2 and 3
	version 1.26.4; written and copyright by Michael Hipp and others
	free software (LGPL) without any warranty but with best wishes

Directory: /tmp/

Terminal control enabled, press 'h' for listing of keys and functions.

Playing MPEG stream 1 of 1: female-say.mp3 ...

MPEG 2.0 L III cbr32 24000 mono

[0:01] Decoding of female-say.mp3 finished.
favourite songs
{'display_name': 'harry989', 'external_urls': {'spotify': 'https://open.spotify.com/user/harry989'}, 'followers': {'href': None, 'total': 0}, 'href': 'https://api.spotify.com/v1/users/harry989', 'id': 'harry989', 'images': [], 'type': 'user', 'uri': 'spotify:user:harry989'}
0
{'Playlists': []}
[]
ON_CONVERSATION_TURN_FINISHED:
  {"with_follow_on_turn": false}
```

Harry989, 2022-03-22:
Let's see if changing it back to the weird numeric username does anything.

Harry989, 2022-03-22:
I also realised that GassistPi doesn't auto launch on boot even though I set it to

shivasiddharth, 2022-03-22:
This is a free module, so in order to prevent any abuse, I have rate limited the number of songs that can be parsed. 
After restart, run this: `journalctl -u gassistpi.service -f`
Share the terminal contents.

Harry989, 2022-03-22:
This made an infinitely looping error, I shall only show two but it just repeats a indefinitely.
```
-- Journal begins at Sat 2021-08-07 04:58:19 BST. --
Mar 22 12:57:25 raspberrypi python[1239]:   File "/home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py", line 101 in __init__
Mar 22 12:57:25 raspberrypi python[1239]:   File "/home/pi/GassistPi/src/main.py", line 1071 in main
Mar 22 12:57:25 raspberrypi python[1239]:   File "/home/pi/GassistPi/src/main.py", line 1105 in <module>
Mar 22 12:57:25 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 22 12:57:25 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 22 12:57:25 raspberrypi systemd[1]: gassistpi.service: Consumed 2.990s CPU time.
Mar 22 12:57:25 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 8.
Mar 22 12:57:25 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 22 12:57:25 raspberrypi systemd[1]: gassistpi.service: Consumed 2.990s CPU time.
Mar 22 12:57:25 raspberrypi systemd[1]: Started Google Assistant-Voice.
Mar 22 12:57:27 raspberrypi python[1260]: [02b9ec60] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 12:57:27 raspberrypi python[1260]: [02bae140] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 12:57:28 raspberrypi python[1260]: USB-MIC-JACK
Mar 22 12:57:28 raspberrypi python[1260]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 22 12:57:28 raspberrypi sudo[1281]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/rm /tmp/GassistPi.log
Mar 22 12:57:28 raspberrypi sudo[1281]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Mar 22 12:57:28 raspberrypi sudo[1281]: pam_unix(sudo:session): session closed for user root
Mar 22 12:57:28 raspberrypi python[1260]: /home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
Mar 22 12:57:28 raspberrypi python[1260]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Mar 22 12:57:28 raspberrypi python[1260]: [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Mar 22 12:57:28 raspberrypi python[1260]: Fatal Python error: Aborted
Mar 22 12:57:28 raspberrypi python[1260]: Current thread 0xb6f66980 (most recent call first):
Mar 22 12:57:28 raspberrypi python[1260]:   File "/home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py", line 101 in __init__
Mar 22 12:57:28 raspberrypi python[1260]:   File "/home/pi/GassistPi/src/main.py", line 1071 in main
Mar 22 12:57:28 raspberrypi python[1260]:   File "/home/pi/GassistPi/src/main.py", line 1105 in <module>
Mar 22 12:57:28 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 22 12:57:28 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 22 12:57:28 raspberrypi systemd[1]: gassistpi.service: Consumed 3.081s CPU time.
Mar 22 12:57:28 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 9.
Mar 22 12:57:28 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 22 12:57:28 raspberrypi systemd[1]: gassistpi.service: Consumed 3.081s CPU time.
Mar 22 12:57:28 raspberrypi systemd[1]: Started Google Assistant-Voice.
Mar 22 12:57:31 raspberrypi python[1284]: [0175ac60] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 12:57:31 raspberrypi python[1284]: [0176a140] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 12:57:32 raspberrypi python[1284]: USB-MIC-JACK
Mar 22 12:57:32 raspberrypi python[1284]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 22 12:57:32 raspberrypi sudo[1302]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/rm /tmp/GassistPi.log
Mar 22 12:57:32 raspberrypi sudo[1302]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Mar 22 12:57:32 raspberrypi sudo[1302]: pam_unix(sudo:session): session closed for user root
Mar 22 12:57:32 raspberrypi python[1284]: /home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
Mar 22 12:57:32 raspberrypi python[1284]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Mar 22 12:57:32 raspberrypi python[1284]: [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Mar 22 12:57:32 raspberrypi python[1284]: Fatal Python error: Aborted
Mar 22 12:57:32 raspberrypi python[1284]: Current thread 0xb6fcc980 (most recent call first):
```

shivasiddharth, 2022-03-22:
What is your Audio IN and OUT device?
Share the o/p of:
```
sudo nano /home/pi/.asoundrc
sudo nano /etc/asound.conf
```

Harry989, 2022-03-22:
There's nothing in /home/pi/.asoundrc or /etc/asound.conf

Harry989, 2022-03-22:
I had to butcher some stuff because I am a Audio jack and usb mic user but the script in the guide had some errors that I couldn't solve, I did initially have this in /home/pi/.asoundrc but it seems to have been removed.
```
pcm.!default {
  type asym
  capture.pcm "mic"
  playback.pcm "speaker"
}
pcm.mic {
  type plug
  slave {
    pcm "hw:2,0"
  }
}
pcm.speaker {
  type plug
  slave {
    pcm "hw:1,0"
  }
}
```

shivasiddharth, 2022-03-22:
Open the `usb-mic-onboard-jack.sh` script and comment out the line:
https://github.com/shivasiddharth/GassistPi/blob/cd416148e46a661d767c371814d398f17bccb460/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh#L31
Re run the script using: 
`sudo ./GassistPi/audio-drivers/USB-MIC-JACK/scripts/usb-mic-onboard-jack.sh`
Before restarting run `speaker-test` in terminal and then left click on the volume icon on the menu bar and select the IN and OUT device.

Harry989, 2022-03-22:
I have done that and restarted, still no launch on boot however, I also didn't get an error and it said it installed

shivasiddharth, 2022-03-22:
Do you have something in:
```
sudo nano /home/pi/.asoundrc
sudo nano /etc/asound.conf
```
??

What is the` journalctl -u gassistpi.service -f` saying?

Harry989, 2022-03-22:
Still got nothing in /home/pi/.asoundrc but /etc/asound.conf now has this in it:
```
pcm.dsnooper {
    type dsnoop
    ipc_key 816357492
    ipc_key_add_uid 0
    ipc_perm 0666
    slave {
        pcm "hw:1,0"
        channels 1
    }
}

pcm.!default {
        type asym
        playback.pcm {
                type plug
                slave.pcm "hw:0,0"
        }
        capture.pcm {
                type plug
                slave.pcm "dsnooper"
        }
}
```
and the output for journalctl -u gassistpi.service -f is the same as before
```
-- Journal begins at Sat 2021-08-07 04:58:19 BST. --
Mar 22 14:40:13 raspberrypi python[3727]:   File "/home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py", line 101 in __init__
Mar 22 14:40:13 raspberrypi python[3727]:   File "/home/pi/GassistPi/src/main.py", line 1071 in main
Mar 22 14:40:13 raspberrypi python[3727]:   File "/home/pi/GassistPi/src/main.py", line 1105 in <module>
Mar 22 14:40:14 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 22 14:40:14 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 22 14:40:14 raspberrypi systemd[1]: gassistpi.service: Consumed 3.122s CPU time.
Mar 22 14:40:14 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 118.
Mar 22 14:40:14 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 22 14:40:14 raspberrypi systemd[1]: gassistpi.service: Consumed 3.122s CPU time.
Mar 22 14:40:14 raspberrypi systemd[1]: Started Google Assistant-Voice.
Mar 22 14:40:16 raspberrypi python[3748]: [02b45c60] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 14:40:16 raspberrypi python[3748]: [02b6d338] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 14:40:17 raspberrypi python[3748]: USB-MIC-JACK
Mar 22 14:40:17 raspberrypi python[3748]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 22 14:40:17 raspberrypi sudo[3767]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/rm /tmp/GassistPi.log
Mar 22 14:40:17 raspberrypi sudo[3767]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Mar 22 14:40:17 raspberrypi sudo[3767]: pam_unix(sudo:session): session closed for user root
Mar 22 14:40:17 raspberrypi python[3748]: /home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
Mar 22 14:40:17 raspberrypi python[3748]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Mar 22 14:40:17 raspberrypi python[3748]: ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
Mar 22 14:40:17 raspberrypi python[3748]: [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Mar 22 14:40:17 raspberrypi python[3748]: Fatal Python error: Aborted
Mar 22 14:40:17 raspberrypi python[3748]: Current thread 0xb6fd9980 (most recent call first):
Mar 22 14:40:17 raspberrypi python[3748]:   File "/home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py", line 101 in __init__
Mar 22 14:40:17 raspberrypi python[3748]:   File "/home/pi/GassistPi/src/main.py", line 1071 in main
Mar 22 14:40:17 raspberrypi python[3748]:   File "/home/pi/GassistPi/src/main.py", line 1105 in <module>
Mar 22 14:40:17 raspberrypi systemd[1]: gassistpi.service: Main process exited, code=killed, status=6/ABRT
Mar 22 14:40:17 raspberrypi systemd[1]: gassistpi.service: Failed with result 'signal'.
Mar 22 14:40:17 raspberrypi systemd[1]: gassistpi.service: Consumed 2.964s CPU time.
Mar 22 14:40:17 raspberrypi systemd[1]: gassistpi.service: Scheduled restart job, restart counter is at 119.
Mar 22 14:40:17 raspberrypi systemd[1]: Stopped Google Assistant-Voice.
Mar 22 14:40:17 raspberrypi systemd[1]: gassistpi.service: Consumed 2.964s CPU time.
Mar 22 14:40:17 raspberrypi systemd[1]: Started Google Assistant-Voice.
Mar 22 14:40:19 raspberrypi python[3770]: [01141c60] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 14:40:19 raspberrypi python[3770]: [01169338] vlcpulse audio output error: PulseAudio server connection failure: Connection refused
Mar 22 14:40:20 raspberrypi python[3770]: USB-MIC-JACK
Mar 22 14:40:20 raspberrypi python[3770]: Initializing GPIOs 5 and 6 for assistant activity indication
Mar 22 14:40:20 raspberrypi sudo[3788]:       pi : PWD=/home/pi ; USER=root ; COMMAND=/usr/bin/rm /tmp/GassistPi.log
Mar 22 14:40:20 raspberrypi sudo[3788]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Mar 22 14:40:20 raspberrypi sudo[3788]: pam_unix(sudo:session): session closed for user root
Mar 22 14:40:20 raspberrypi python[3770]: /home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
Mar 22 14:40:20 raspberrypi python[3770]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Mar 22 14:40:20 raspberrypi python[3770]: ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave
Mar 22 14:40:20 raspberrypi python[3770]: [FATAL:audio_input_stream.cc(47)] Input device could not be opened: default
Mar 22 14:40:20 raspberrypi python[3770]: Fatal Python error: Aborted
Mar 22 14:40:20 raspberrypi python[3770]: Current thread 0xb6fa4980 (most recent call first):
```

Harry989, 2022-03-22:
Here you go.
```
pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: b1 [bcm2835 HDMI 1], device 0: bcm2835 HDMI 1 [bcm2835 HDMI 1]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
card 1: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 4/4
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
pi@raspberrypi:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 2: Device [(LCS) USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
```

shivasiddharth, 2022-03-22:
Are you using https://static.cytron.io/image/cache/catalog/products/USB-MINI-MIC/Mini%20USB%20(3)-800x800.JPG this kind of mic?

Harry989, 2022-03-22:
> Are you using https://static.cytron.io/image/cache/catalog/products/USB-MINI-MIC/Mini%20USB%20(3)-800x800.JPG this kind of mic?

No I'm using some random ass microphone I got second hand off of my brother but it's pretty cheap

Harry989, 2022-03-22:
I am unsure if you are aware but both the speaker and the microphone have been fully functioning the whole time

Harry989, 2022-03-22:
I have rebooted and the assistant still hasn't launched on boot

shivasiddharth, 2022-03-22:
I am aware of that. `[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default` means that the USB Mic is not working on boot. The `journalctl -u gassistpi.service -f` still says `[FATAL:audio_input_stream.cc(47)] Input device could not be opened: default` ??

Harry989, 2022-03-22:
Well, some progress, the assistant is definitely running now, but I now have no audio, I presume this is because in the /etc/asound.conf the output device has been set to 0,0 instead of 1,0 as 0,0 is HDMI audio output instead of AUX output.
```
-- Journal begins at Sat 2021-08-07 04:58:19 BST. --
Mar 22 14:51:08 raspberrypi sudo[1105]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Mar 22 14:51:08 raspberrypi python[1027]: /home/pi/env/lib/python3.9/site-packages/google/assistant/library/assistant.py:90: DeprecationWarning: Google Assistant Library for Python is deprecated
Mar 22 14:51:08 raspberrypi python[1027]:   warnings.warn('Google Assistant Library for Python is deprecated', DeprecationWarning)
Mar 22 14:51:08 raspberrypi sudo[1105]: pam_unix(sudo:session): session closed for user root
Mar 22 14:51:08 raspberrypi python[1027]: device_model_id: gassistpi-cc2e4-gassistpi-6g86gf
Mar 22 14:51:08 raspberrypi python[1027]: device_id: 4B4361CAB1B959D5D43B41AEAD7E6B05
Mar 22 14:51:09 raspberrypi python[1027]: ON_MUTED_CHANGED:
Mar 22 14:51:09 raspberrypi python[1027]:   {"is_muted": false}
Mar 22 14:51:12 raspberrypi python[1027]: ON_START_FINISHED
Mar 22 14:51:12 raspberrypi python[1027]: ON_MEDIA_STATE_IDLE
Mar 22 14:55:09 raspberrypi python[1027]: ON_CONVERSATION_TURN_STARTED
Mar 22 14:55:14 raspberrypi python[1027]: ON_END_OF_UTTERANCE
Mar 22 14:55:14 raspberrypi python[1027]: ON_END_OF_UTTERANCE
Mar 22 14:55:14 raspberrypi python[1027]: ON_RECOGNIZING_SPEECH_FINISHED:
Mar 22 14:55:14 raspberrypi python[1027]:   {"text": "what's the weather like today"}
Mar 22 14:55:14 raspberrypi python[1027]: ON_RESPONDING_STARTED:
Mar 22 14:55:14 raspberrypi python[1027]:   {"is_error_response": false}
Mar 22 14:55:23 raspberrypi python[1027]: ON_RESPONDING_FINISHED
Mar 22 14:55:23 raspberrypi python[1027]: ON_CONVERSATION_TURN_FINISHED:
Mar 22 14:55:23 raspberrypi python[1027]:   {"with_follow_on_turn": false}
```

Harry989, 2022-03-22:
Yes, changing the audio output to 1,0 has fixed the issue. It now launches on boot with audio.

shivasiddharth, 2022-03-22:
Cool ! Open the raspi-config and force audio via 3.5mm jack. Raspberry Pi OS software is notorious to reset the audio out to HDMI often.

Harry989, 2022-03-22:
Weird thing is I disabled the HDMI output. Also I am no longer getting spotify playlists to work, so that's interesting

Harry989, 2022-03-22:
Nevermind another reboot fixed it.

closed 2022-03-22 by shivasiddharth